rockowitz/ddcutil #327: ddcutil core dump when DSA2 cache file not found/created

agspoon, 2023-07-14:
ddcutil will throw a SEGV if it can't find or create the DSA2 cache file.  Found while running from a Linux udev rule where $HOME is not defined.

The function `dsa2_save_persistent_stats()` calls `dsa2_stats_cache_file_name()`, which returns a NULL char *.  The return value is passed unchecked to `fopen_mkdir`, which passes along the NULL pointer to `strrchr()` (also unchecked).  The NULL de-reference at this point triggers a SEGV signal.

I'm currently working around the issue by using the ` --disable-dsa2` option which has the side-effect of not generating the statistics file.

Even if the NULL pointer bug is fixed, it would good to have a way to avoid any error output in the event the statistics cannot be written (e.g a read-only file system, etc.), or a means to explicitly disable writing the statistics file (or any file I/O for that matter).

rockowitz, 2023-07-14:
Thank you for the detailed report.  Will follow up.

rockowitz, 2023-07-14:
ddcutil now checks for NULL values returned by dsa2_stats_cache_file_name() and related functions.  which should resolve the segfault problem.  The changes have been pushed to branch 2.0.0-dev. 

If NULL values are encountered, ddcutil writes error messages to the terminal and system log. Use ***--disable-dynamic-sleep*** etc. to avoid them.

agspoon, 2023-07-14:
Thanks!  User's, the bane of every developer :)

My use case is basically that I'm too cheap to buy a KVM to switch between my two computers.  A low level aggravation that I suffer a couple dozen times a day.  I have a small USB A/B switch that I use to swap the keyboard/mouse, but had to go through a awkward button and OSD interface on my LG monitor to switch inputs.

Thanks to someone figuring out how to programmatically switch inputs on these broken LG monitors (Issue #100) , and your implementation of the `--i2c-source-addr` option, I can now automate the process.

I wrote a couple udev rules that fire whenever the USB keyboard is added or removed, that runs ddcutil to switch the monitor input accordingly.  Works great.  A poor man's KVM.  I noticed only later that my journal was filled with backtrace logs from the above SEGV fault in ddcutil (apparently the stats are written after the work is done).

I agree that suppressing the error message is not a good idea, I was just wondering if I was missing out on some important functionality by disabling the dynamic-sleep capability, when all I really wanted to avoid was writing to the cache file.  Regardless, it works for my weird corner use case, so I'm good.  Just thought I'd pass on what I found regarding the NULL pointer.

Thanks again :)

rockowitz, 2023-07-14:
Thank you for letting me know the use case. I'll keep it in mind as I'm working on ddcutil.  I'm continually amazed at the clever ways people find to use ddcutil.  

Dynamic sleep depends on having a cache file to tune the sleep-multiplier over many program executions, so disabling dynamic sleep has little impact on execution time in your case. If execution time matters, you can experiment with option ***--sleep-multiplier***.

closed 2023-07-16 by rockowitz