rockowitz/ddcutil #328: DELL U4924DW - videoswap/usbswitch

nobanzai, 2023-07-14:
For a few days now I am using a DELL U4924DW monitor with usb-c, kvm and pip/pbp capabilities.
I found out, that feature 0xe5 can be used for swapping video, 0xe7 for switching usb.
Quering these features, returns:

+ DISPLAY=1
+ ddcutil --display 1 getvcp 0xe5
VCP code 0xe5 (Manufacturer Specific         ): mh=0x00, ml=0x02, sh=0x00, sl=0x01
+ ddcutil --display 1 getvcp 0xe7
VCP code 0xe7 (Manufacturer Specific         ): mh=0xff, ml=0xff, sh=0x15, sl=0x00

I then tried to switch video and usb with

+ DISPLAY=1
+ ddcutil --display 1 setvcp 0xe5 0xf000
Setting value failed for feature e5, rc=DDCRC_NULL_RESPONSE(-3002): received DDC null response
+ ddcutil --display 1 setvcp 0xe7 0xff00
Setting value failed for feature e7, rc=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message

Both video and usb are swapped as desired, but I also get the above shown messages.
Afterwards the OSD can't be activated until the monitor is switched off and back on.

What am I doing wrong here?

ddcutil-1.4.1-2.2.x86_64 on openSUSE Tumbleweed 20230712

TIA.
Bye.
Michael.

rockowitz, 2023-07-14:
1) Try using ***--noverify*** on the **setvcp** command.  Features xE5 and xE7 may be write-only.  As they are manufacturer-specific features and not in the MCCS spec, **ddcutil** has no way of knowing that they are write-only. 

2) If your monitor supports feature xCA (OSD/Button Control) it may be possible to re-enable the OSD using that feature.  See **ddcutil vcpinfo ca --verbose** for values.  (If the VCP version of your monitor is 2.2, the high order byte (SH) of the value is used for extended button control.  See the [MCCS spec](https://milek7.pl/ddcbacklight/mccs.pdf) for details.)

rockowitz, 2023-07-14:
**ddcutil setvcp ca 2**

nobanzai, 2023-07-14:
Doesn't help - osd still can't be activated after setting 0xe5/0xe7.
Besides that "ddcutil getvcp 0xca" already returns 0x02, so osd should be active.

nobanzai, 2023-07-14:
0xe7 is working without problems, only 0xe5 leads to the osd problem.
Besides that, 0xe5 seems to destabilize the ucb-c connection. A few times the monitor lost the connection to the notebook connected via usb-c.

nobanzai, 2023-07-14:
It seems to be a timing problem - if I wait for 2 seconds between the two setvcp commands, everything is ok: OSD menu can be activated and no usb-c problems so far.

nobanzai, 2023-07-14:
Works for me.

closed 2023-07-14 by nobanzai