rockowitz/ddcutil #329: `make DESTDIR="/path/to/dir" install` is broken

orestisfl, 2023-07-14:
After https://github.com/rockowitz/ddcutil/commit/994f4aa8fe4f514d42a4022ac70e7358c9dc7b79, it seems that running `make install` with a custom `DESTDIR` is not working:

```
Making install in src
make[1]: Entering directory '/tmp/xx/ddcutil/src'
Making install in util
make[2]: Entering directory '/tmp/xx/ddcutil/src/util'
make[3]: Entering directory '/tmp/xx/ddcutil/src/util'
make[3]: Nothing to be done for 'install-exec-am'.
make[3]: Nothing to be done for 'install-data-am'.
make[3]: Leaving directory '/tmp/xx/ddcutil/src/util'
make[2]: Leaving directory '/tmp/xx/ddcutil/src/util'
Making install in usb_util
make[2]: Entering directory '/tmp/xx/ddcutil/src/usb_util'
make[3]: Entering directory '/tmp/xx/ddcutil/src/usb_util'
make[3]: Nothing to be done for 'install-exec-am'.
make[3]: Nothing to be done for 'install-data-am'.
make[3]: Leaving directory '/tmp/xx/ddcutil/src/usb_util'
make[2]: Leaving directory '/tmp/xx/ddcutil/src/usb_util'
Making install in base
make[2]: Entering directory '/tmp/xx/ddcutil/src/base'
make[3]: Entering directory '/tmp/xx/ddcutil/src/base'
make[3]: Nothing to be done for 'install-exec-am'.
make[3]: Nothing to be done for 'install-data-am'.
make[3]: Leaving directory '/tmp/xx/ddcutil/src/base'
make[2]: Leaving directory '/tmp/xx/ddcutil/src/base'
Making install in vcp
make[2]: Entering directory '/tmp/xx/ddcutil/src/vcp'
make[3]: Entering directory '/tmp/xx/ddcutil/src/vcp'
make[3]: Nothing to be done for 'install-exec-am'.
make[3]: Nothing to be done for 'install-data-am'.
make[3]: Leaving directory '/tmp/xx/ddcutil/src/vcp'
make[2]: Leaving directory '/tmp/xx/ddcutil/src/vcp'
Making install in i2c
make[2]: Entering directory '/tmp/xx/ddcutil/src/i2c'
make[3]: Entering directory '/tmp/xx/ddcutil/src/i2c'
make[3]: Nothing to be done for 'install-exec-am'.
make[3]: Nothing to be done for 'install-data-am'.
make[3]: Leaving directory '/tmp/xx/ddcutil/src/i2c'
make[2]: Leaving directory '/tmp/xx/ddcutil/src/i2c'
Making install in usb
make[2]: Entering directory '/tmp/xx/ddcutil/src/usb'
make[3]: Entering directory '/tmp/xx/ddcutil/src/usb'
make[3]: Nothing to be done for 'install-exec-am'.
make[3]: Nothing to be done for 'install-data-am'.
make[3]: Leaving directory '/tmp/xx/ddcutil/src/usb'
make[2]: Leaving directory '/tmp/xx/ddcutil/src/usb'
Making install in dynvcp
make[2]: Entering directory '/tmp/xx/ddcutil/src/dynvcp'
make[3]: Entering directory '/tmp/xx/ddcutil/src/dynvcp'
make[3]: Nothing to be done for 'install-exec-am'.
make[3]: Nothing to be done for 'install-data-am'.
make[3]: Leaving directory '/tmp/xx/ddcutil/src/dynvcp'
make[2]: Leaving directory '/tmp/xx/ddcutil/src/dynvcp'
Making install in ddc
make[2]: Entering directory '/tmp/xx/ddcutil/src/ddc'
make[3]: Entering directory '/tmp/xx/ddcutil/src/ddc'
make[3]: Nothing to be done for 'install-exec-am'.
make[3]: Nothing to be done for 'install-data-am'.
make[3]: Leaving directory '/tmp/xx/ddcutil/src/ddc'
make[2]: Leaving directory '/tmp/xx/ddcutil/src/ddc'
Making install in libmain
make[2]: Entering directory '/tmp/xx/ddcutil/src/libmain'
make[3]: Entering directory '/tmp/xx/ddcutil/src/libmain'
make[3]: Nothing to be done for 'install-exec-am'.
make[3]: Nothing to be done for 'install-data-am'.
make[3]: Leaving directory '/tmp/xx/ddcutil/src/libmain'
make[2]: Leaving directory '/tmp/xx/ddcutil/src/libmain'
Making install in app_sysenv
make[2]: Entering directory '/tmp/xx/ddcutil/src/app_sysenv'
[... 81 lines cut ...]
 /usr/bin/install -c -m 644 etc/modprobe.d/nvidia-i2c.conf etc/udev/rules.d/60-ddcutil-i2c.rules etc/udev/rules.d/60-ddcutil-usb.rules etc/X11/xorg.conf.d/90-nvidia-i2c.conf '/tmp/xx/dest//usr/share/ddcutil/data'
 /usr/bin/mkdir -p '/tmp/xx/dest//usr/lib/modules-load.d'
 /usr/bin/install -c -m 644 usr/lib/modules-load.d/ddcutil.conf '/tmp/xx/dest//usr/lib/modules-load.d'
 /usr/bin/mkdir -p '/tmp/xx/dest//usr/lib/udev/rules.d'
 /usr/bin/install -c -m 644 usr/lib/udev/rules.d/60-ddcutil-i2c.rules usr/lib/udev/rules.d/60-ddcutil-usb.rules usr/lib/udev/rules.d/60-libddcutil-i2c.rules '/tmp/xx/dest//usr/lib/udev/rules.d'
make  install-data-hook
make[3]: Entering directory '/tmp/xx/ddcutil/data'
(data/install-data-hook)===> Executing rule: install-data-hook
sed  -i "s|/usr|/usr|" /usr/lib/udev/rules.d/60-ddcutil-usb.rules 
sed: can't read /usr/lib/udev/rules.d/60-ddcutil-usb.rules: No such file or directory
make[3]: *** [Makefile:701: install-data-hook] Error 2
make[3]: Leaving directory '/tmp/xx/ddcutil/data'
make[2]: *** [Makefile:604: install-data-am] Error 2
make[2]: Leaving directory '/tmp/xx/ddcutil/data'
make[1]: *** [Makefile:556: install-am] Error 2
make[1]: Leaving directory '/tmp/xx/ddcutil/data'
make: *** [Makefile:543: install-recursive] Error 1
```

Full command used:
```
git clean -fdx . && NOCONFIGURE=1 ./autogen.sh && ./configure --prefix=/usr && make -j && make DESTDIR='/tmp/xx/dest/' install
```

Initially reported at https://aur.archlinux.org/packages/ddcutil-dev-git

rockowitz, 2023-07-15:
Fixed by 32cd on branch 2.0.0-dev

orestisfl, 2023-07-15:
Thank you, confirmed

closed 2023-07-16 by rockowitz