rockowitz/ddcutil #330: libddcutil segfaults with dlopen

CarterLi, 2023-07-15:
Make sure `i2c-dev` is disabled

```
~/test via C v13.1.1-gcc 
❯ lsmod | rg i2c-dev

~/test via C v13.1.1-gcc 
❯ command cat test.c
#include <ddcutil_c_api.h>
#include <dlfcn.h>
#include <unistd.h>

int main()
{
        void* ddcutil = dlopen("libddcutil.so", RTLD_LAZY);
        dlclose(ddcutil);
        sleep(1000);
}

~/test via C v13.1.1-gcc 
❯ gcc test.c -o test -g

~/test via C v13.1.1-gcc 
❯ ./test
fish: Job 1, './test' terminated by signal SIGSEGV (Address boundary error)
```

stack trace

```
#0  0x00007ffff7d51639 in watch_displays_using_udev (data=0x417890) at ddc/ddc_watch_displays.c:408
#1  0x00007ffff7af2983 in g_thread_proxy (data=0x4128f0) at ../glib/gthread.c:831
#2  0x00007ffff7e59907 in start_thread (arg=<optimized out>) at pthread_create.c:444
#3  0x00007ffff7edf870 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

ddcutil version

```
❯ ddcutil --version
ddcutil 1.4.1
Built with support for USB connected displays.
Built without function failure simulation.
Built with libdrm services.

Copyright (C) 2015-2023 Sanford Rockowitz
License GPLv2: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

CarterLi, 2023-07-15:
Ref: https://github.com/fastfetch-cli/fastfetch/issues/499

If I remove the `dlclose` it won't crash

rockowitz, 2023-07-15:
Thank you for the report.  There had been vague assertions in the past about dlopen()/dlclose() not working.  Your detailed report makes the locus of the problem obvious.

I have pushed the needed changes to branch 2.0.0-dev.  For now trace messages are emitted unconditionally - that of course will change.  More significantly, the termination function in the mainline sleeps for 4 seconds to allow for the sleep in the polling loop to complete, after which the polling loop checks if the terminate flag is set.  Is this a problem for you?  If so, the sleep duration in the polling loop could be decreased, increasing the frequency with with the loop checks for rare display hotplug events.

CarterLi, 2023-07-15:
> More significantly, the termination function in the mainline sleeps for 4 seconds to allow for the sleep in the polling loop to complete, after which the polling loop checks if the terminate flag is set. Is this a problem for you?

It's a serious problem for me. My program is performance sensitive and I don't want any delays. I'd rather not dlclose it and leak the memories.

I don't need `watch`. My program should end in seconds. Can we have a flag / option to disable the watch thread?

rockowitz, 2023-07-15:
Yes. Function **ddca_init()**, new in release 2.0.0, could be made to control whether watching for display hotplug events occurs, either through a flag on argument **opts**, or using the option string argument **libopts**.  I need to think through which is better and what the defaults should be.

rockowitz, 2023-07-15:
Function **ddca_init()** now recognizes strings "--enable-watch-displays" and "--disable-watch-displays" in its **libopts** string argument. Haven't decided yet what the default should be, but in any event if you specify "--disable-watch-displays" explicitly that will override the default.  I have pushed the changes to branch 2.0.0-dev.

rockowitz, 2023-07-16:
Why did you choose to use dlopen()/dlclose() instead of just linking to the shared library?  Is it so that your program will execute even if the required version of libddcutil is not present, or is there some other reason?

CarterLi, 2023-07-16:
> Is it so that your program will execute even if the required version of libddcutil is not present?

Yes

closed 2023-07-16 by rockowitz