rockowitz/ddcutil #332: Extra/new output from the capabilities command

digitaltrails, 2023-07-21:
I'm seeing new/unexpected text on the stdout of `capabilities` results from 2.0-dev.  **Is this just debugging or is it the new normal?**  

Should `capabilities` errors go to stderr?

The capabilities parser in `vdu_controls` doesn't parse this extra info (but could do).

```
Unparsed capabilities string: (prot(monitor)type(lcd)model(ZR24w)cmds(01 02 03 07 0C F3)vcp(02 04 05 08 0B 0C 10 12 14(04 05 08 0B) 16 18 1A 52 60(01 03 0F) 6C 6E 70 AC AE B2 B6 C0 C6C 8 C9 CA CC(01 02 03 04 05 06 08 0A 0D 14) D6(01 04 05) DF FA(00 01 02) FB FC FD FE(00 01 02 04))mswhql(1)mccs_ver(2.2)asset_eep(32)mpu(01))
Errors parsing capabilities string:
   Feature C (Invalid code)
   Feature 8 (Invalid code)
Model: ZR24w
MCCS version: 2.2
```

rockowitz, 2023-07-22:
Neither.  The capabilities string is malformed.  See the segment "C0 C6C 8 C9".  The capabilities command is reporting that instead of its usual output.

Keep in mind that the capabilities string is unreliable.  ddcutil simply reports it as best it can and does not rely on it.  (ddcui unfortunately does make use of it, but that is another discussion.)  See page [Command Capabilities](http://www.ddcutil.com/command_capabilities/) on the website. 

Unfortunately, there's no clear line between what belongs in stderr vs stdout.  At least I couldn't define one.  For every rule I came up with, there was a case for which the split felt wrong.  For example, suppose the monitor correctly reports that it does not support a feature. Seems like that belongs in stderr, but what if the getvcp command specifies a feature group instead of a specific feature. It's unlikely that all the features in a group are implemented.  Does the output belong in stdout or stderr, or should it be split up?  ddcutil generally uses stderr only for serious errors.  Dealing with user error reports is harder when the output is split up.

As a workaround, you could edit the capabilities string, and save the corrected string in $HOME/.cache/ddcutil/capabilities.

digitaltrails, 2023-07-22:
Everything works as is (with vdu_controls), so it's probably not an issue.  I think perhaps I edited out of past config files (months ago) and was surprised to see it today when I was testing running from scratch.

closed 2023-07-22 by digitaltrails