rockowitz/ddcutil #333: last_io_event.c:119: record_io_finish: Assertion `finish_time > tsrec->finish_time' failed

nishanthkarthik, 2023-08-23:
I was using my mouse wheel to scroll on the monitor icon rapidly to reduce the brightness of my desktop monitor. The brightness was coming down when suddenly powerdevil crashed. I don't have the backtrace as unfortunately drkonqi on KDE was supposed to capture the backtrace but powerdevil dying also triggered systemd to kill drkonqi. 

[image]

ddcutil: 1.4.1-1
powerdevil: 5.27.7-1

**Journal**

```
systemd[777]: plasma-powerdevil.service: Killing process 1306 (drkonqi) with signal SIGKILL.
systemd[777]: plasma-powerdevil.service: Killing process 1001 (org_kde_powerde) with signal SIGKILL.
systemd[777]: plasma-powerdevil.service: State 'stop-sigterm' timed out. Killing.
drkonqi[1306]: org.kde.drkonqi: Mapping found despite product information being provided by the application. Consider removing the mapping entry "org_kde_powerdevil"
org_kde_powerdevil[1001]: KCrash: Attempting to start /usr/lib/drkonqi
org_kde_powerdevil[1001]: KCrash: Arguments: /usr/lib/org_kde_powerdevil
org_kde_powerdevil[1001]: KCrash: Application Name = org_kde_powerdevil path = /usr/lib pid = 1001
org_kde_powerdevil[1001]: KCrash: crashing... crashRecursionCounter = 2
org_kde_powerdevil[1001]: 15 -- restarted=true
org_kde_powerdevil[1001]: 23 -- productname=Powerdevil
org_kde_powerdevil[1001]: 31 -- bugaddress=submit@bugs.kde.org
org_kde_powerdevil[1001]: 40 -- programname=KDE Power Management System
org_kde_powerdevil[1001]: 18 -- appversion=5.27.7
org_kde_powerdevil[1001]: 9 -- pid=1001
org_kde_powerdevil[1001]: 9 -- signal=6
org_kde_powerdevil[1001]: 17 -- apppath=/usr/lib
org_kde_powerdevil[1001]: 27 -- appname=org_kde_powerdevil
org_kde_powerdevil[1001]: 17 -- platform=wayland
org_kde_powerdevil[1001]: 32 -- exe=/usr/lib/org_kde_powerdevil
org_kde_powerdevil[1001]: KCrash: Attempting to start /usr/lib/org_kde_powerdevil
org_kde_powerdevil[1001]: org_kde_powerdevil: last_io_event.c:119: record_io_finish: Assertion `finish_time > tsrec->finish_time' failed.
```

[ddcutil-interrogate.log](https://github.com/rockowitz/ddcutil/files/12415168/ddcutil-interrogate.log)

rockowitz, 2023-08-24:
Thank you for the report.  I'm not sure what to make of it.  This ancient code could use some cleanup, but the assert failure is saying that an operation took 0 nanoseconds.  If this is a recurring problem I'll give it more attention. otherwise I'm just going to relax the test in release 2.0.0.

nishanthkarthik, 2023-08-24:
No problem at all! ddcutil crashes very rarely and this is the first time I am hitting this assert in the past couple of years I've used ddcutil. Otherwise, it has been incredibly robust. Thank you for considering relaxing this check in v2.

rockowitz, 2023-08-26:
@nishanthkarthik  Thanks again for the report.  On closer analysis the assert() failure was in a vestigial piece of code that has been eliminated on the 2.0.0-dev branch.

closed 2023-08-27 by rockowitz