rockowitz/ddcutil #334: Dell dock WD19S blocks DDC?

ngrigoriev, 2023-08-26:
Hi!

I have been using ddcutil for a while, successfully. I have got a new dock - Dell WD19S. Pretty cool device...except that I have noticed that ddcutil no longer works.

```
❯ sudo ddcutil detect
Invalid display
   I2C bus:  /dev/i2c-1
   DRM connector:           card0-DP-4
   EDID synopsis:
      Mfg id:               VSC - ViewSonic Corporation
      Model:                VP3881
      Product code:         57908  (0xe234)
      Serial number:        V2W201700123
      Binary serial number: 16843026 (0x01010112)
      Manufacture year:     2020,  Week: 17
   DDC communication failed

Invalid display
   I2C bus:  /dev/i2c-12
   DRM connector:           card1-eDP-1
   EDID synopsis:
      Mfg id:               CSO - California Institute of Technology
      Model:
      Product code:         5376  (0x1500)
      Serial number:
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2019,  Week: 0
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.
```

If I plug the same monitor into laptop's HDMI port directly, it works. If I plug it into HDMI port of another dock (some cheap Lenovo one), it works as well. 

Environment: Arch Linux, NVidia :( driver 535.104.05.

Attached the interrogate log: 
[ddcutil-interrogate.txt](https://github.com/rockowitz/ddcutil/files/12444406/ddcutil-interrogate.txt)

Any suggestions? It is not a bad dock, especially when it is free ;)

rockowitz, 2023-08-26:
What are the models of the Dell and Lenovo docks?  What I suspect that the Dell is a true dock, with multiple video outputs, implementing DisplayPort Multi Stream Transport (MST) to split the video signals, whereas the Lenovo has just a single video output, not implementing MST.   User reports and my testing strongly suggest that the proprietary Nvidia driver does not implement MST.  You could confirm this by (at least temporarily) using the open source Nouveau driver.

ngrigoriev, 2023-08-27:
Indeed, this is quite possible. Dell model is WD19S, pretty complex product, with multiple video outputs - a true dock, as you said. The Lenovo is ‎4X90S92381, very simple travel USB-C hub. Funny, Lenovo laptop and Lenovo hub are having some troubles working together, so I had an opportunity to switch that Dell toy. So far I love it, but the inability to switch monitor's inputs programmatically is a huge disappointment, this monitor has very inconvenient controls :)

Thanks for the explanation, it must be the reason.

closed 2023-08-27 by rockowitz