rockowitz/ddcutil #336: add-feature to configue sync monitor brightness

cosmos1721, 2023-09-03:
I've been reading the code and most find it extremely useful in many scenarios. But an additional feature/option to sync color/brightness and contrast level of different monitors would be much more beneficial. For example:

There are 3 monitors connected to a system say eDP-1(primary laptop monitor), HDMI-1-0, HDMI-1-1.

What can be done

(checklist) to sync brightness and color either for HDMI1-0 and HDMI-1-1 or subsequent available monitors.
even the brightness and color be individually synced.
brightness can be managed using the non-function key shortcuts available on the keyboard.

rockowitz, 2023-09-24:
I believe what you're asking for is to have ddcutil simultaneously apply the same changes to the brightness, contrast, color etc on multiple monitors.  This unfortunately is not possible.

First, ddcutil cannot control the laptop monitor.  eDP devices do not support DDC/CI.  

More generally, ddcutil does not know how to modify settings on different monitors to achieve the same visual effect.  For example, the maximum brightness value differs between monitors.  Even if the values were normalized, e.g.. as  a percent of maximum value, using the same normalized value would not set the same absolute brightness on all monitors.   Furthermore, brightness changes are not linear in their effect.  For example, suppose that a normalized value of 20% on monitor A produces the same brightness as a normalized value of 30% on monitor B.  There is no guarantee that normalized values of 40% on A and 60% on B produce the same brightness.

As to keyboard shortcuts, that is a function of a program or script using ddcutil.  ddcutil does not run as a background process watching for keyboard input.

cosmos1721, 2023-09-25:
Actually, me and some of my linux user friends were facing such issues, so tried creating a similar bash script and pushed on my github, you can check out [here](https://github.com/cosmos1721/display-utils).
Here, i'm fetching input of my internal monitor from dbus which i believe gives an input in percentage, so do all the monitors irrespective of any kind.

closed 2023-11-17 by rockowitz