rockowitz/ddcutil #337: Feature: detect when VCP value is unsettable

digitaltrails, 2023-09-17:
If possible, it would be good to be able to determine that a setvcp is failing because it is not permitted.  Either not permitted at all or not permitted at the moment.

**Background**: When my LG monitor is set its _**European Broadcasting Union** Custom Picture Mode_, contrast is no longer allowed to be changed:

```
% ddcutil --display 2 getvcp 12                                                                                                   
VCP code 0x12 (Contrast                      ): current value =    70, max value =   100
% ddcutil --display 2 setvcp 12 50
Verification failed for feature 12
% echo $?
1
``` 
When I encountered this situation I had recently updated my kernel and GPU driver. I figured DDC/CI was failing and blamed Nvidia.  After first spending quite a bit of time reversing out the GPU-driver and kernel, the reason became obvious, so then I had to spend quite a bit of time reversing the reversals.

If a feature has gone read only, it would be nice if the message or the return code indicated that was the reason for setvcp failures.  This is only a nice to have if easily achieved, I don't think it's worth a lot of effort.

rockowitz, 2023-09-17:
Unfortunately, you can't get there from here.

At the I2C level, the only errors when sending a Set VCP Feature request are ones that indicate that that the request was not successfully received.   There's no way of knowing whether the request was executed successfully, which is why setvcp internally performs a getvcp  (assuming verification is enabled) to test if the value has actually changed.  The only thing that ddcutil can know as to whether a feature is RW or RO is statically obtained from the Monitor Control Command Spec (possibly overridden using the user defined feature facility ). There's simply no way to query a monitor as to whether a feature currently is RW or RO.

digitaltrails, 2023-09-17:
I thought it might be intractable - at least that makes for an easy resolution.

closed 2023-09-17 by digitaltrails