rockowitz/ddcutil #338: Second display not recognized in MST/daisy-chain setup using Dell U2723QE

rnwst, 2023-09-24:
The purpose of this issue is to potentially help someone else coming across this issue. 

`ddcutil` didn't recognize my second display, which was the second/last display in a USB-C MST setup. The first display is a Dell U2723QE. This was despite my user having read-write permissions for all i2c devices, and the `i2c-dev` kernel module being loaded.

The solution turned out to be a setting in the OSD menu of the Dell U2723QE. 'Display' -> 'Multi-Monitor Sync' was turned on. Once it was turned off, everything worked correctly.

### Version/Config info
ddcutil 1.4.1
kernel version 6.5.3
driver: i915 (intel graphics)

rnwst, 2023-09-24:
Solution: In the monitor's OSD, go to 'Display' -> 'Multi-Monitor Sync' and turn it off. The second display will subsequently be recognized by `ddcutil`.

closed 2023-09-24 by rnwst

rockowitz, 2023-09-24:
@rnwst Thank you for posting this information. I'm sure it will save some other people hours of aggravation.

rockowitz, 2023-09-24:
@rnwst You may want to follow this freedesktop.org bug thread [Same MST display appears as 2 /dev/i2c devices](https://gitlab.freedesktop.org/drm/intel/-/issues/3605).  Video driver support for I2C over MST is a work in progress.

rockowitz, 2023-09-24:
@rnwst  Your posting prompted me to create a new [wiki page](https://github.com/rockowitz/ddcutil/wiki/Notes-on-Specific-Monitors) that can be a community maintained repository for information about specific monitors.  It is intended as a replacement for website page [Notes on Specific Monitors](https://www.ddcutil.com/monitor_notes/), which has not kept up with the torrent of information about specific monitors.

rnwst, 2023-09-24:
> @rnwst You may want to follow this freedesktop.org bug thread [Same MST display appears as 2 /dev/i2c devices](https://gitlab.freedesktop.org/drm/intel/-/issues/3605). Video driver support for I2C over MST is a work in progress.

Thank you for that link. I noticed that as well (`ddcutil detect` reporting the first monitor in the chain twice). Based on the thread you linked above, it seems that you do not have an MST setup - I'd be happy to supply you with logs and the likes for debugging purposes, if desired. This issue doesn't seem to have any disastrous consequences though - I'm using `ddcutil` to set monitor brightness on all three reported (two actual) monitors at once, and it works.

rnwst, 2023-09-24:
> @rnwst Your posting prompted me to create a new [wiki page](https://github.com/rockowitz/ddcutil/wiki/Notes-on-Specific-Monitors) that can be a community maintained repository for information about [...]

That's awesome, thank you!

rockowitz, 2023-09-25:
@rnwst Thank you for the offer of help.  

At this point I have a StarTech MST hub and a Dell U2419 (which supports MST chaining) on the test bench. What I'm seeing with the amdgpu driver is that the phantom display problem is solved with the hub, but still exists when chaining through the monitor.  Once release 2.0.0 is out I'll hopefully find the time to build and submit a proper case for the bug report.

Unfortunately, I don't have an i915 driver environment in which to test.  I picked up an Intel Arc A380 video card for testing, but I have not been able to get it to work.  So I will take you up on your offer of help at some point.  In particular, if you're able to build the development version of the i915 driver it would be helpful for you to test the sysfs changes described in the bugzilla report once they go back in.