rockowitz/ddcutil #340: Why is the 0x37 address missing from my computer?

RiKkzu, 2023-10-10:
I'm writing through a translator, sorry.
Devices used: laptop with Intel Atom Z8350 processor (HDMI connector), HDMI - HDMI cable, HDMI - DVI-D adapter, BENQ GW2260 monitor (DVI-D connector).
In the Windows 10 system with the ClickMonitorDC_7_2 program, everything works fine, I can adjust the brightness, contrast and turn off-turn on the monitor.
I can't control the monitor in the Debian system. Help me understand the reason. Below is some information.

sudo ddcutil detect:
>Invalid display
>I2C bus:  /dev/i2c-9
>DRM connector:           card0-HDMI-A-2
   EDID synopsis:
      Mfg id:               BNQ - UNK
      Model:                BenQ GW2260
      Product code:         30916  (0x78c4)
      Serial number:        9CC06359019
      Binary serial number: 21573 (0x00005445)
      Manufacture year:     2012,  Week: 49
   DDC communication failed

lspci:
>00:00.0 Host bridge: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series SoC Transaction Register (rev 36)
00:02.0 VGA compatible controller: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Integrated Graphics Controller (rev 36)
00:03.0 Multimedia controller: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series Imaging Unit (rev 36)
00:0b.0 Signal processing controller: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series Power Management Controller (rev 36)
00:14.0 USB controller: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series USB xHCI Controller (rev 36)
00:1a.0 Encryption controller: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series Trusted Execution Engine (rev 36)
00:1f.0 ISA bridge: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series PCU (rev 36)

lspci -vvv:
>00:02.0 VGA compatible controller: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Integrated Graphics Controller (rev 36) (prog-if 00 [VGA controller])
        DeviceName:  Onboard IGD
        Subsystem: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Integrated Graphics Controller
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 162
        Region 0: Memory at 90000000 (64-bit, non-prefetchable) [size=16M]
        Region 2: Memory at 80000000 (64-bit, prefetchable) [size=256M]
        Region 4: I/O ports at f000 [size=64]
        Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
        Capabilities: <access denied>
        Kernel driver in use: i915
        Kernel modules: i915

sudo i2cdetect -l:
>i2c-0   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-1   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-2   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-3   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-4   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-5   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-6   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-7   i2c             i915 gmbus ssc                          I2C adapter
i2c-8   i2c             i915 gmbus vga                          I2C adapter
i2c-9   i2c             i915 gmbus panel                        I2C adapter
i2c-10  i2c             i915 gmbus dpc                          I2C adapter
i2c-11  i2c             i915 gmbus dpb                          I2C adapter
i2c-12  i2c             i915 gmbus dpd                          I2C adapter
i2c-13  i2c             AUX D/DP D                              I2C adapter
i2c-14  i2c             CH341 I2C USB bus 001 device 008        I2C adapter

sudo i2cdetect -y 9:
>     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- 49 -- -- -- -- -- -- 
50: 50 -- -- -- -- -- -- -- -- 59 -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --      

sudo i2cdump -y 9 0x49:
>No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................

sudo i2cdump -y 9 0x50:
>No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 00 ff ff ff ff ff ff 00 09 d1 c4 78 45 54 00 00    ........???xET..
10: 31 16 01 03 80 30 1b 78 2e 9a d5 a3 55 55 9f 27    1????0?x.???UU?'
20: 0c 50 54 21 08 00 81 80 81 c0 81 00 a9 c0 b3 00    ?PT!?.?????.???.
30: d1 c0 01 01 01 01 02 3a 80 18 71 38 2d 40 58 2c    ???????:??q8-@X,
40: 45 00 dd 0c 11 00 00 1e 00 00 00 ff 00 39 43 43    E.???..?.....9CC
50: 30 36 33 35 39 30 31 39 0a 20 00 00 00 fd 00 32    06359019? ...?.2
60: 4c 1e 53 11 00 0a 20 20 20 20 20 20 00 00 00 fc    L?S?.?      ...?
70: 00 42 65 6e 51 20 47 57 32 32 36 30 0a 20 00 9d    .BenQ GW2260? .?
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................

sudo i2cdump -y 9 0x59:
>No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................

Is there a problem in the driver? What can be done about it?

rockowitz, 2023-10-10:
I strongly suspect that the video driver on your system doesn't support I<sup>2</sup>C communication.   The **i2cdetect** output indicates that slave address x37 is not responsive. Since DDC works with the monitor on Windows, we know the monitor is not the problem.  

Please execute **ddcutil environment --verbose** and submit the output as an attachment.  I may be able to say a bit more.

RiKkzu, 2023-10-11:
The result is in the attachment.
[ddcutil.txt](https://github.com/rockowitz/ddcutil/files/12869892/ddcutil.txt)

rockowitz, 2023-10-12:
I see the following in your **environment** output: 

~~~
Checking for possibly conflicting programs...
   root        2970  0.0  0.1 267632  7076 ?        Ssl  окт10   0:00 /usr/lib/x86_64-linux-gnu/ddccontrol/ddccontrol_service
~~~

I doubt this is the problem, but please disable ddccontrol.service  before we proceed to a report a bug against the i915 driver.

RiKkzu, 2023-10-12:
Deleted ddccontrol and restarted the computer.
[ddcutil1.txt](https://github.com/rockowitz/ddcutil/files/12883534/ddcutil1.txt)

rockowitz, 2023-10-17:
I'm seeing inconsistent information in your output from **ddcutil interrogate**.  DKMS reports "I2C disabled in this kernel, not building module.", but **i2cdetect** works and **ddcutil** is able to read the EDID.

Please send output from command **lsmod** and the config file for your kernel from /boot.

RiKkzu, 2023-10-17:
Done.
[lsmod.txt](https://github.com/rockowitz/ddcutil/files/12924888/lsmod.txt)
[config-6.1.0-13-amd64.txt](https://github.com/rockowitz/ddcutil/files/12924893/config-6.1.0-13-amd64.txt)

RiKkzu, 2023-11-02:
[rockowitz](https://github.com/rockowitz) Sorry, you didn't find out what the reason is?

rockowitz, 2023-11-02:
It's pretty clearly a problem in the i915 video driver.  The i2cdetect output clearly shows that slave address x37 is unresponsive.  We know from the fact that the various I2C utilities work and the kernel config file that i2c is enabled in your kernel.  I have no idea why DKMS reports that I2C is missing, but I suspect it's a DKMS issue and independent of your problem.

Instructions for filing an i915 driver bug are found [here](https://gitlab.freedesktop.org/drm/intel/-/wikis/How-to-file-i915-bugs).  You'll want to include 
- output of **ddcutil detect**, which reports that DDC communication doesn't work for the monitor on Linux
- the fact that DDC communication does work for the monitor on Windows
- output of i2cdetect, which shows that slave address x50 (EDID) is responsive, but slave address x37 (DDC) is not, indicating the problem is in the video driver
- details of your SOC system

Let me know the bug report id when the bug is filed  - and I'll add myself to the list of those interested.

closed 2023-11-17 by rockowitz