rockowitz/ddcutil #341: Error messages/warnings are printed to stdout

gadefox, 2023-10-11:
Hi, is there any chance to print these err messages/warnings with `--verbose` option only or to `stderr`, otherwise it's pretty hard to parse the output. Thanks

```
ddcutil -t getvcp 10
(is_nvidia_einval_bug          ) nvida/i2c-dev bug encountered. Forcing future io I2C_IO_STRATEGY_FILEIO. Retrying
VCP 10 C 80 100
```
```
ddcutil 1.4.1
Built with support for USB connected displays.
Built without function failure simulation.
Built with libdrm services.
```

rockowitz, 2023-10-11:
That message is, I'm afraid, meant to be hard to ignore. It's an internal debug message that's emitted unconditionally. I'm sorry it tripped you up, but I'm glad that it got your attention and you reported it.

For background, see issue #282.  The problem arises if the nvidia driver (or more precisely the portion that is rebuilt for each kernel, e.g. by DKMS, is built with a copy of i2c in which symbol I2C_M_DMA_SAFE is undefined.  The problem was reported by a single user, and while the report enabled me to identify  how the error arose, I've never been able to find a copy of i2c.h in which the symbol is undefined.

Possibly TMI for you, but in case you're curious, I2C_M_DMA_SAFE is used internally by driver i2c-dev to tell the video driver that it's safe to use DMA.  It does not require that DMA be used.  Normally, the Nvidia driver can be called with this bit set, but it doesn't make any use of the bit.  However, if the driver is built with a copy of i2c.h in which the bit is undefined, a call into the driver with the bit set fails as having an invalid argument.

Which leads to the following questions:
- What is your distribution?
-  How is the recompiled portion of the Nvidia driver being built (e.g. DKMS) or us it being downloaded from somewhere?
-  What copies of i2c.h exist on your system, and are there any in which I2C_M_DMA_SAFE is undefined?  If so, please submit a copy of the file.

Thank you.

gadefox, 2023-10-11:
Thanks.. Well, I'm using Debian `Trixie` and yes, the proprietary NVIDIA driver (legacy `340.108`) is built by `dkms`. The `i2c.h` contains the following definition. I hope this helps..
```
#define I2C_M_DMA_SAFE          0x0200  /* use only in kernel space */
```

rockowitz, 2023-10-12:
Can you send me a copy of file nv-i2c.c from the driver source that DKMS is compiling?

Option ***--use-file-io*** applies only to the current **ddcutil** invocation.  The need to use file type io is not remembered from one execution to the next.  Or are you saying that the **is_nvidia_einval_bug** message occurs multiple times per invocation?

gadefox, 2023-10-12:
I'm not getting the error message (`is_nvidia_einval_bug`) with `--use-file-io` option..
[nv.tar.gz](https://github.com/rockowitz/ddcutil/files/12882473/nv.tar.gz)

rockowitz, 2023-10-12:
My focus on i2c.h was incomplete.  Function nv_i2c_algo_master_xfer() in current versions of nv-i2c.c rejects valid calls from i2c-dev  if nv-i2c-c was compiled with some copy of i2c.h in which I2C_M_DMA_SAFE is not defined.  But I was puzzled because I could never find such a copy of i2c.h in the wild.   

You reported that you are using legacy nvidia driver version 340.108.  In the copy of nv-i2c.h. which you sent, function nv_2c_algo_xfer() fails any call in which any flag bit other than I2C_M_READ (e.g. I2C_M_DMA_SAFE) is set.  So the problem is that an old version of the nvidia driver rejects calls from current versions of i2c-dev.  This is a much more plausible reason for why this error is seen.

Is there a hardware reason you're running the legacy driver?  If so, then ***--use-file-io*** is the proper solution. 

Thank you for your help in diagnosing this obscure problem.

gadefox, 2023-10-12:
It's the old laptop. Anyway, it seems the issue is solved.
I can find the following definitions in `i2c.h`:
```
#define I2C_M_RD                0x0001  /* guaranteed to be 0x0001! */
#define I2C_M_TEN               0x0010  /* use only if I2C_FUNC_10BIT_ADDR */
#define I2C_M_DMA_SAFE          0x0200  /* use only in kernel space */
#define I2C_M_RECV_LEN          0x0400  /* use only if I2C_FUNC_SMBUS_READ_BLOCK_DATA */
#define I2C_M_NO_RD_ACK         0x0800  /* use only if I2C_FUNC_PROTOCOL_MANGLING */
#define I2C_M_IGNORE_NAK        0x1000  /* use only if I2C_FUNC_PROTOCOL_MANGLING */
#define I2C_M_REV_DIR_ADDR      0x2000  /* use only if I2C_FUNC_PROTOCOL_MANGLING */
#define I2C_M_NOSTART           0x4000  /* use only if I2C_FUNC_NOSTART */
#define I2C_M_STOP              0x8000  /* use only if I2C_FUNC_PROTOCOL_MANGLING */
```
I changed the `nvidia` driver's code (fn. `nv_i2c_algo_master_xfer`) so the condition looks like this:
```
        /* we don't support I2C_FUNC_10BIT_ADDR, I2C_FUNC_PROTOCOL_MANGLING */
        if (msgs[i].flags & (I2C_M_TEN | I2C_M_NO_RD_ACK | I2C_M_IGNORE_NAK | I2C_M_REV_DIR_ADDR | I2C_M_STOP))
        {
            rc = -EINVAL;
            rmStatus = RM_ERR_INVALID_ARGUMENT;
        }
        else
        {
            rmStatus = rm_i2c_transfer(sp, nv, (void *)adap,
                                       (msgs[i].flags & I2C_M_RD) ?
                                           NV_I2C_CMD_READ : NV_I2C_CMD_WRITE,
                                       (NvU8)(msgs[i].addr & 0x7f), 0,
                                       (NvU32)(msgs[i].len & 0xffffUL),
                                       (NvU8 *)msgs[i].buf);
        }
```
Now I don't need to use `--use-file-io` option or I'm not getting the error message (`is_nvidia_einval_bug`). Thanks..

gadefox, 2023-10-13:
Btw. the safest change is to extend `I2C_M_DMA_SAFE` bit only.
```
if (msgs[i].flags & ~(I2C_M_RD | I2C_M_DMA_SAFE))
```

closed 2023-10-13 by gadefox

rockowitz, 2025-12-15:
In src/base/parms.h, comment out or delete the line **#define TEST_DISPSEL 1**.  There's a new algorithm for display selection.  If the macro is defined both the old and new algorithms are executed and compared.

In situations such as this, please create a new open issue.