rockowitz/ddcutil #348: ddcutil 2.2.0, strange warnings (Feature 0xXX should not exist)

OPerepadia, 2023-10-31:
I see these warnings every time when I run `ddcutil` command, for example

```
$ ddcutil detect
busno=6, Feature 0x41 should not exist but ddc_get_nontable_vcp_value() succeeds, returning mh=0x00 ml=0x64 sh=0x00 sl=0x64
busno=6, Feature 0xdd should not exist but ddc_get_nontable_vcp_value() succeeds, returning mh=0x00 ml=0x64 sh=0x00 sl=0x64
busno=6, Feature 0x00 should not exist but ddc_get_nontable_vcp_value() succeeds, returning mh=0x00 ml=0x64 sh=0x00 sl=0x64
Display 1
   I2C bus:  /dev/i2c-6
   DRM connector:           card1-DP-1
   EDID synopsis:
      Mfg id:               DEL - Dell Inc.
      Model:                DELL P2418D
      Product code:         53441  (0xd0c1)
      Serial number:        MY3ND91O0AKT
      Binary serial number: 809585492 (0x30414b54)
      Manufacture year:     2019,  Week: 4
   VCP version:         2.1
```

Here is the output of `sudo ddcutil interrogate --verbose`. Thank you
[output.txt](https://github.com/rockowitz/ddcutil/files/13214389/output.txt)

rockowitz, 2023-10-31:
As part of display detection, **ddcutil** checks how the display reports an unsupported feature.   The proper way is to set the unsupported feature flag in the getvcp reply packet.  Some monitors do not adhere to the spec and report an unsupported feature in some other way, e.g. by returning a DDC Null Response, or don't give any indication that a feature is unsupported.  The messages you are seeing indicate the monitor does not use the unsupported feature flag.

What is the output of **ddcutil detect --verbose** and **ddcutil getvcp  dd**?

OPerepadia, 2023-10-31:
ddcutil detect --verbose
```
busno=6, Feature 0x41 should not exist but ddc_get_nontable_vcp_value() succeeds, returning mh=0x00 ml=0x64 sh=0x00 sl=0x64
busno=6, Feature 0xdd should not exist but ddc_get_nontable_vcp_value() succeeds, returning mh=0x00 ml=0x64 sh=0x00 sl=0x64
busno=6, Feature 0x00 should not exist but ddc_get_nontable_vcp_value() succeeds, returning mh=0x00 ml=0x64 sh=0x00 sl=0x64
Display 1
   I2C bus:  /dev/i2c-6
      DRM connector:                         card1-DP-1
      /sys/class/drm/card1-DP-1/dpms         On
      /sys/class/drm/card1-DP-1/enabled      enabled
      /sys/class/drm/card1-DP-1/status       connected
      Driver:                                amdgpu
      I2C address 0x50 (EDID) responsive:    true 
      I2C address 0x37 (DDC)  responsive:    true 
      Is eDP device:                         false
      Is LVDS device:                        false
      /sys/bus/i2c/devices/i2c-6/name        AMDGPU DM aux hw bus 1
      PCI device path:                       /sys/devices/pci0000:00/0000:00:08.1/0000:30:00.0/drm/card1/card1-DP-1/i2c-6
   EDID synopsis:
      Mfg id:               DEL - Dell Inc.
      Model:                DELL P2418D
      Product code:         53441  (0xd0c1)
      Serial number:        MY3ND91O0AKT
      Binary serial number: 809585492 (0x30414b54)
      Manufacture year:     2019,  Week: 4
      EDID version:         1.4
      Extra descriptor:        
      Video input definition:    0xa5 - Digital Input (DisplayPort), Bit depth: 8
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:4:4 + YCrCb 4:2:2
         Standard sRGB color space: True
      White x,y:        0.313, 0.329
      Red   x,y:        0.662, 0.330
      Green x,y:        0.304, 0.614
      Blue  x,y:        0.149, 0.062
      Extension blocks: 1
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 10 ac c1 d0 54 4b 41 30   ............TKA0
      +0010   04 1d 01 04 a5 35 1e 78 3a ad 75 a9 54 4d 9d 26   .....5.x:.u.TM.&
      +0020   0f 50 54 a5 4b 00 81 00 b3 00 d1 00 71 4f a9 40   .PT.K.......qO.@
      +0030   81 80 d1 c0 01 01 56 5e 00 a0 a0 a0 29 50 30 20   ......V^....)P0 
      +0040   35 00 0e 28 21 00 00 1a 00 00 00 ff 00 4d 59 33   5..(!........MY3
      +0050   4e 44 39 31 4f 30 41 4b 54 0a 00 00 00 fc 00 44   ND91O0AKT......D
      +0060   45 4c 4c 20 50 32 34 31 38 44 0a 20 00 00 00 fd   ELL P2418D. ....
      +0070   00 31 56 1d 71 1c 01 0a 20 20 20 20 20 20 01 c1   .1V.q...      ..
   VCP version:         2.1
   Controller mfg:      RealTek
   Firmware version:    1.2
   Unable to determine how monitor reports unsupported features
   Current dynamic sleep adjustment multiplier:  0.10

```

ddcutil getvcp dd
```
busno=6, Feature 0x41 should not exist but ddc_get_nontable_vcp_value() succeeds, returning mh=0x00 ml=0x64 sh=0x00 sl=0x64
busno=6, Feature 0xdd should not exist but ddc_get_nontable_vcp_value() succeeds, returning mh=0x00 ml=0x64 sh=0x00 sl=0x64
busno=6, Feature 0x00 should not exist but ddc_get_nontable_vcp_value() succeeds, returning mh=0x00 ml=0x64 sh=0x00 sl=0x64
VCP code 0xdd (Unknown feature               ): mh=0xff, ml=0xff, sh=0x02, sl=0x01
```

rockowitz, 2023-10-31:
Your output confirms my diagnosis.   The output of **ddcutil detect --verbose** reports "Unable to determine how monitor reports unsupported features", and **getvcp** for non-existent feature xDD appears to succeed.

I will look into how to report this situation in a less confusing way, and into possibly suppressing these messages altogether for terse output.

rockowitz, 2023-11-02:
I have changed the message to hopefully be more clear that this is a monitor problem.  The message is emitted only if option ***--verbose*** is specified.  The detailed message is always written to the system log. 

Changes were applied to branch 2.0.2-dev.

ti-mo, 2023-11-10:
Hi! I have a similar setup to OP, running two S2721DGF. Also started noticing it this week after an upgrade.

On 2.0.2-dev, the `Feature 0x41 should not exist but ddc_get_nontable_vcp_value() succeeds` warnings are gone, both in `detect` and `getvcp` subcommands. Thanks for the quick fix!

closed 2023-11-17 by rockowitz

paulequilibrio, 2024-11-22:
> I have changed the message to hopefully be more clear that this is a monitor problem. The message is emitted only if option _**--verbose**_ is specified. The detailed message is always written to the system log.
> 
> Changes were applied to branch 2.0.2-dev.

@rockowitz How can I supress the message even in system log?

rockowitz, 2024-11-22:
This is a WARN level message written to the system log.  You can suppress it using option ***--syslog error*** which restricts messages written to the system long to only those with that severity level or higher.   You will. of course, lose any messages severity WARN or lower.