rockowitz/ddcutil #349: Koorui 27E3Q: detection works, setting brightness also works but changes aren't physically reflected

leledumbo, 2023-11-09:
```
$ ddcutil detect
Display 1
   I2C bus:  /dev/i2c-12
   DRM connector:           card1-DP-1
   EDID synopsis:
      Mfg id:               HKC - UNK
      Model:                27E3Q
      Product code:         10038  (0x2736)
      Serial number:        0000000000001
      Binary serial number: 1 (0x00000001)
      Manufacture year:     2023,  Week: 9
   VCP version:         2.1
```

```
$  ddcutil getvcp --bus 12 10
VCP code 0x10 (Brightness                    ): current value =     0, max value =   100
$  ddcutil setvcp --bus 12 10 50
$  ddcutil getvcp --bus 12 10
VCP code 0x10 (Brightness                    ): current value =    50, max value =   100
```
but my monitor brightness doesn't change at all, only when changed through OSD then it really changes.

rockowitz, 2023-11-09:
I;m afraid there are some monitors with poor DDC implementations where it appears that DDC has made a change but in fact only the OSD works.  

The onlyother  possibility is that the monitor is in some mode where it disables most controls.  But then I wouldn't have expected getvcp to report the change worked.

leledumbo, 2023-11-10:
Yeah, this monitor's firmware seems to be made in a rush. Some things are quite non-standard, guess I just have to live with it.

closed 2023-11-10 by leledumbo