rockowitz/ddcutil #353: ubsan int overflow

nekopsykose, 2023-11-20:
during a `ddcutil environment` when built with clang and -fsanitize=undefined:

```
device_id_util.c:546:54: runtime error: left shift of 32902 by 16 places cannot be represented in type 'int'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior device_id_util.c:546:54 in 
```

the following diff uses a uint instead which makes it fit:

```diff
diff --git a/src/util/device_id_util.c b/src/util/device_id_util.c
index b93a2c32..e6c347fb 100644
--- a/src/util/device_id_util.c
+++ b/src/util/device_id_util.c
@@ -534,10 +534,10 @@ load_device_ids(Device_Id_Type id_type, GPtrArray * all_lines) {
           {
              if (cur_node[tabct-1]) {
                 if (id_type == ID_TYPE_PCI) {
-                   gushort this_subvendor_id = 0;
-                   gushort this_subdevice_id = 0;
+                   guint this_subvendor_id = 0;
+                   guint this_subdevice_id = 0;
                    char * this_name = NULL;
-                   int ct = sscanf(a_line+tabct, "%4hx %4hx %m[^\n]",
+                   int ct = sscanf(a_line+tabct, "%4ux %4ux %m[^\n]",
                                    &this_subvendor_id,
                                    &this_subdevice_id,
                                    &this_name);

```

don't know if i'm missing something though which would make it the wrong solution

rockowitz, 2023-11-22:
Line 546 has been changed to include an explicit cast() which should address the -fsanitize=undefined complaint:

~~~
              guint this_id = (guint) this_subvendor_id << 16 | this_subdevice_id;
~~~

nekopsykose, 2023-11-22:
that works, thanks!

closed 2023-11-22 by nekopsykose