rockowitz/ddcutil #358: ddcutil 2.0 regression. (Invalid Display. DDC communication failed. causes: EIO / ETIMEDOUT). works in 1.4.1-1

CrashTD, 2023-12-20:
`ddcutil detect` on 1.4.1-1 works fine:

```
Display 1
   I2C bus:  /dev/i2c-4
   DRM connector:           card1-DP-3
   EDID synopsis:
      Mfg id:               SAM - Samsung Electric Company
      Model:                SyncMaster
      Product code:         953  (0x03b9)
      Serial number:        HS3Q800096
      Binary serial number: 1279865394 (0x4c493232)
      Manufacture year:     2008,  Week: 33
   VCP version:         2.0

Display 2
   I2C bus:  /dev/i2c-5
   DRM connector:           card1-HDMI-A-1
   EDID synopsis:
      Mfg id:               PHL - Philips Consumer Electronics Company
      Model:                247ELPH
      Product code:         49286  (0xc086)
      Serial number:        AU51211007766
      Binary serial number: 7766 (0x00001e56)
      Manufacture year:     2012,  Week: 11
   VCP version:         2.2

Display 3
   I2C bus:  /dev/i2c-6
   DRM connector:           card1-DP-1
   EDID synopsis:
      Mfg id:               ENC - Eizo Nanao Corporation
      Model:                S2202W
      Product code:         6517  (0x1975)
      Serial number:        73068119
      Binary serial number: 16843009 (0x01010101)
      Manufacture year:     2009,  Week: 45
   VCP version:         2.0
```

but 2.0.0-1 gives me this:

```
Invalid display
   I2C bus:  /dev/i2c-4
   DRM connector:           card1-DP-3
   EDID synopsis:
      Mfg id:               SAM - Samsung Electric Company
      Model:                SyncMaster
      Product code:         953  (0x03b9)
      Serial number:        HS3Q800096
      Binary serial number: 1279865394 (0x4c493232)
      Manufacture year:     2008,  Week: 33
   DDC communication failed. (getvcp of feature x10 returned Error_Info[EIO in ddc_write_read_with_retry, causes: EIO])

Display 1
   I2C bus:  /dev/i2c-5
   DRM connector:           card1-HDMI-A-1
   EDID synopsis:
      Mfg id:               PHL - Philips Consumer Electronics Company
      Model:                247ELPH
      Product code:         49286  (0xc086)
      Serial number:        AU51211007766
      Binary serial number: 7766 (0x00001e56)
      Manufacture year:     2012,  Week: 11
   VCP version:         2.2

Display 2
   I2C bus:  /dev/i2c-6
   DRM connector:           card1-DP-1
   EDID synopsis:
      Mfg id:               ENC - Eizo Nanao Corporation
      Model:                S2202W
      Product code:         6517  (0x1975)
      Serial number:        73068119
      Binary serial number: 16843009 (0x01010101)
      Manufacture year:     2009,  Week: 45
   VCP version:         2.0

Invalid display
   I2C bus:  /dev/i2c-8
   DRM connector:           card1-DP-3
   EDID synopsis:
      Mfg id:               SAM - Samsung Electric Company
      Model:                SyncMaster
      Product code:         953  (0x03b9)
      Serial number:        HS3Q800096
      Binary serial number: 1279865394 (0x4c493232)
      Manufacture year:     2008,  Week: 33
   DDC communication failed. (getvcp of feature x10 returned Error_Info[DDCRC_RETRIES in ddc_write_read_with_retry, causes: ETIMEDOUT(10)])
```
which then breaks all tools that rely on correct detection.

this is on Arch Linux. The monitor is connected via DVI to DP adapter. I will be happy to help you obtain all the necessary information and carry out all the necessary tests.

rockowitz, 2023-12-22:
I reconfigured the system I'm on now to use a DVI to DP adapter instead of plugging a DVI monitor directly into the DVI output on my RX580. I wasn't able to recreate your problem on ddcutil 2.0.0, but I did trigger similar errors on 2.0.2-dev.  The problem seems to be  related to aggressive parallelization in display detection and attempting to avoid explicitly reading the EDID by obtaining it from /sys.  

If you can, please build from branch 2.0.2-dev.  If the problem does not resolve, try disabling dynamic the dynamic sleep adjustment (***--disable-dynamic-sleep***) and disabling parallel DDC checks (***--ddc-check-async-threshold 9***).  On the other hand, if no problems are encountered, try either or both of the following:  hidden option ***--i2c-bus-check-async-threshold 1*** turns on parallel checking for EDIDs, and hidden utility option ***--f14*** enables reading the EDID from /sys.  Thanks.

CrashTD, 2023-12-24:
2.0.2-dev detect without arguments works partially. at least the double detected display is gone.

```
Invalid display
   I2C bus:  /dev/i2c-4
   DRM connector:           card1-DP-3
   EDID synopsis:
      Mfg id:               SAM - Samsung Electric Company
      Model:                SyncMaster
      Product code:         953  (0x03b9)
      Serial number:        HS3Q800096
      Binary serial number: 1279865394 (0x4c493232)
      Manufacture year:     2008,  Week: 33
   DDC communication failed. (getvcp of feature x10 returned Error_Info[EIO in ddc_write_read_with_retry, causes: EIO])

Display 1
   I2C bus:  /dev/i2c-5
   DRM connector:           card1-HDMI-A-1
   EDID synopsis:
      Mfg id:               PHL - Philips Consumer Electronics Company
      Model:                247ELPH
      Product code:         49286  (0xc086)
      Serial number:        AU51211007766
      Binary serial number: 7766 (0x00001e56)
      Manufacture year:     2012,  Week: 11
   VCP version:         2.2

Display 2
   I2C bus:  /dev/i2c-6
   DRM connector:           card1-DP-1
   EDID synopsis:
      Mfg id:               ENC - Eizo Nanao Corporation
      Model:                S2202W
      Product code:         6517  (0x1975)
      Serial number:        73068119
      Binary serial number: 16843009 (0x01010101)
      Manufacture year:     2009,  Week: 45
   VCP version:         2.0
```

with arguments seems to make things better but now its "detection failed":
```
./ddcutil detect --disable-dynamic-sleep --ddc-check-async-threshold 9   
Display 1
   I2C bus:  /dev/i2c-4
   DRM connector:           card1-DP-3
   EDID synopsis:
      Mfg id:               SAM - Samsung Electric Company
      Model:                SyncMaster
      Product code:         953  (0x03b9)
      Serial number:        HS3Q800096
      Binary serial number: 1279865394 (0x4c493232)
      Manufacture year:     2008,  Week: 33
   VCP version:         Detection failed

Display 2
   I2C bus:  /dev/i2c-5
   DRM connector:           card1-HDMI-A-1
   EDID synopsis:
      Mfg id:               PHL - Philips Consumer Electronics Company
      Model:                247ELPH
      Product code:         49286  (0xc086)
      Serial number:        AU51211007766
      Binary serial number: 7766 (0x00001e56)
      Manufacture year:     2012,  Week: 11
   VCP version:         2.2

Display 3
   I2C bus:  /dev/i2c-6
   DRM connector:           card1-DP-1
   EDID synopsis:
      Mfg id:               ENC - Eizo Nanao Corporation
      Model:                S2202W
      Product code:         6517  (0x1975)
      Serial number:        73068119
      Binary serial number: 16843009 (0x01010101)
      Manufacture year:     2009,  Week: 45
   VCP version:         2.0
```

I was just going to test ` --i2c-bus-check-async-threshold 1` as well but with that my monitors turned 2 x black and 1 complete green screen and my system rebooted. i'm not sure how to interpret the journal but looks like it somehow crashed KDE Power management system.

rockowitz, 2023-12-27:
There are several things going on here. First, attempting to read the EDID in parallel from all /dev/i2c devices can sometimes, but not always fail.  Reading the EDID from /sys (using ***--f14***) avoids this problem, but can encounter obscure error in the EDID that amdgpu writes to sys, which I was hoping to avoid.  For now, the default is ***--disable-i2c-bus-check-async_threshold***. 

Have you tried setting a high  fixed sleep-multiplier, e.g. ***--sleep-multiplier 2.0 --disable-dynamic-sleep***.  That Samsung monitor is rather old, and its DDC implementation may be marginal.  Have you tried another monitor in the DP-3 slot?  And do you know if your DP-DVI cable/adapter is active or passive?

CrashTD, 2023-12-30:
sorry for the late reply. busy days.

`--sleep-multiplier 2.0 --disable-dynamic-sleep` jumps between working and giving correct output to "DDC communication failed" and "VCP Version: Detection failed". the adapter is a active one, but as far i remember not exactly high-quality. the Monitor in the DP-1 slot is connected via the same adapter, so i don't think this is the problem. but i can switch it with the Samsung when i get the chance and test.

i'm not at home until new year so i can't do further testing. will update here as soon i can. 

I would like to thank you for the fast and friendly communication so far and wish you a happy new year!

CrashTD, 2024-01-04:
I think i can can close this. The monitor in question got replaced now and the new gets detected perfect. so I'll file this under strange quirk of this specific device.

closed 2024-01-04 by CrashTD

Matt-1-2-3, 2024-01-28:
I am having the same issue with my LG OLED 4k C3 tv

:ddcutil detect
Invalid display
   I2C bus:  /dev/i2c-5
   DRM connector:           card1-HDMI-A-1
   EDID synopsis:
      Mfg id:               GSM - Goldstar Company Ltd (LG)
      Model:                LG TV SSCR2
      Product code:         33229  (0x81cd)
      Serial number:        
      Binary serial number: 16843009 (0x01010101)
      Manufacture year:     2023,  Week: 1
   DDC communication failed. (getvcp of feature x10 returned Error_Info[EIO in ddc_write_read_with_retry, causes: EIO])

Display 1
   I2C bus:  /dev/i2c-6
   DRM connector:           card1-DP-1
   EDID synopsis:
      Mfg id:               NEC - NEC Corporation
      Model:                EA243WM
      Product code:         26725  (0x6865)
      Serial number:        32117493NA
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2013,  Week: 6
   VCP version:         2.0

rockowitz, 2024-01-28:
@Matt-1-2-3  Please ensure you are running the new release 2.1.2?  Then submit the output of **ddcutil interrogate**

Matt-1-2-3, 2024-01-28:
I am attaching the output of interrogate as a text file. It's long and my user name was replaced with "myuser" and hostname with "myhost"

ddcutil --version
ddcutil 2.1.2
Copyright (C) 2015-2024 Sanford Rockowitz
License GPLv2: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
[output.txt](https://github.com/rockowitz/ddcutil/files/14078150/output.txt)

rockowitz, 2024-01-29:
The output of command **i2cdetect -y 5** on line 631 shows that while slave address x50 (EDID)  is responsive, slave address x37 (DDC) is not.  So whatever problem you're having is outside of **ddcutil**'s control.  I see that you're running on a Zen kernel.  I suggest you try using a stock kernel.  If the problem resolves, i.e. **i2cdetect** reports slave address x37 as responsive, the problem is an issue for the Zen developers.

Matt-1-2-3, 2024-01-29:
Ill reboot and try stock kernel

Matt-1-2-3, 2024-01-29:
Unfortunately stock kernel with the same results, new log uploaded
[output_stock_kernel.txt](https://github.com/rockowitz/ddcutil/files/14078482/output_stock_kernel.txt)

rockowitz, 2024-01-29:
The i2cdetect output in the interrogate log is unchanged.  I don't see why it would show anything different, but for absolute certainty try running i2cdetect directly from the terminal, outside of ddcutil.

Matt-1-2-3, 2024-01-29:
from the cli itself

me@pc[~]:i2cdetect -y 5
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: 30 -- -- -- -- -- -- -- -- -- 3a -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: 50 -- -- -- 54 -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

sazawal, 2024-09-27:
> I am having the same issue with my LG OLED 4k C3 tv
>
> :ddcutil detect Invalid display I2C bus: /dev/i2c-5 DRM connector: card1-HDMI-A-1 EDID synopsis: Mfg id: GSM - Goldstar Company Ltd (LG) Model: [...]

I have the same issue with LG C4 OLED

rockowitz, 2024-09-28:
@sazawal Have you tried running command **i2cdetect -y 5** as above?  Does it show slave address x37 as being active? This is a TV, not a monitor, so I would not expect it to support DDC/CI.

sazawal, 2024-09-28:
I get blank lines, so I guess it is the TV issue

$ sudo i2cdetect -y 5
Warning: Can't use SMBus Quick Write command, will skip some addresses
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                                                 
10:                                                 
20:                                                 
30: -- -- -- -- -- -- -- --                         
40:                                                 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60:                                                 
70:

sazawal, 2024-09-28:
For those who come here with the LG TV issue, I found a package for controlling LG TVs over wifi. Check this out, https://aur.archlinux.org/packages/lgwebosremote-git