rockowitz/ddcutil #363: Release 2.1.0

rockowitz, 2024-01-17:
ddcutil 2.1.0 is now available as branch 2.1.0-release and as tag v2.1.0.  
For details, see the [ddcutil website](https://www.ddcutil.com/release_notes) and file CHANGELOG.md in the source tree.

The development branch is now 2.1.1-dev.

captn3m0, 2024-01-19:
Thanks for this release, using the `--bus` argument now seems stable. `setvcp` would earlier error out while targeting a single bus, but now works fine for me.

solarisfire, 2024-01-19:
This is really not playing well with KDE Plasma... Had to roll back to 2.0.0

rockowitz, 2024-01-19:
@solisfire Can you be more specific as to how 2.1.0 is not "playing well" with Plasma?  There is an issue with PowerDevil. which loads libddcutil dynamically without regard for whether it has loaded libddcutil.4 or libddcutil.5, but that incompatibility has existed since release 2.0.0.

solarisfire, 2024-01-19:
The powerdevil stuff is way more pronounced in 2.1.0 than in 2.0.0, leading to constant restarts, core dumps, etc.

See: https://github.com/rockowitz/ddcutil/issues/365

And: https://bbs.archlinux.org/viewtopic.php?id=292028

And: https://bugs.kde.org/show_bug.cgi?id=476375

Until I rolled back I just had this repeating in journal every few seconds:

<pre>
Jan 19 11:08:40 solaris-arion libddcutil[205753]: Initializing libddcutil.  ddcutil version: 2.1.0, shared library: /usr/lib/libddcutil.so.5.1.0
Jan 19 11:08:40 solaris-arion org_kde_powerdevil[205753]: org.kde.powerdevil: org.kde.powerdevil.chargethresholdhelper.getthreshold failed "Charge thresholds are not supported by the kernel for this hardware"
Jan 19 11:08:40 solaris-arion org_kde_powerdevil[205753]: org.kde.powerdevil: org.kde.powerdevil.backlighthelper.brightness failed
Jan 19 11:08:40 solaris-arion systemd-coredump[205756]: Process 205730 (org_kde_powerde) of user 1000 dumped core.
                                                        
                                                        Stack trace of thread 205730:
                                                        #0  0x00007557f9cac83c n/a (libc.so.6 + 0x8e83c)
                                                        #1  0x00007557f9c5c668 raise (libc.so.6 + 0x3e668)
                                                        #2  0x00007557fb20441c _ZN6KCrash19defaultCrashHandlerEi (libKF5Crash.so.5 + 0x741c)
                                                        #3  0x00007557f9c5c710 n/a (libc.so.6 + 0x3e710)
                                                        #4  0x00007557f9cac83c n/a (libc.so.6 + 0x8e83c)
                                                        #5  0x00007557f9c5c668 raise (libc.so.6 + 0x3e668)
                                                        #6  0x00007557f9c444b8 abort (libc.so.6 + 0x264b8)
                                                        #7  0x00007557f9c443dc n/a (libc.so.6 + 0x263dc)
                                                        #8  0x00007557f9c54d26 __assert_fail (libc.so.6 + 0x36d26)
                                                        #9  0x00007557f429d2b9 n/a (libddcutil.so.5 + 0x4c2b9)
                                                        #10 0x00007557f429ff35 n/a (libddcutil.so.5 + 0x4ef35)
                                                        #11 0x00007557f42a01f5 n/a (libddcutil.so.5 + 0x4f1f5)
                                                        #12 0x00007557f42c416b n/a (libddcutil.so.5 + 0x7316b)
                                                        #13 0x00007557f42cb8a2 ddca_get_display_info_list2 (libddcutil.so.5 + 0x7a8a2)
                                                        #14 0x00007557f43b2d37 n/a (powerdevilupowerbackend.so + 0x11d37)
                                                        #15 0x00007557f43b462e _ZN23PowerDevilUPowerBackend18initWithBrightnessEb (powerdevilupowerbackend.so + 0x1362e)
                                                        #16 0x00007557fa4d0e27 n/a (libQt5Core.so.5 + 0x2d0e27)
                                                        #17 0x00007557f43a9634 _ZN23PowerDevilUPowerBackend24brightnessSupportQueriedEb (powerdevilupowerbackend.so + 0x8634)
                                                        #18 0x00007557fa4d0e27 n/a (libQt5Core.so.5 + 0x2d0e27)
                                                        #19 0x00007557fb05456a _ZN4KJob6resultEPS_NS_14QPrivateSignalE (libKF5CoreAddons.so.5 + 0x5e56a)
                                                        #20 0x00007557fb05a53c n/a (libKF5CoreAddons.so.5 + 0x6453c)
                                                        #21 0x00007557fa4d0e27 n/a (libQt5Core.so.5 + 0x2d0e27)
                                                        #22 0x00007557f499a7bd n/a (kauth_helper_plugin.so + 0xa7bd)
                                                        #23 0x00007557f499b0a1 n/a (kauth_helper_plugin.so + 0xb0a1)
                                                        #24 0x00007557faf9676e n/a (libQt5DBus.so.5 + 0x2176e)
                                                        #25 0x00007557fa4c3964 _ZN7QObject5eventEP6QEvent (libQt5Core.so.5 + 0x2c3964)
                                                        #26 0x00007557fa49bef8 _ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent (libQt5Core.so.5 + 0x29bef8)
                                                        #27 0x00007557fa4a0e5b _ZN23QCoreApplicationPrivate16sendPostedEventsEP7QObjectiP11QThreadData (libQt5Core.so.5 + 0x2a0e5b)
                                                        #28 0x00007557fa4e6ec8 n/a (libQt5Core.so.5 + 0x2e6ec8)
                                                        #29 0x00007557f8f0df69 n/a (libglib-2.0.so.0 + 0x59f69)
                                                        #30 0x00007557f8f6c367 n/a (libglib-2.0.so.0 + 0xb8367)
                                                        #31 0x00007557f8f0c162 g_main_context_iteration (libglib-2.0.so.0 + 0x58162)
                                                        #32 0x00007557fa4ead0c _ZN20QEventDispatcherGlib13processEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE (libQt5Core.so.5 + 0x2ead0c)
                                                        #33 0x00007557fa49ac04 _ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE (libQt5Core.so.5 + 0x29ac04)
[... cut ...]
                                                        #3  0x00007557f452cf32 n/a (libQt5XcbQpa.so.5 + 0x62f32)
                                                        #4  0x00007557fa2f379a n/a (libQt5Core.so.5 + 0xf379a)
                                                        #5  0x00007557f9caa9eb n/a (libc.so.6 + 0x8c9eb)
                                                        #6  0x00007557f9d2e7cc n/a (libc.so.6 + 0x1107cc)
                                                        ELF object binary architecture: AMD x86-64
</pre>

rockowitz, 2024-01-19:
I'm trying to diagnose the problem on my Fedora 39 system, which is showing the repeated starts.  I have the following lines in $HOME/.config/ddcutil/ddcutirc file;
~~~
[libddcutil]
options: --trcfunc i2c_check_ bus
options: --trcfunc scan_sys_drm_connectors
options: --syslog debug --libddcutil-trace-file xxx.log 

I get diagnostic output as expected when I use libddcutil from ddcui, but there's no trace output, either in the system log or in the file specified, when PowerDevil is executing. All I see is PowerDevil's backtrace and other messages.  Can someone tell me what I need to do to capture the libddcuti output?