rockowitz/ddcutil #364: --async deprecated but documented config options don't exist

EdJoPaTo, 2024-01-19:
Arch Linux packages 2.1.0 now.

Called with `--async` it prints: `Deprecated option ignored: --async. Use --i2c_bus_check_async_min or --ddc_check_async_min`. Using them they don't exist (`Unknown option --i2c_bus_check_async_min`)

The options in the man page don't exist either. I got suspicious as one of them seems to have a typo, a duplicate c on asynCc: `--asyncc-i2c-bus-checks-min`

https://github.com/rockowitz/ddcutil/blob/056098c30d350e5a7d53a668f7399ba8d67f443d/man/ddcutil.1#L288-L297

In the sources they are written differently and don't contain underlines:

https://github.com/rockowitz/ddcutil/blob/056098c30d350e5a7d53a668f7399ba8d67f443d/src/cmdline/cmd_parser_goption.c#L1096-L1107

I think this should be synced to be the same in all three places, the `--async` deprecation stdout hint, the man page and the actual command line parsing.

rockowitz, 2024-01-19:
Thanks for the report.  It looks like a patch release will be coming soon. 

 The options are parsed properly, but the informational messages and man page need to be corrected.

rockowitz, 2024-01-24:
The man page in branch 2.1.1-dev has been extensively edited and should reflect the discrepancies you identified.

closed 2024-02-08 by rockowitz