rockowitz/ddcutil #367: FPS drops on get/set - Nvidia non-free - Wayland

postcert, 2024-01-19:
Just reporting that I'm seeing FPS drops when performing gets and sets as I did not see an issue or related information.
I do have to say the new cache features or simply upgrading from 2.0 to 2.1.1-dev has helped mitigate the issue a bit.

Here is my interrogation log:
[ddcutil-interrogation.log](https://github.com/rockowitz/ddcutil/files/13995286/ddcutil-interrogation.log)

Hardware:
Arch - 6.6.10-lts
Nvidia 3070 - Nvidia non-free - Wayland
- Lg C3 over HDMI (Doesn't work as expected)
- Dell U2719D
- Gigabyte M27Q

postcert, 2024-01-19:
I saw in another issue you had mentioned trying `--trace i2c` so I've attached that for both of my monitors that are supported.

`❯ ddcutil --bus $NUMBER --tid --trace i2c setvcp 10 100 > i2c_trace_bus$NUMBER.log`
[i2c_trace_bus2.log](https://github.com/rockowitz/ddcutil/files/13995346/i2c_trace_bus2.log)
[i2c_trace_bus3.log](https://github.com/rockowitz/ddcutil/files/13995347/i2c_trace_bus3.log)

Otherwise the application works great :+1:

rockowitz, 2024-01-20:
1)  I don't see any mechanism by which DDC communication would affect FPS. 
2) What I see in the interrogate log is the the Dell U2719D doesn't properly report an invalid feature.  You might try option --skip-ddc-checks which improve performance slightly since you're probably only checking on features such as x10 known to be valid.
3) I think what you want to show in the logs is an invalid response coming back for getvcp 10.  Is that correct?  If ddcutil is having to do a lot of retries to get a valid response you might try --disable-dynamic-sleep.  To see how many DDC retries are required, use --stats.

postcert, 2024-01-20:
It's just a momentary drop of less than 10% fps but wanted to bring it up as I haven't seen it documented elsewhere and it affects all of the outputs. For the logs I just provided them as they seemed possibly relevant checking other past issues.
For my usage I've added caching for the $current_brightness now to avoid the lag when polling but I'll test out those flags, especially the skip-checks.
Also I was curious if there was any information on the capabilities/display cache and their expiry as I didn't see any documentation on it. I was thinking that the features (invalid or present) statuses would be cached as I was seeing an execution time improvement with the newest dev branch.

Thanks for the info!

rockowitz, 2024-01-21:
1) $current_brightness ??  Seems like a reference to some script using ddcutil.  Not my domain.
2) There's no need to expire capabilities cache entries.  For any given model of monitor, the capabilities string is constant.
3) Keep in mind that display caching is an unsupported feature, as there were too many edge cases and given all the other performance enhancements display characteristics caching turned out to provide little additional performance gain.  The --help option does not show it.

closed 2024-01-22 by postcert