rockowitz/ddcutil #369: 2.1.0: "--disable-usb" --> error: 'DEFAULT_ENABLE_USB' undeclared

AndrewAmmerlaan, 2024-01-22:
```
libtool: compile:  x86_64-pc-linux-gnu-gcc -DHAVE_CONFIG_H -I. -I../.. -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I../../src -I../../src/public -Wall -std=c11 -Werror=unused-result -Wimplicit-function-declaration -Wno-compound-token-split-by-macro -fPIC -O3 -pipe -march=skylake -mtune=skylake -flto=16 -c cmd_parser_goption.c  -fPIC -DPIC -o .libs/cmd_parser_goption.o
parsed_cmd.c: In function 'new_parsed_cmd':
parsed_cmd.c:138:8: error: 'DEFAULT_ENABLE_USB' undeclared (first use in this function); did you mean 'DEFAULT_ENABLE_UDF'?
138 |    if (DEFAULT_ENABLE_USB)
|        ^~~~~~~~~~~~~~~~~~
|        DEFAULT_ENABLE_UDF
parsed_cmd.c:138:8: note: each undeclared identifier is reported only once for each function it appears in
At top level:
cc1: note: unrecognized command-line option '-Wno-compound-token-split-by-macro' may have been intended to silence earlier diagnostics
```

Compilation is successful if `--enable-usb`. Originally reported here: https://bugs.gentoo.org/922715

closed 2024-01-23 by rockowitz

rockowitz, 2024-01-23:
Fixed, as you've already noticed, in commit a3ad61.  Thanks for the report.

AndrewAmmerlaan, 2024-01-23:
Thanks, backported this patch for Gentoo.

commit a3ad6164 (2024-01-23): missing ifdef around USB only setting (fixes #369)