rockowitz/ddcutil #370: Crash and restart loop in org_kde_powerdevil on ArchLinux KDE Plasma 5

mrvolive, 2024-01-23:
## Description
Powerdevil (KDE Power Management tool) goes on through a crash and restart loop on ArchLinux. This seems to be caused by ddcutil 2.1.0-3 as downgrading to 2.0 fixes the issue.

Here is the crash log for investigation.

## System Info
- **Operating System**: Arch Linux
- **KDE Plasma Version**: 5.27.10-1
- **Powerdevil Version**: 5.27.10-2 

## Crash Log

```
           PID: 9496 (org_kde_powerde)
           UID: 1000 (oli)
           GID: 1000 (oli)
        Signal: 6 (ABRT)
     Timestamp: Mon 2024-01-22 18:10:16 CET (15h ago)
  Command Line: /usr/lib/org_kde_powerdevil
    Executable: /usr/lib/org_kde_powerdevil
 Control Group: /user.slice/user-1000.slice/user@1000.service/background.slice/plasma-powerdevil.service
          Unit: user@1000.service
     User Unit: plasma-powerdevil.service
         Slice: user-1000.slice
     Owner UID: 1000 (oli)
       Boot ID: df3068760f3945fb92abad4bb33c58ed
    Machine ID: e4785762de3745599c5d9c6d93a22bba
      Hostname: oli-arch
       Storage: /var/lib/systemd/coredump/core.org_kde_powerde.1000.df3068760f3945fb92abad4bb33c58ed.9496.1705943416000000.zst (inaccessible)
       Message: Process 9496 (org_kde_powerde) of user 1000 dumped core.
                
                Stack trace of thread 9497:
                #0  0x000074ddaa6ac83c n/a (libc.so.6 + 0x8e83c)
                #1  0x000074ddaa65c668 raise (libc.so.6 + 0x3e668)
                #2  0x000074ddabadb88f _ZN6KCrash19defaultCrashHandlerEi (libKF5Crash.so.5 + 0x788f)
                #3  0x000074ddaa65c710 n/a (libc.so.6 + 0x3e710)
                #4  0x000074ddaa6ac83c n/a (libc.so.6 + 0x8e83c)
                #5  0x000074ddaa65c668 raise (libc.so.6 + 0x3e668)
                #6  0x000074ddaa65c710 n/a (libc.so.6 + 0x3e710)
                #7  0x000074ddaa720f6f __poll (libc.so.6 + 0x102f6f)
                #8  0x000074dda976c2b6 n/a (libglib-2.0.so.0 + 0xb82b6)
                #9  0x000074dda970c162 g_main_context_iteration (libglib-2.0.so.0 + 0x58162)
                #10 0x000074ddaaeead2f _ZN20QEventDispatcherGlib13processEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE (libQt5Core.so.5 + 0x2ead2f)
                #11 0x000074ddaae9ac04 _ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE (libQt5Core.so.5 + 0x29ac04)
                #12 0x000074ddaacf7576 _ZN7QThread4execEv (libQt5Core.so.5 + 0xf7576)
                #13 0x000074ddab197a9a n/a (libQt5DBus.so.5 + 0x18a9a)
                #14 0x000074ddaacf379a n/a (libQt5Core.so.5 + 0xf379a)
                #15 0x000074ddaa6aa9eb n/a (libc.so.6 + 0x8c9eb)
                #16 0x000074ddaa72e7cc n/a (libc.so.6 + 0x1107cc)
                
                Stack trace of thread 9499:
                #0  0x000074ddaa72c73d syscall (libc.so.6 + 0x10e73d)
                #1  0x000074dda97672f7 g_cond_wait (libglib-2.0.so.0 + 0xb32f7)
                #2  0x000074dda96d91b4 n/a (libglib-2.0.so.0 + 0x251b4)
                #3  0x000074dda9741a8e n/a (libglib-2.0.so.0 + 0x8da8e)
                #4  0x000074dda973fa05 n/a (libglib-2.0.so.0 + 0x8ba05)
                #5  0x000074ddaa6aa9eb n/a (libc.so.6 + 0x8c9eb)
                #6  0x000074ddaa72e7cc n/a (libc.so.6 + 0x1107cc)
                
                Stack trace of thread 9501:
                #0  0x000074ddaa720f6f __poll (libc.so.6 + 0x102f6f)
                #1  0x000074dda976c2b6 n/a (libglib-2.0.so.0 + 0xb82b6)
                #2  0x000074dda970eb97 g_main_loop_run (libglib-2.0.so.0 + 0x5ab97)
                #3  0x000074dda563119c n/a (libgio-2.0.so.0 + 0x11219c)
                #4  0x000074dda973fa05 n/a (libglib-2.0.so.0 + 0x8ba05)
                #5  0x000074ddaa6aa9eb n/a (libc.so.6 + 0x8c9eb)
                #6  0x000074ddaa72e7cc n/a (libc.so.6 + 0x1107cc)
                
                Stack trace of thread 9496:
                #0  0x000074ddaa720f6f __poll (libc.so.6 + 0x102f6f)
                #1  0x000074ddabadabca n/a (libKF5Crash.so.5 + 0x6bca)
                #2  0x000074ddabadb82c _ZN6KCrash19defaultCrashHandlerEi (libKF5Crash.so.5 + 0x782c)
                #3  0x000074ddaa65c710 n/a (libc.so.6 + 0x3e710)
                #4  0x000074ddaa6ac83c n/a (libc.so.6 + 0x8e83c)
                #5  0x000074ddaa65c668 raise (libc.so.6 + 0x3e668)
                #6  0x000074ddaa6444b8 abort (libc.so.6 + 0x264b8)
                #7  0x000074ddaa6443dc n/a (libc.so.6 + 0x263dc)
                #8  0x000074ddaa654d26 __assert_fail (libc.so.6 + 0x36d26)
                #9  0x000074dda42fff14 n/a (/usr/lib/libddcutil.so.5.1.0 (deleted) + 0x2bf14)
                ELF object binary architecture: AMD x86-64
```

joeydesimone, 2024-01-24:
Same issue.

**System Info**
---
**Operating System:** Arch Linux
**KDE Plasma Version:** 5.27.10-1
**Powerdevil Version:** 5.27.10-2
**Kernel Version:** 6.7.0-arch3-1 (64-bit)

**Crash Log**
---
```
                      PID: 1222 (org_kde_powerde)
           UID: 1000 (joe)
           GID: 1000 (joe)
        Signal: 6 (ABRT)
     Timestamp: Mon 2024-01-22 18:21:34 EST (1 day 2h ago)
  Command Line: /usr/lib/org_kde_powerdevil
    Executable: /usr/lib/org_kde_powerdevil
 Control Group: /user.slice/user-1000.slice/user@1000.service/background.slice/plasma-powerdevil.service
          Unit: user@1000.service
     User Unit: plasma-powerdevil.service
         Slice: user-1000.slice
     Owner UID: 1000 (joe)
       Boot ID: cb2fedfaed4c4fc48f9ff7cb3856ab07
    Machine ID: 3fb99bccbe504df58da6e09802dc221b
      Hostname: jdt
       Storage: /var/lib/systemd/coredump/core.org_kde_powerde.1000.cb2fedfaed4c4fc48f9ff7cb3856ab07.1222.1705965694000000.zst (inaccessible)
       Message: Process 1222 (org_kde_powerde) of user 1000 dumped core.
                
                Stack trace of thread 1274:
                #0  0x0000776c3feac83c n/a (libc.so.6 + 0x8e83c)
                #1  0x0000776c3fe5c668 raise (libc.so.6 + 0x3e668)
                #2  0x0000776c4130588f _ZN6KCrash19defaultCrashHandlerEi (libKF5Crash.so.5 + 0x788f)
                #3  0x0000776c3fe5c710 n/a (libc.so.6 + 0x3e710)
                #4  0x0000776c3feac83c n/a (libc.so.6 + 0x8e83c)
                #5  0x0000776c3fe5c668 raise (libc.so.6 + 0x3e668)
                #6  0x0000776c3fe444b8 abort (libc.so.6 + 0x264b8)
                #7  0x0000776c3fe443dc n/a (libc.so.6 + 0x263dc)
                #8  0x0000776c3fe54d26 __assert_fail (libc.so.6 + 0x36d26)
                #9  0x0000776c39b49f14 n/a (libddcutil.so.5 + 0x2bf14)
                #10 0x0000776c39b4ac85 n/a (libddcutil.so.5 + 0x2cc85)
                #11 0x0000776c39b54f51 n/a (libddcutil.so.5 + 0x36f51)
                #12 0x0000776c39b55009 n/a (libddcutil.so.5 + 0x37009)
                #13 0x0000776c3ef3fa05 n/a (libglib-2.0.so.0 + 0x8ba05)
                #14 0x0000776c3feaa9eb n/a (libc.so.6 + 0x8c9eb)
                #15 0x0000776c3ff2e7cc n/a (libc.so.6 + 0x1107cc)
                
                Stack trace of thread 1238:
                #0  0x0000776c3ff20f6f __poll (libc.so.6 + 0x102f6f)
                #1  0x0000776c3ef6c2b6 n/a (libglib-2.0.so.0 + 0xb82b6)
                #2  0x0000776c3ef0eb97 g_main_loop_run (libglib-2.0.so.0 + 0x5ab97)
                #3  0x0000776c3aebe19c n/a (libgio-2.0.so.0 + 0x11219c)
                #4  0x0000776c3ef3fa05 n/a (libglib-2.0.so.0 + 0x8ba05)
                #5  0x0000776c3feaa9eb n/a (libc.so.6 + 0x8c9eb)
                #6  0x0000776c3ff2e7cc n/a (libc.so.6 + 0x1107cc)
                
                Stack trace of thread 1236:
                #0  0x0000776c3ff2c73d syscall (libc.so.6 + 0x10e73d)
                #1  0x0000776c3ef672f7 g_cond_wait (libglib-2.0.so.0 + 0xb32f7)
                #2  0x0000776c3eed91b4 n/a (libglib-2.0.so.0 + 0x251b4)
                #3  0x0000776c3ef41a8e n/a (libglib-2.0.so.0 + 0x8da8e)
                #4  0x0000776c3ef3fa05 n/a (libglib-2.0.so.0 + 0x8ba05)
                #5  0x0000776c3feaa9eb n/a (libc.so.6 + 0x8c9eb)
                #6  0x0000776c3ff2e7cc n/a (libc.so.6 + 0x1107cc)
                
                Stack trace of thread 1222:
                #0  0x0000776c3fea74ae n/a (libc.so.6 + 0x894ae)
                #1  0x0000776c3feac5f3 n/a (libc.so.6 + 0x8e5f3)
                #2  0x0000776c3ef40b7d g_thread_join (libglib-2.0.so.0 + 0x8cb7d)
[... 175 lines cut ...]
                #2  0x00007b171e8d91b4 n/a (libglib-2.0.so.0 + 0x251b4)
                #3  0x00007b171e941a8e n/a (libglib-2.0.so.0 + 0x8da8e)
                #4  0x00007b171e93fa05 n/a (libglib-2.0.so.0 + 0x8ba05)
                #5  0x00007b171f8aa9eb n/a (libc.so.6 + 0x8c9eb)
                #6  0x00007b171f92e7cc n/a (libc.so.6 + 0x1107cc)
                
                Stack trace of thread 1280:
                #0  0x00007b171f920f6f __poll (libc.so.6 + 0x102f6f)
                #1  0x00007b1720cc1bca n/a (libKF5Crash.so.5 + 0x6bca)
                #2  0x00007b1720cc282c _ZN6KCrash19defaultCrashHandlerEi (libKF5Crash.so.5 + 0x782c)
                #3  0x00007b171f85c710 n/a (libc.so.6 + 0x3e710)
                #4  0x00007b171f8ac83c n/a (libc.so.6 + 0x8e83c)
                #5  0x00007b171f85c668 raise (libc.so.6 + 0x3e668)
                #6  0x00007b171f8444b8 abort (libc.so.6 + 0x264b8)
                #7  0x00007b171f8443dc n/a (libc.so.6 + 0x263dc)
                #8  0x00007b171f854d26 __assert_fail (libc.so.6 + 0x36d26)
                #9  0x00007b1719549f14 n/a (libddcutil.so.5 + 0x2bf14)
                #10 0x00007b171954ac85 n/a (libddcutil.so.5 + 0x2cc85)
                #11 0x00007b1719554f51 n/a (libddcutil.so.5 + 0x36f51)
                #12 0x00007b1719555009 n/a (libddcutil.so.5 + 0x37009)
                #13 0x00007b171e93fa05 n/a (libglib-2.0.so.0 + 0x8ba05)
                #14 0x00007b171f8aa9eb n/a (libc.so.6 + 0x8c9eb)
                #15 0x00007b171f92e7cc n/a (libc.so.6 + 0x1107cc)
                ELF object binary architecture: AMD x86-64
```

rockowitz, 2024-01-24:
The bug has been addressed by commit 7f157f6 in branch 2.1.1-dev.  The [arch source](https://gitlab.archlinux.org/archlinux/packaging/packages/ddcutil/-/commit/fbb0b3aed1af1cf4a79b214dcab9bc3546b6c564) has been patched.  For the issue thread that led to the diagnosis and resolution, see issue #365.

closed 2024-01-24 by rockowitz