rockowitz/ddcutil #371: Crash due to invalid assertion

kreudom, 2024-01-24:
The crash occurs during initialization, here:

https://github.com/rockowitz/ddcutil/blob/5f9d97f6d530dfe1616b8983077092acb7a3981f/src/ddc/ddc_displays.c#L424-L427

The assertion fails, because the following branch is taken, which does not set any of the 4 flags:

https://github.com/rockowitz/ddcutil/blob/5f9d97f6d530dfe1616b8983077092acb7a3981f/src/ddc/ddc_displays.c#L416-L420

This issues causes KDE powerdevil to crash for me on Arch Linux.

rockowitz, 2024-01-24:
The bug has been addressed by commit 7f157f6 in branch 2.1.1-dev.  The [arch source](https://gitlab.archlinux.org/archlinux/packaging/packages/ddcutil/-/commit/fbb0b3aed1af1cf4a79b214dcab9bc3546b6c564) has been patched.  For the issue thread that led to the diagnosis and resolution, see issue #365.

The SIGABRT failure in command INTERROGATE due to an incorrect assert() will be addressed in an upcoming commit.

rockowitz, 2024-01-24:
The SIGABRT failure in command INTERROGATE is addressed in commit 8fd63cfd3f4378aff737be9f4402f0435dd1e367

closed 2024-01-24 by rockowitz

kreudom, 2024-01-24:
Just to clarify, this was a separate issue from #365 that still caused powerdevil to crash after the fix. But your commit fixed that. Thanks!

rockowitz, 2024-01-24:
@kreudom Yes, you're quite right.  Now that it's working, can you submit the **interrogate** output?  I'm curious about your monitor that seems to be using Linux error EIO to report an unsupported feature. Thanks.

kreudom, 2024-01-24:
Here is the output: [ddcutil-interrogate.txt](https://github.com/rockowitz/ddcutil/files/14038833/ddcutil-interrogate.txt)

Let me know if you need anything else or want me to debug something.