rockowitz/ddcutil #381: VG248 : Invalid display

ThibF, 2024-02-13:
I have two screens connected to my desktop. Both are displaying and working correctly.
One is fully working including DDC with it.
One is detected but `DDC communication failed`.
I verified that DDC is enabled on the screen menu.
Attached the output of `interrogate`.
[ddcutil.log](https://github.com/rockowitz/ddcutil/files/14258284/ddcutil.log)

rockowitz, 2024-02-13:
It's not clear if invalid data or no data is being returned from the monitor.  Option ***--ddc*** should give more information.

Using a large sleep-multiplier value is the first thing to try.  It might help to upgrade to the latest 2.1.3 ddcutil release.

ThibF, 2024-02-13:
I added the ddc option , see attached log.
[ddcutil.log](https://github.com/rockowitz/ddcutil/files/14271956/ddcutil.log)

Right now, glib-2.0 is required to be >= 2.40 to build, which isn't distributed in Ubuntu 24.04 (currently 2.38). So it won't be trivial to test the latest version, will check see how I can test that.

rockowitz, 2024-02-13:
I think you're confusing glib with glibc.  glib 2.40 was released in 2014. glib 2.40  glibc is currently  under development.

rockowitz, 2024-02-13:
Try running **ddcutil getvcp 10 --ddc --bus 5** and **ddcutil getvcp 10 --ddc --bus 6".  I expect you'll lots of garbage response packets for the former.

ThibF, 2024-02-14:
```
➜  ~ ddcutil getvcp 10 --ddc --bus 6
VCP code 0x10 (Brightness                    ): current value =    10, max value =   100
➜  ~ ddcutil getvcp 10 --ddc --bus 5
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC communication failed for monitor on bus /dev/i2c-5
```

ThibF, 2024-02-14:
After figuring out my mistake about glib and building the latest version, here is more logs about the issue.
Surprisingly, sleep multiplier make ddcutil crash at exit time in a reproducible manner.

```
➜  ddcutil git:(master)  ./src/ddcutil getvcp 10 --ddc --bus 5 --sleep-multiplier 4
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 88 88 02 00 00 10 10 00 00 00 00 64 00 1a 1a da 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Actual checksum 0x00, expected 0x3c
DDC: i2c_response_bytes: 6e 88 88 02 00 00 10 10 00 00 00 64 00 00 1a 1a da 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
ddcutil: dsa2.c:931: dsa2_record_final: Assertion `rtable->cur_step <= rtable->cur_retry_loop_step' failed.
[1]    294820 IOT instruction (core dumped)  ./src/ddcutil getvcp 10 --ddc --bus 5 --sleep-multiplier 4
➜  ddcutil git:(master) ./src/ddcutil getvcp 10 --ddc --bus 5                     
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Quirk: response packet starts with double 0x6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC communication failed for monitor on bus /dev/i2c-5
```

[ddcutil.log](https://github.com/rockowitz/ddcutil/files/14278582/ddcutil.log)

rockowitz, 2024-02-14:
Let's deal with the crash first.  Add tracing to the command that triggers the crash: 

.**ddcutil getvcp 10 --ddc --bus 5 --sleep-multiplier 4 --trcfile dsa2.c**

Setting the sleep multiplier without disabling dynamic sleep adjustment just discards cached dsa data and starts with the specified sleep multiplier.  I very much doubt it changes anything, but try: 

**ddcutil getvcp 10 --bus 5  --ddc --sleep_multiplier 4 --disable-dynamic-sleep**

Drive i2c_dev actually has 2 interfaces, one using ioctl() and one using file write()/read().  Normally ddcutil uses the ioctl() interface.  To use the file io interface: 

**ddcutil getvcp 10 --bus 5 --ddc --sleep-multiplier 4 --disable-dynamic-sleep --use-file-io**

Please submit the output of these 3 commands.  (Note that tracing is enabled only on the first.) There's no need to continue to send output from **ddcutil interrogate**.

ThibF, 2024-02-26:
Here is all the logs for the different commands:
[trcfile.381.log](https://github.com/rockowitz/ddcutil/files/14402823/trcfile.381.log)
[disable_dynamic_sleep.381.log](https://github.com/rockowitz/ddcutil/files/14402824/disable_dynamic_sleep.381.log)
[use_file_io.381.log](https://github.com/rockowitz/ddcutil/files/14402825/use_file_io.381.log)

ThibF, 2024-06-10:
If you have time or any hint where I could read about DDC to investigate myself, that would be great.