rockowitz/ddcutil #382: AOC I2475PXQU: Invalid display

setaur, 2024-02-15:
Hello.
I'm trying to use ddcutil to control external monitor AOC I2475PXQU connected via hdmi cable into laptop Lenovo z510 with Debian 11 Bullseye onboard, but I'm hitting into a wall.
I have installed from Debian repository ddcutil version 0.9.9 and also compiled and installed from this site ddcutil 2.1.3. Both failed to control the monitor. Function DDC/CI is enabled in a AOC settings.
Manual for this monitor can be accessed here:
https://manuals.coolblue.nl/eb/aoc-bundel-collectie.pdf

ddcutil 0.9.9:
```
# ddcutil detect --verbose
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Performance and Retry Options:
   Deferred sleep enabled:                      false
   Sleep suppression (reduced sleeps) enabled:  false
   Dynamic sleep adjustment enabled:            false

Invalid display
   I2C bus:             /dev/i2c-5
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      false
      I2C address 0x37 (DDC) responsive:  true 
      /sys/bus/i2c/devices/i2c-5/name:   i915 gmbus dpb
   EDID synopsis:
      Mfg id:           AOC
      Model:            2475W1
      Serial number:    P08H4QA007082
      Manufacture year: 2017
      EDID version:     1.3
      Product code:     9333
      Binary sn:        16843009 (0x01010101)
      Extra descriptor: 
      Video input definition: 0x80 - Digital Input
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.639, 0.338
      Green x,y:        0.297, 0.616
      Blue  x,y:        0.151, 0.053
      Extension blocks: 1
   EDID source: 
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 05 e3 75 24 01 01 01 01   ..........u$....
      +0010   10 1b 01 03 80 35 1e 78 2a a3 e5 a3 56 4c 9d 26   .....5.x*...VL.&
      +0020   0d 50 54 bf ef 00 d1 c0 b3 00 95 00 81 80 81 40   .PT............@
      +0030   81 c0 01 01 01 01 02 3a 80 18 71 38 2d 40 58 2c   .......:..q8-@X,
      +0040   45 00 0f 28 21 00 00 1e 00 00 00 ff 00 50 30 38   E..(!........P08
      +0050   48 34 51 41 30 30 37 30 38 32 00 00 00 fd 00 32   H4QA007082.....2
      +0060   4c 0f 53 11 00 0a 20 20 20 20 20 20 00 00 00 fc   L.S...      ....
      +0070   00 32 34 37 35 57 31 0a 20 20 20 20 20 20 01 16   .2475W1.      ..
   DDC communication failed
   Is DDC/CI enabled in the monitor's on-screen display?

Invalid display
   I2C bus:             /dev/i2c-7
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      true 
      /sys/bus/i2c/devices/i2c-7/name:   AUX A/port A
   EDID synopsis:
      Mfg id:           CMN
      Model:            
      Serial number:    
[... 102 lines cut ...]
      Video input definition:    0x95 - Digital Input (DisplayPort), Bit depth: 6
      Supported features:
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.585, 0.354
      Green x,y:        0.338, 0.575
      Blue  x,y:        0.152, 0.118
      Extension blocks: 0
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 0d ae bd 15 00 00 00 00   ................
      +0010   11 16 01 04 95 22 13 78 02 e9 15 95 5a 56 93 27   .....".x....ZV.'
      +0020   1e 50 54 00 00 00 01 01 01 01 01 01 01 01 01 01   .PT.............
      +0030   01 01 01 01 01 01 79 1d 56 c2 50 00 26 30 1f 41   ......y.V.P.&0.A
      +0040   4c 00 58 c2 10 00 00 18 00 00 00 fe 00 4e 31 35   L.X..........N15
      +0050   36 42 47 45 2d 45 41 31 0a 20 00 00 00 fe 00 43   6BGE-EA1. .....C
      +0060   4d 4e 0a 20 20 20 20 20 20 20 20 20 00 00 00 fe   MN.         ....
      +0070   00 4e 31 35 36 42 47 45 2d 45 41 31 0a 20 00 6a   .N156BGE-EA1. .j
   This is a laptop display.  Laptop displays do not support DDC/CI
```
[ddcutil_2.1.3-interrogate.txt](https://github.com/rockowitz/ddcutil/files/14299383/ddcutil_2.1.3-interrogate.txt)

Can anything be done to make this function work?

rockowitz, 2024-02-16:
The monitor is returning reply packets that are entirely zero, or more precisely the video adapter is reading zeros from the I2C bus after it puts a request packet on the bus.

First, option ***--ddc*** will report DDC data errors.

I'm not optimistic, but some things to try. 

- Turn off dynamic sleep adjustment and force a high sleep multiplier: ***--disable-dynamic-sleep  --sleep-multiplier 3***
- If I'm reading the laptop and monitor specs correctly, in addition to HDMI they both have VGA connectors. I'm sure you don't want to normally use a VGA connection, but it would be informative if the problem persists when using that connection.

setaur, 2024-02-16:
Thanks for quick answer.
Yes, both laptop and monitor have VGA, but unfortunately I don't have cable at this time. I will try, when I get one.
I think all those commands failed to produce anything useful:
```
/usr/local/bin# ./ddcutil --ddc detect
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
Invalid display
   I2C bus:  /dev/i2c-5
   DRM connector:           card0-HDMI-A-1
   EDID synopsis:
      Mfg id:               AOC - UNK
      Model:                2475W1
      Product code:         9333  (0x2475)
      Serial number:        P08H4QA007082
      Binary serial number: 16843009 (0x01010101)
      Manufacture year:     2017,  Week: 16
   DDC communication failed. (getvcp of feature x10 returned Error_Info[DDCRC_RETRIES in ddc_write_read_with_retry, causes: DDCRC_READ_ALL_ZERO(10)])

Invalid display
   I2C bus:  /dev/i2c-7
   DRM connector:           card0-eDP-1
   EDID synopsis:
      Mfg id:               CMN - Chimei Innolux Corporation
      Model:                
      Product code:         5565  (0x15bd)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2012,  Week: 17
   This is a laptop display.  Laptop displays do not support DDC/CI
```

```
/usr/local/bin# ./ddcutil --disable-dynamic-sleep --sleep-multiplier 3 detect
Invalid display
   I2C bus:  /dev/i2c-5
   DRM connector:           card0-HDMI-A-1
   EDID synopsis:
      Mfg id:               AOC - UNK
      Model:                2475W1
      Product code:         9333  (0x2475)
      Serial number:        P08H4QA007082
      Binary serial number: 16843009 (0x01010101)
      Manufacture year:     2017,  Week: 16
   DDC communication failed. (getvcp of feature x10 returned Error_Info[DDCRC_RETRIES in ddc_write_read_with_retry, causes: DDCRC_READ_ALL_ZERO(10)])

Invalid display
   I2C bus:  /dev/i2c-7
   DRM connector:           card0-eDP-1
   EDID synopsis:
      Mfg id:               CMN - Chimei Innolux Corporation
      Model:                
      Product code:         5565  (0x15bd)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2012,  Week: 17
   This is a laptop display.  Laptop displays do not support DDC/CI
```

```
/usr/local/bin# ./ddcutil --use-file-io detect
Invalid display
   I2C bus:  /dev/i2c-5
   DRM connector:           card0-HDMI-A-1
   EDID synopsis:
      Mfg id:               AOC - UNK
      Model:                2475W1
      Product code:         9333  (0x2475)
      Serial number:        P08H4QA007082
      Binary serial number: 16843009 (0x01010101)
      Manufacture year:     2017,  Week: 16
   DDC communication failed. (getvcp of feature x10 returned Error_Info[DDCRC_RETRIES in ddc_write_read_with_retry, causes: DDCRC_READ_ALL_ZERO(10)])

Invalid display
   I2C bus:  /dev/i2c-7
   DRM connector:           card0-eDP-1
   EDID synopsis:
      Mfg id:               CMN - Chimei Innolux Corporation
      Model:                
      Product code:         5565  (0x15bd)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2012,  Week: 17
   This is a laptop display.  Laptop displays do not support DDC/CI
```