rockowitz/ddcutil #383: DDC communication failed (docking station + HDMI)

lameventanas, 2024-02-25:
Hello,
I have a pair of Philips PHL24E1N5500 monitors.
I was using them with a Thinkpad X270 and docking station before, connected via HDMI and DP respectively, and ddcutil worked great.
But now I have migrated to a Thinkpad T14 gen 10, and Lenovo Ultradock 4. One monitor via HDMI, the other with DP.
Now ddcutil can't communicate with the HDMI one.
And I also get a weird message about a "phantom display":

```
$ ddcutil detect --verbose
Invalid display
   I2C bus:  /dev/i2c-9
      DRM connector:                         card1-eDP-1
      /sys/class/drm/card1-eDP-1/dpms        Off
      /sys/class/drm/card1-eDP-1/enabled     disabled
      /sys/class/drm/card1-eDP-1/status      connected
      Driver:                                i915
      I2C address 0x50 (EDID) responsive:    true 
      I2C address 0x37 (DDC)  responsive:    false
      Is LVDS or EDP display:                true 
      Is laptop display by EDID:             false
      Is laptop display:                     true 
      /sys/bus/i2c/devices/i2c-9/name        AUX A/DDI A/PHY A
      PCI device path:                       /sys/devices/pci0000:00/0000:00:02.0/drm/card1/card1-eDP-1/i2c-9
   EDID synopsis:
      Mfg id:               BOE - BOE
      Model:                
      Product code:         2526  (0x09de)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2020,  Week: 32
      EDID version:         1.4
      Extra descriptor:        NV140FHM-N4V
      Video input definition:    0xa5 - Digital Input (DisplayPort), Bit depth: 8
      Supported features:
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.587, 0.364
      Green x,y:        0.335, 0.573
      Blue  x,y:        0.163, 0.116
      Extension blocks: 0
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 09 e5 de 09 00 00 00 00   ................
      +0010   20 1e 01 04 a5 1f 11 78 03 5f f5 96 5d 55 92 29    ......x._..]U.)
      +0020   1d 50 54 00 00 00 01 01 01 01 01 01 01 01 01 01   .PT.............
      +0030   01 01 01 01 01 01 85 40 80 2c 71 38 a0 40 30 20   .......@.,q8.@0 
      +0040   36 00 35 ae 10 00 00 1a 00 00 00 fd 00 28 3c 4a   6.5..........(<J
      +0050   4a 11 01 0a 20 20 20 20 20 20 00 00 00 fe 00 42   J...      .....B
      +0060   4f 45 20 43 51 0a 20 20 20 20 20 20 00 00 00 fe   OE CQ.      ....
      +0070   00 4e 56 31 34 30 46 48 4d 2d 4e 34 56 0a 00 a5   .NV140FHM-N4V...
   This is a laptop display.  Laptop displays do not support DDC/CI

Phantom display
   Associated non-phantom display: bus /dev/i2c-16
   I2C bus:  /dev/i2c-12
      DRM connector:                         card1-DP-3
      /sys/class/drm/card1-DP-3/dpms         Off
      /sys/class/drm/card1-DP-3/enabled      disabled
      /sys/class/drm/card1-DP-3/status       disconnected
      Driver:                                i915
      I2C address 0x50 (EDID) responsive:    true 
      I2C address 0x37 (DDC)  responsive:    false
      Is LVDS or EDP display:                false
      Is laptop display by EDID:             false
      Is laptop display:                     false
      /sys/bus/i2c/devices/i2c-12/name       AUX USBC3/DDI TC3/PHY TC3
      PCI device path:                       /sys/devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-3/i2c-12
   EDID synopsis:
      Mfg id:               PHL - Philips Consumer Electronics Company
      Model:                PHL24E1N5500
      Product code:         49772  (0xc26c)
      Serial number:        UK02348024114
      Binary serial number: 24114 (0x00005e32)
      Manufacture year:     2023,  Week: 48
      EDID version:         1.4
      Extra descriptor:        
      Video input definition:    0xa5 - Digital Input (DisplayPort), Bit depth: 8
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:4:4 + YCrCb 4:2:2
         Standard sRGB color space: True
      White x,y:        0.313, 0.329
      Red   x,y:        0.655, 0.311
      Green x,y:        0.286, 0.650
      Blue  x,y:        0.151, 0.061
      Extension blocks: 1
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 41 0c 6c c2 32 5e 00 00   ........A.l.2^..
      +0010   30 21 01 04 a5 35 1e 78 3b e6 e5 a7 4f 49 a6 26   0!...5.x;...OI.&
      +0020   0f 50 54 bf ef 00 d1 c0 b3 00 95 00 81 80 81 40   .PT............@
      +0030   81 c0 01 01 01 01 56 5e 00 a0 a0 a0 29 50 30 20   ......V^....)P0 
      +0040   35 00 0f 28 21 00 00 1e 00 00 00 ff 00 55 4b 30   5..(!........UK0
      +0050   32 33 34 38 30 32 34 31 31 34 00 00 00 fc 00 50   2348024114.....P
      +0060   48 4c 32 34 45 31 4e 35 35 30 30 0a 00 00 00 fd   HL24E1N5500.....
      +0070   00 30 4b 72 72 21 01 0a 20 20 20 20 20 20 01 e8   .0Krr!..      ..
   DDC communication failed
   Use non-phantom device bus /dev/i2c-16

Display 1
   I2C bus:  /dev/i2c-16
      DRM connector:                         card1-DP-8
      /sys/class/drm/card1-DP-8/dpms         On
      /sys/class/drm/card1-DP-8/enabled      enabled
      /sys/class/drm/card1-DP-8/status       connected
      Driver:                                i915
      I2C address 0x50 (EDID) responsive:    true 
      I2C address 0x37 (DDC)  responsive:    true 
      Is LVDS or EDP display:                false
      Is laptop display by EDID:             false
      Is laptop display:                     false
      /sys/bus/i2c/devices/i2c-16/name       DPMST
      PCI device path:                       /sys/devices/pci0000:00/0000:00:02.0/i2c-16
   EDID synopsis:
      Mfg id:               PHL - Philips Consumer Electronics Company
      Model:                PHL24E1N5500
      Product code:         49772  (0xc26c)
      Serial number:        UK02348024114
      Binary serial number: 24114 (0x00005e32)
      Manufacture year:     2023,  Week: 48
      EDID version:         1.4
      Extra descriptor:        
      Video input definition:    0xa5 - Digital Input (DisplayPort), Bit depth: 8
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:4:4 + YCrCb 4:2:2
         Standard sRGB color space: True
      White x,y:        0.313, 0.329
      Red   x,y:        0.655, 0.311
      Green x,y:        0.286, 0.650
      Blue  x,y:        0.151, 0.061
      Extension blocks: 1
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 41 0c 6c c2 32 5e 00 00   ........A.l.2^..
      +0010   30 21 01 04 a5 35 1e 78 3b e6 e5 a7 4f 49 a6 26   0!...5.x;...OI.&
      +0020   0f 50 54 bf ef 00 d1 c0 b3 00 95 00 81 80 81 40   .PT............@
      +0030   81 c0 01 01 01 01 56 5e 00 a0 a0 a0 29 50 30 20   ......V^....)P0 
      +0040   35 00 0f 28 21 00 00 1e 00 00 00 ff 00 55 4b 30   5..(!........UK0
      +0050   32 33 34 38 30 32 34 31 31 34 00 00 00 fc 00 50   2348024114.....P
      +0060   48 4c 32 34 45 31 4e 35 35 30 30 0a 00 00 00 fd   HL24E1N5500.....
      +0070   00 30 4b 72 72 21 01 0a 20 20 20 20 20 20 01 e8   .0Krr!..      ..
   VCP version:         2.2
   Controller mfg:      RealTek
   Firmware version:    0.1
   Monitor uses invalid feature flag in DDC reply packet to indicate unsupported feature.
   Current dynamic sleep adjustment multiplier:  0.00
```

rockowitz, 2024-02-29:
I'm pretty sure that what has changed is that the video connection for the X270 dock is standard DisplayPort, whereas the new T14/Lenovo dock combination uses USB-C alt mode to transmit the DisplayPort signal, and the I<sup>2</sup>C signalling is being lost. 

Note that when a docking station is used the connection to the dock is always DP Multi-Stream Transport (MST).  If there is a HDMI connector on the dock the DP signal is converted to HDMI.

"Phantom display" is a way of describing the phenomenon of a display appearing at two different /dev/i2c devices.  Typically only one actually supports DDC.  **ddcutil** reports the invalid /dev/i2c device as "phantom".  

See [Docking Connected Displays](https://www.ddcutil.com/release_notes/#docking-station-connected-displays), [The same DisplayPort connected monitor appears twice in the output of ddcutil detect.](https://www.ddcutil.com/faq/#the-same-displayport-connected-monitor-appears-twice-in-the-output-of-ddcutil-detect) and also this extensive [freedesktop.org bug report thread](https://gitlab.freedesktop.org/drm/intel/-/issues/37).

This warrants a [i915 bug report] ( https://drm.pages.freedesktop.org/intel-docs/how-to-file-i915-bugs.html).  Let me know when you've filed it and I'll add myself to the CC list.  But I wouldn't expect a fix any time soon.

lameventanas, 2024-03-01:
> This warrants a [i915 bug report] ( https://drm.pages.freedesktop.org/intel-docs/how-to-file-i915-bugs.html). Let me know when you've filed it and I'll add myself to the CC list. But I wouldn't expect a fix any time soon.

I don't get it, what is the i915 bug that I should file? Is the communication between ddcutil and the HDMI monitor failing due to a bug in the i915 driver?

rockowitz, 2024-03-01:
In your new setup, the T14 <-> Docking Station communication is by a USB Type-C connector, operating in display port alt mode. I<sup>2</sup>C signalling is handled over the DP Aux channel.  The dock is a DP Multi-Stream Transport device.  The single DP stream from laptop to dock actually carries several DP streams, which are split out at the dock.  For the HDMI connector on the dock, the dock converts  DP signalling to HDMI signalling.  In particular, the I<sup>2</sup>C signal that is "multiplexed" over the the DP Aux channel becomes simple I<sup>2</sup>C signalling on the appropriate pins of the HDMI connector.  Somewhere along this chain the I<sup>2</sup>C signal is being lost.

lameventanas, 2024-03-03:
Thanks for the explanation, but I still don't know what would I file in such a bug report to the i915 driver developers.
And they will just brush it off and say that the dock is defective or that the dock doesn't implement that function.

lameventanas, 2024-03-04:
The docking station model is a Lenovo Universal Thunderbolt 4 (40b0) (I just updated the issue to reflect this).
I tried updating the firmware to the latest version, it didn't help.
I also tried connecting both monitors via DP, but still only one of them is detected.
I'm about to buy a 3rd monitor, but I'm wondering if I should get one that supports USB TB instead of just DP or HDMI.
My aim is to be able to control 3 monitors, so I can easily change the brightness in a script, instead of using their menu system one by one, with the controls in the back. Is there any hope for this?

Mousketeer, 2024-10-20:
> In your new setup, the T14 <-> Docking Station communication is by a USB Type-C connector, operating in display port alt mode. I2C signalling is handled over the DP Aux channel.
Pardon my enormous ignorance, but is this still the case even if he's connected with thunderbolt?

I'm largely in the same boat as @lameventanas: the same Lenovo thunderbolt docking station, using two identical models of monitors (also Philips, but a different model) from a TB4 port (not on a Lenovo) and I have the exact same symptoms.

I'll attempt to borrow a different TB-enabled laptop later, but I'm not sure if I can get one without i915.

@rockowitz Would you say this warrants a bugreport to the developers of the firmware of the dock? (Not sure how to approach that, but..)

Mousketeer, 2024-10-20:
After re-reading some of @rockowitz 's comments, I checked my hardware setup, and I'm actually using both monitors over displayport, so that's a significant difference.

For completeness, using the ddcutil 2.1.4, kernel 6.1.0 (Debian) and the very latest dock firmware 1.1.18:
`ddcutil detect --verbose
Invalid display
   I2C bus:  /dev/i2c-13
      DRM connector:                         card0-eDP-1
      /sys/class/drm/card0-eDP-1/dpms        On
      /sys/class/drm/card0-eDP-1/enabled     enabled
      /sys/class/drm/card0-eDP-1/status      connected
      Driver:                                i915
      I2C address 0x50 (EDID) responsive:    true 
      I2C address 0x37 (DDC)  responsive:    false
      Is LVDS or EDP display:                true 
      Is laptop display by EDID:             false
      Is laptop display:                     true 
      /sys/bus/i2c/devices/i2c-13/name       AUX A/DDI A/PHY A
      PCI device path:                       /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/i2c-13
   EDID synopsis:
      Mfg id:               AUO - UNK
      Model:                
      Product code:         25149  (0x623d)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2018,  Week: 0
      EDID version:         1.4
      Extra descriptor:        B140HAN06.2
      Video input definition:    0xa5 - Digital Input (DisplayPort), Bit depth: 8
      Supported features:
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.640, 0.330
      Green x,y:        0.300, 0.600
      Blue  x,y:        0.150, 0.060
      Extension blocks: 0
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 06 af 3d 62 00 00 00 00   ..........=b....
      +0010   00 1c 01 04 a5 1f 11 78 02 ee 95 a3 54 4c 99 26   .......x....TL.&
      +0020   0f 50 54 00 00 00 01 01 01 01 01 01 01 01 01 01   .PT.............
      +0030   01 01 01 01 01 01 14 37 80 b8 70 38 24 40 10 10   .......7..p8$@..
      +0040   3e 00 35 ae 10 00 00 18 00 00 00 0f 00 00 00 00   >.5.............
      +0050   00 00 00 00 00 00 00 00 00 20 00 00 00 fe 00 41   ......... .....A
      +0060   55 4f 0a 20 20 20 20 20 20 20 20 20 00 00 00 fe   UO.         ....
      +0070   00 42 31 34 30 48 41 4e 30 36 2e 32 20 0a 00 2a   .B140HAN06.2 ..*
   This is a laptop display.  Laptop displays do not support DDC/CI

Phantom display
   Associated non-phantom display: bus /dev/i2c-19
   I2C bus:  /dev/i2c-14
      DRM connector:                         card0-DP-1
      /sys/class/drm/card0-DP-1/dpms         Off
      /sys/class/drm/card0-DP-1/enabled      disabled
      /sys/class/drm/card0-DP-1/status       disconnected
      Driver:                                i915
      I2C address 0x50 (EDID) responsive:    true 
      I2C address 0x37 (DDC)  responsive:    false
      Is LVDS or EDP display:                false
      Is laptop display by EDID:             false
      Is laptop display:                     false
      /sys/bus/i2c/devices/i2c-14/name       AUX USBC1/DDI TC1/PHY TC1
      PCI device path:                       /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-1/i2c-14
   EDID synopsis:
      Mfg id:               PHL - Philips Consumer Electronics Company
      Model:                PHL 275B1
      Product code:         2374  (0x0946)
      Serial number:        UK02049051504
      Binary serial number: 51504 (0x0000c930)
[... 52 lines cut ...]
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:4:4 + YCrCb 4:2:2
         Standard sRGB color space: True
      White x,y:        0.313, 0.329
      Red   x,y:        0.667, 0.313
      Green x,y:        0.290, 0.655
      Blue  x,y:        0.152, 0.047
      Extension blocks: 1
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 41 0c 46 09 30 c9 00 00   ........A.F.0...
      +0010   31 1e 01 04 a5 3c 22 78 3b d7 05 aa 50 4a a7 27   1....<"x;...PJ.'
      +0020   0c 50 54 bf ef 00 d1 c0 b3 00 95 00 81 80 81 40   .PT............@
      +0030   81 c0 01 01 01 01 56 5e 00 a0 a0 a0 29 50 30 20   ......V^....)P0 
      +0040   35 00 55 50 21 00 00 1e 00 00 00 ff 00 55 4b 30   5.UP!........UK0
      +0050   32 30 34 39 30 35 31 35 30 34 00 00 00 fc 00 50   2049051504.....P
      +0060   48 4c 20 32 37 35 42 31 0a 20 20 20 00 00 00 fd   HL 275B1.   ....
      +0070   00 30 4b 72 72 1e 01 0a 20 20 20 20 20 20 01 37   .0Krr...      .7
   VCP version:         2.2
   Controller mfg:      RealTek
   Firmware version:    0.1
   Monitor uses invalid feature flag in DDC reply packet to indicate unsupported feature.
   Current dynamic sleep adjustment multiplier:  0.00
`

lameventanas, 2024-10-23:
> This warrants a [i915 bug report] ( https://drm.pages.freedesktop.org/intel-docs/how-to-file-i915-bugs.html). Let me know when you've filed it and I'll add myself to the CC list. But I wouldn't expect a fix any time soon.

I created a [bug report in i915](https://gitlab.freedesktop.org/drm/i915/kernel/-/issues/10901), they asked me to try their latest drmtip. After compiling, testing, etc, I uploaded the logs, which they have ignored for 4 months already.