rockowitz/ddcutil #385: AOC Q27P1B input source values

denilsonsa, 2024-02-27:
I'm giving some feedback regarding the [AOC Q27P1B monitor](https://aoc.com/uk/products/monitors/q27p1/specification). It has four video inputs:

* VGA (D-SUB)
* DVI
* HDMI
* DP (DisplayPort)

Unfortunately, its implementation of the VCP 0x60 is… well… unique. And it probably behaves similarly to [other AOC displays](https://github.com/rockowitz/ddcutil/issues?q=AOC).

It claims to support:

    Feature: 60 (Input Source)
       Values:
          01: VGA-1
          03: DVI-1

<details><summary>Full output of capabilities</summary>

```console
$ ddcutil capabilities
Model: Q27P1
MCCS version: 2.1
Commands:
   Op Code: 01 (VCP Request)
   Op Code: 02 (VCP Response)
   Op Code: 03 (VCP Set)
   Op Code: 07 (Timing Request)
   Op Code: 0C (Save Settings)
   Op Code: F3 (Capabilities Request)
VCP Features:
   Feature: 02 (New control value)
   Feature: 04 (Restore factory defaults)
   Feature: 05 (Restore factory brightness/contrast defaults)
   Feature: 08 (Restore color defaults)
   Feature: 10 (Brightness)
   Feature: 12 (Contrast)
   Feature: 14 (Select color preset)
      Values:
         01: sRGB
         05: 6500 K
         06: 7500 K
         08: 9300 K
         0b: User 1
   Feature: 16 (Video gain: Red)
   Feature: 18 (Video gain: Green)
   Feature: 1A (Video gain: Blue)
   Feature: 60 (Input Source)
      Values:
         01: VGA-1
         03: DVI-1
   Feature: 62 (Audio speaker volume)
   Feature: 6C (Video black level: Red)
   Feature: 6E (Video black level: Green)
   Feature: 70 (Video black level: Blue)
   Feature: C8 (Display controller type)
   Feature: C9 (Display firmware level)
   Feature: B0 (Settings)
   Feature: B6 (Display technology type)
   Feature: D6 (Power mode)
   Feature: DF (VCP Version)
   Feature: F8 (Manufacturer specific feature)
```

</details>

But obviously it supports more values for VCP 0x60.

I also know this monitor is not yet added to any DDC database, because I get the *Feature definition file not found: AOC-Q27P1B-9985.mccs* message.

## Limitations from my tests
[... 165 lines cut ...]

</details>

## System

* Hardware
    * HP EliteDesk 800 G2 DM 65W
    * Intel i7-6700
    * Intel HD Graphics 530
    * 2x [AOC Q27P1B](https://aoc.com/uk/products/monitors/q27p1/specification) connected via DisplayPort (direct cables, no adapters)
* Software
    * ddcutil version 2.1.3
    * OS: Manjaro Linux x86_64
    * Kernel: 6.6.16-2-MANJARO

## Action points

Can this kind of info be added to some database somewhere?

Do you need any additional information?

## See also

* #188
* #314

rockowitz, 2024-02-29:
There is no central database of monitor feature definitions.  Given the huge and expanding range of monitors, creating and maintaining one would be a hopeless task.  The message "Feature definition file not found: AOC-Q27P1B-9985.mccs" indicates that you have not created a local feature definition file for the monitor. See [User Defined Features](https://www.ddcutil.com/udf/).

I suggest that you add your findings to the wiki page [Notes on Specific Monitors](https://github.com/rockowitz/ddcutil/wiki/Notes-on-Specific-Monitors) or perhaps add a new page for AOC displays.  This is also an appropriate location for people to report feature definition (mccs) file they have created.

digitaltrails, 2024-04-04:
Could this feature be handled by a ddcutil *user defined feature*, would it be CNC?

```
MFG_ID       AOC
MODEL        AOC Q27P1B
PRODUCT_CODE 12345
MCCS_VERSION 2.2
FEATURE_CODE 60  Input selection
    ATTRS NC RW
    VALUE 01 VGA
	VALUE 02 VGA (switch to)
	VALUE 03 DVI (switch to)
	VALUE 04 DVI (switch to)
	VALUE 0f DisplayPort (switch to)
	VALUE 11 HDMI (switch to)
	VALUE 300 DVI/HDMI (read-only)
	VALUE 301 DVI/HDMI (read-only)
	VALUE 302 DVI/HDMI (read-only)
	VALUE 305 DisplayPort (read-only)
```

leinardi, 2026-10-01:
@denilsonsa thanks for this write-up, it's the most careful one I've found for an AOC monitor. I used your values (`0x0F`, `0x11`, `0x03`, `0x01`) in `monmux` (https://github.com/leinardi/monmux), a small tool that calls `ddcutil` to switch inputs, but only for monitors that someone has tested.

Since you have the monitor, would you be up for running it once? It's three commands, and the first two don't change anything: one reads the monitor's ID, the other prints the exact command before you run it. Steps: https://github.com/leinardi/monmux/issues/18. No worries if not.

denilsonsa, 2026-10-01:
@leinardi Poke me again in a few months. I'm [busy enough](https://indieweb.org/life_happens) for now that I can't really give enough attention to test the display.