rockowitz/ddcutil #386: "Invalid display" talking to Philips 436M6VBP over HDMI

l0b0, 2024-02-28:
Not sure what to do about this error:

```console
❯ ddcutil detect
Invalid display
   I2C bus:  /dev/i2c-7
   DRM connector:           card0-HDMI-A-1
   EDID synopsis:
      Mfg id:               PHL - Philips Consumer Electronics Company
      Model:                PHL 436M6VBP
      Product code:         49529  (0xc179)
      Serial number:        
      Binary serial number: 1335 (0x00000537)
      Manufacture year:     2019,  Week: 9
   DDC communication failed. (getvcp of feature x10 returned Error_Info[EIO in ddc_write_read_with_retry, causes: EIO])
```

After looking around at similar issues I've tried both `--disable-dynamic-sleep --sleep-multiplier 3` and `--use-file-io`, but to no effect.

[interrogate.log](https://github.com/rockowitz/ddcutil/files/14431807/interrogate.log)

rnhmjoj, 2024-02-29:
I'm not sure, given it has identified the display I'd say i2c is set up correctly. 
There is a number of similar issues in the upstream tracker, so it's probably unrelated.

rockowitz, 2024-03-01:
I<sup>2</sup>C communication is working on /dev/i2c-7. The EDID (slave address x50) can be read.  Error EIO is an indication that slave address x37 (DDC) is unresponsive.

If you install **i2cdetect** what I expect it will show is that, for i2c bus 7, slave address x50 is responsive, and slave address x37 is not.  So no amount of fiddling with **ddcutil** settings will make things work. 

You have 2 video devices (card0 and card1) enabled.  Apparently one is on the Ryzen CPU. and the other is a discrete card.  Do you need both?  That may be adding complexity that confuses the amdgpu driver.

Have you tested whether I<sup>2</sup>C communication works on X11 instead of Wayland?

Beyond that, I would guess that the problem is in some way related to how NixOS is configured, but I have no experience with NixOS.

l0b0, 2024-03-02:
> There is a number of similar issues in the upstream tracker, so it's probably unrelated.

Which upstream tracker do you mean? i2c, the kernel, the display manufacturer, or something else?

l0b0, 2024-03-02:
> I2C communication is working on /dev/i2c-7. The EDID (slave address x50) can be read. Error EIO is an indication that slave address x37 (DDC) is unresponsive.
> 
> If you install **i2cdetect** what I expect it will show is that, for i2c bus 7, slave address x50 is responsive, and slave address x37 is not. So no amount of fiddling with **ddcutil** settings will make things work.

Oh, OK, that clarifies things.
 
> You have 2 video devices (card0 and card1) enabled. Apparently one is on the Ryzen CPU. and the other is a discrete card. Do you need both? That may be adding complexity that confuses the amdgpu driver.

Disabling the onboard GPU didn't help.

> Have you tested whether I2C communication works on X11 instead of Wayland?

Just did, and the result was the same on X11 as on Wayland.

> Beyond that, I would guess that the problem is in some way related to how NixOS is configured, but I have no experience with NixOS.

I'd expect it to be something wrong with the display configuration, but I don't know enough about this to debug any of it.