rockowitz/ddcutil #388: Cant switch between DP1 and DP2

yosev, 2024-03-02:
Hey,

first of all, thanks for your work on this project!

I have a Samsung Odyssey G7 and i managed to use ddc from windows, mac and linux (raspberry) via HDMI and DP (1&2).
Despite the fact that the values for `ddcutil vcpinfo x60 --verbose` are completely wild, i managed to find the correct values for HDMI1, DP1 and DP2. The weird thing is, that i can not toggle between DP1 and DP2. Whenever i switch to one of the two DP, the displays overlay shows DP1 but does indeed show the output of the machine connected on DP2. Is there something i miss when switching the channels? The display's overlay even marks both as the active input (DP1 and DP2) at the same time, if i navigate to the input source manually.

I have:
- HDMI1 -> 0x06
- DP1 -> 0x03
- DP2 -> 0x09

Output of `ddcutil interrogate` & `ddcutil vcpinfo x60 --verbose`:
https://pastebin.com/eZX4YxNx

Thanks!

yosev, 2024-03-02:
I updated the Monitor to the latest firmware 'M-TG700CCAA-1016.1[9222]' which did **not** bring any changes.

rockowitz, 2024-03-06:
I'm afraid I don't have anything to tell you that you don't already know.  Many recent large monitors with advanced picture-by-picture capabilities  manage the input sources through non-standard use of feature x60 and features in the manufacturer reserved range xE0..xFF, The only hope for a solution is to sniff the I<sup>2</sup>C packets and reverse engineer the protocol.

At least please add what you've found to the [wiki](l/wiki/Notes-on-Specific-Monitors) so that others can benefit from what you've found.

closed 2024-03-06 by yosev