rockowitz/ddcutil #390: libddcutil -3021 "DDCRC_INTERNAL_ERROR"

digitaltrails, 2024-03-07:
I plugged in my old beng monitor, it works fine with ddcutil-service+libddcutil-1.4 and with with ddcutil 2.1.5-dev command.  But errors when I use ddcutil-service with libddcutil-2.1.5-dev it errors:

```
2.1.5-dev v2.1.4 % ./src/ddcutil --display 2 getvcp 10
VCP code 0x10 (Brightness                    ): current value =    78, max value =   100``

2.1.5-dev v2.1.4 % busctl --user call com.ddcutil.DdcutilService /com/ddcutil/DdcutilObject com.ddcutil.DdcutilInterface GetVcp isyu "2" "" "16" "0"
qqsis 0 0 "" -3021 "DDCRC_INTERNAL_ERROR"

2.1.5-dev v2.1.4 % busctl --user get-property com.ddcutil.DdcutilService /com/ddcutil/DdcutilObject com.ddcutil.DdcutilInterface DdcutilVersion   
s "2.1.4"
```

Detect is working fine:
```
busctl --user call com.ddcutil.DdcutilService /com/ddcutil/DdcutilObject com.ddcutil.DdcutilInterface Detect u "0"
ia(iiisssqsu)is 3 3 1 0 0 
"HWP" "HP ZR24w" "CNT00811J6" 10345 "AP///////wAi8GkoAQEBAQgUAQSlNiN4LvyBpFVNnSUSUFQhCACBQIGAlQCpQLMA0cABAQEBKDyAoHCwI0AwIDYAImAhAAAaAAAA/QA7PRhQEQAKICAgICAgAAAA/ABIUCBaUjI0dwogICAgAAAA/wBDTlQwMDgxMUo2CiAgAD0=" 16843009 2 0 0 "BNQ" 
"BenQ T2200HD" "2C803527SL0" 30502 "AP///////wAJ0SZ3RVQAADMSAQOAMBt4LjWBplZImiQSUFSlSwBxT4GA0cABAQEBAQEBAQEBAjqAGHE4LUBYLEUA3QwRAAAeAAAA/wAyQzgwMzUyN1NMMAoKAAAA/QA4TB5TEQAKICAgICAgAAAA/ABCZW5RIFQyMjAwSEQKACo=" 21573 3 0 0 "GSM" 
"LG HDR 4K" "" 30471 "AP///////wAebQd3e1EBAAQdAQS1PCJ4nj4xrlBHrCcMUFQhCABxQIGAgcCpwNHAgQABAQEBTdAAoPBwPoAwIGUMWFQhAAAaKGgAoPBwPoAIkGUMWFQhAAAaAAAA/QA4PR6HOAAKICAgICAgAAAA/ABMRyBIRFIgNEsKICAgAZ4=" 86395 0 "OK"
```

I'll look into this more later, but I thought I let you know ASAP.  It may have been happening with earlier version too, but I had Nvidia issues so I hadn't plugged in this monitor.

rockowitz, 2024-03-07:
Try passing option ***--trcapi ddca_get_non_table_vcp_value*** to libddcutil by putting it config file ddcutilrc.  This should give more detail as to what the internal error is.

digitaltrails, 2024-03-07:
The return code is coming from ddc_validate_display_ref called from ddca_open_display3 via ddca_open_display2.  Still investigating...

digitaltrails, 2024-03-07:
If I disabled drm by adding --f12 to ~/.config/ddcutil/ddcutilrc, then the problem goes away.

Maybe this nvidia 550 driver's DRM is not quite all there.

rockowitz, 2024-03-07:
What's the output of **ddcutil det -v --f2**?

digitaltrails, 2024-03-07:
> What's the output of **ddcutil det -v --f2**?

[ddcutil-det-f2.log](https://github.com/rockowitz/ddcutil/files/14528227/ddcutil-det-f2.log)

digitaltrails, 2024-03-07:
The log I attached might not contain anything useful.  This is beginning to look like some driver weirdness. Yesterday I connected the monitor while the system was up. This morning I cold booted this morning and the issue no longer occurs.    

I'll see if I can cause the problem to reoccur.  

BTW, the beng is connected by DVI-DP, the previous Nvidia driver had issues with DVI-DP, so maybe that's a factor.

digitaltrails, 2024-03-07:
I did a cold boot and then a hot-plug of the Beng, and the fault reoccurred. Log attached.  Diff'ing the two logs shows that DRM doesn't appear to be properly initialized for the hotplugged monitor.

[ddcutil-det-f2-problem.log](https://github.com/rockowitz/ddcutil/files/14529816/ddcutil-det-f2-problem.log)

Exact process.
1. I powered everything down at the wall.  
2. Unplugged the Beng DVI-DP cable.  
3. Waited a while. 
4. Powered back up.
5. Logged in.
6. Started ddcutil-service with ddcutillib 2.1.5-dev
7. Plugged the Beng DVI-DP cable back in.
8. Ran the Nvidia-Settings to enable the Beng.

digitaltrails, 2024-03-07:
It's the driver.  I powered down an disconnected my HP DP-DP connected monitor.  Same story on hotplugging.

```
Display 1
   I2C bus:  /dev/i2c-0
   EDID synopsis:
      Mfg id:               HWP - Hewlett Packard
      Model:                HP ZR24w
      Product code:         10345  (0x2869)
      Serial number:        CNT00811J6
      Binary serial number: 16843009 (0x01010101)
      Manufacture year:     2010,  Week: 8
   VCP version:         2.2

Display 2
   I2C bus:  /dev/i2c-3
   DRM connector:           card0-DP-3
   EDID synopsis:
      Mfg id:               GSM - Goldstar Company Ltd (LG)
      Model:                LG HDR 4K
      Product code:         30471  (0x7707)
      Serial number:        
      Binary serial number: 86395 (0x0001517b)
      Manufacture year:     2019,  Week: 4
   VCP version:         2.1

```

This is not a big issue for me, but it might be for anyone with a laptop that might routinely hotplug a monitor. 

[hp.log](https://github.com/rockowitz/ddcutil/files/14530017/hp.log)

digitaltrails, 2024-03-08:
Posted a query to forums.developer.nvidia.com: [Driver 550.54 - no update to /sys/class/drm/… on hotplug of monitor](https://forums.developer.nvidia.com/t/driver-550-54-no-update-to-sys-class-drm-on-hotplug-of-monitor/285316)

Sometimes these kinds of post get no official response - but often things eventually get fixed. I figure it can't hurt to make some noise.

rockowitz, 2024-03-08:
I've seen this behaviour.   On my deskstop system, I've found that if a monitor is removed the edid can still be there, and if a different monitor is connected the edid value is unchanged.  I backed off on using the sysfs edid value instead of reading the edid directly - I may need to back off even more.

A couple days ago I tried creating some test cases to demonstrate the behaviour for an intended amdgpu drm bug report,
but I was unable to reproduce the problem.

digitaltrails, 2024-03-08:
> I've seen this behaviour. On my deskstop system, I've found that if a monitor is removed the edid can still be there, and if a different monitor is connected the edid value is unchanged. I backed [...]

It appears that DRM is not all there yet. Perhaps it's not yet essential to desktop environment, so the imperfections are not making enough waves.

> A couple days ago I tried creating some test cases to demonstrate the behaviour for an intended amdgpu drm bug report, but I was unable to reproduce the problem.

Generally ddcutil/libddcutil has been pretty solid. Having dodgy drivers undermine that experience is a shame.  I'd prefer to back off to the point where users aren't bothered by these problems.  

Possibly some advanced options should default to off.  Documentation could explain that an option may require specific GPUs and GPU-drivers.  Perhaps even mentioning which GPUs and drivers are currently known to work well.

rockowitz, 2024-03-09:
As a first step, I've added some asserts, which along with valgrind should indicate where the g_path_get_basename() calls are failing and suggest where to look further.  As you can see from the branch 2.1.5 git log I've also improved tracing which should help with the next step.

digitaltrails, 2024-03-09:
So I should test with the latest code? Or is this something you are pursuing yourself?

rockowitz, 2024-03-10:
Yes, please test with the latest code. Until I can reliably recreate the problem here, all I can do is stare at the code.  Thanks.

digitaltrails, 2024-03-10:
Only the 2.1.5-dev branch has any recent commits, but they're not recent as in today.  I'm I not seeing things right?

rockowitz, 2024-03-10:
Doing too many things today .. I just pushed the changes.

digitaltrails, 2024-03-10:
I'm not seeing anything interesting.  `ddc_validate_display_ref()` is still erroring due to a NULL `drm_connector`.  I presume this may be because `/sys/class/drm/card0-DP-1/device/card0-DP-2/edid` is empty for the hotplugged monitor, whereas for all monitors present at boot, that file does contain an edid.

So I'm seeing no assertions firing on a getvcp.  And --trcapi ddca_get_non_table_vcp_value didn't log anything extra.

rockowitz, 2024-03-10:
So if you run the exact same steps that produced  ddcutil-det-f2-problem.log,  you get the same output as before with the  GLib-CRITICAL msgs re g_path_get_basename, but no asserts in libddcutil.  Is that correct?

digitaltrails, 2024-03-10:
> So if you run the exact same steps that produced ddcutil-det-f2-problem.log, you get the same output as before with the GLib-CRITICAL msgs re g_path_get_basename, but no asserts in libddcutil. Is that correct?

Accessing `libddcutil` from `busctl`, I get no asserts, but it occured to me that you were really asking for the result of `./src/ddcutil detect -v --f2`, and that does have an assert failure:
```
% ./src/ddcutil detect -v --f2
*** Sys_Drm_Connector report: Detailed /sys/class/drm report: ***

   Display connectors reported by DRM:
   Connector:   card0-DP-1
      i2c_busno:   -1
      name:        (null)
      dev:         (null)
      enabled:     disabled
      status:      connected
      edid:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 22 f0 69 28 01 01 01 01   ........".i(....
      +0010   08 14 01 04 a5 36 23 78 2e fc 81 a4 55 4d 9d 25   .....6#x....UM.%
      +0020   12 50 54 21 08 00 81 40 81 80 95 00 a9 40 b3 00   .PT!...@.....@..
      +0030   d1 c0 01 01 01 01 28 3c 80 a0 70 b0 23 40 30 20   ......(<..p.#@0 
      +0040   36 00 22 60 21 00 00 1a 00 00 00 fd 00 3b 3d 18   6."`!........;=.
      +0050   50 11 00 0a 20 20 20 20 20 20 00 00 00 fc 00 48   P...      .....H
      +0060   50 20 5a 52 32 34 77 0a 20 20 20 20 00 00 00 ff   P ZR24w.    ....
      +0070   00 43 4e 54 30 30 38 31 31 4a 36 0a 20 20 00 3d   .CNT00811J6.  .=

   Connector:   card0-DP-2
      i2c_busno:   -1
      name:        (null)
      dev:         (null)
      enabled:     disabled
      status:      disconnected
      edid:        None

   Connector:   card0-DP-3
      i2c_busno:   -1
      name:        (null)
      dev:         (null)
      enabled:     disabled
      status:      connected
      edid:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 1e 6d 07 77 7b 51 01 00   .........m.w{Q..
      +0010   04 1d 01 04 b5 3c 22 78 9e 3e 31 ae 50 47 ac 27   .....<"x.>1.PG.'
      +0020   0c 50 54 21 08 00 71 40 81 80 81 c0 a9 c0 d1 c0   .PT!..q@........
      +0030   81 00 01 01 01 01 4d d0 00 a0 f0 70 3e 80 30 20   ......M....p>.0 
      +0040   65 0c 58 54 21 00 00 1a 28 68 00 a0 f0 70 3e 80   e.XT!...(h...p>.
      +0050   08 90 65 0c 58 54 21 00 00 1a 00 00 00 fd 00 38   ..e.XT!........8
      +0060   3d 1e 87 38 00 0a 20 20 20 20 20 20 00 00 00 fc   =..8..      ....
      +0070   00 4c 47 20 48 44 52 20 34 4b 0a 20 20 20 01 9e   .LG HDR 4K.   ..
      +0080   02 03 19 71 44 90 04 03 01 23 09 07 07 83 01 00   ...qD....#......
      +0090   00 e3 05 c0 00 e3 06 05 01 02 3a 80 18 71 38 2d   ..........:..q8-
      +00a0   40 58 2c 45 00 58 54 21 00 00 1e 56 5e 00 a0 a0   @X,E.XT!...V^...
      +00b0   a0 29 50 30 20 35 00 58 54 21 00 00 1a 00 00 00   .)P0 5.XT!......
      +00c0   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
      +00d0   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
      +00e0   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
      +00f0   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 29   ...............)

   Connector:   card0-HDMI-A-1
      i2c_busno:   -1
      name:        (null)
      dev:         (null)
      enabled:     disabled
      status:      disconnected
      edid:        None

*** Sysfs_I2C_Info report ***
   All Sysfs_I2C_Info records
      Sysfs_I2C_Info at: 0x556eda44e820
         busno:                     0
         name:                      NVIDIA i2c adapter 3 at 8:00.0
         adapter_path:              (null)
         adapter_class:             (null)
         driver:                    (null)
         driver_version:            (null)
         conflicting_driver_names:  
ddcutil: drm_common.c:87: adapter_supports_drm_using_drm_api: Assertion `adapter_path' failed.
[1]    8852 IOT instruction (core dumped)  ./src/ddcutil detect -v --f2``

```

rockowitz, 2024-03-11:
The assert failure in drm_common.c has been fixed.  

I came across a clue this afternoon as to why a test case for for the missing DRM information is elusive.  See https://super-unix.com/unixlinux/linux-how-to-make-linux-detect-re-probe-monitors-with-intel-i915-driver/ and section "Force-probing a connector" in https://dri.freedesktop.org/docs/drm/gpu/drm-uapi.html

digitaltrails, 2024-03-11:
Works here too.  But I still get -3021 "DDCRC_INTERNAL_ERROR" if I do:
```
busctl --user call com.ddcutil.DdcutilService /com/ddcutil/DdcutilObject com.ddcutil.DdcutilInterface GetVcp isyu 2 "" "0x10" "0"
```

rockowitz, 2024-03-11:
It's a proof of concept hack, but try the following in your failure scenario:
  -  plug the monitor in
  - verify that status=disconnected and edid has no value
  - execute **echo detect > /sys/class/drm/<connector>/status** for the relevant connector., e.g. "card0-DP-1"
  - status should now be connected, and edid have a value
  
If the above works, then ensure that ddcutil-service calls ddca_redetect_displays() at least once.
- GetVcp should now succeed

digitaltrails, 2024-03-25:
With the latest code detect is now correctly reporting the EDID for card-0-dp2, although --f2 causes a segmentation fault.
```
./src/ddcutil detect -v --f2
*** Sys_Drm_Connector report: Detailed /sys/class/drm report: ***

   Display connectors reported by DRM:
   Connector:   card0-DP-1
      i2c_busno:   -1
      connector_number: -1
      name:        (null)
      dev:         (null)
      enabled:     disabled
      status:      connected
      edid:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 22 f0 69 28 01 01 01 01   ........".i(....
      +0010   08 14 01 04 a5 36 23 78 2e fc 81 a4 55 4d 9d 25   .....6#x....UM.%
      +0020   12 50 54 21 08 00 81 40 81 80 95 00 a9 40 b3 00   .PT!...@.....@..
      +0030   d1 c0 01 01 01 01 28 3c 80 a0 70 b0 23 40 30 20   ......(<..p.#@0 
      +0040   36 00 22 60 21 00 00 1a 00 00 00 fd 00 3b 3d 18   6."`!........;=.
      +0050   50 11 00 0a 20 20 20 20 20 20 00 00 00 fc 00 48   P...      .....H
      +0060   50 20 5a 52 32 34 77 0a 20 20 20 20 00 00 00 ff   P ZR24w.    ....
      +0070   00 43 4e 54 30 30 38 31 31 4a 36 0a 20 20 00 3d   .CNT00811J6.  .=

   Connector:   card0-DP-2
      i2c_busno:   -1
      connector_number: -1
      name:        (null)
      dev:         (null)
      enabled:     disabled
      status:      disconnected
      edid:        None

   Connector:   card0-DP-3
      i2c_busno:   -1
      connector_number: -1
      name:        (null)
      dev:         (null)
      enabled:     disabled
      status:      connected
      edid:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 1e 6d 07 77 7b 51 01 00   .........m.w{Q..
      +0010   04 1d 01 04 b5 3c 22 78 9e 3e 31 ae 50 47 ac 27   .....<"x.>1.PG.'
      +0020   0c 50 54 21 08 00 71 40 81 80 81 c0 a9 c0 d1 c0   .PT!..q@........
      +0030   81 00 01 01 01 01 4d d0 00 a0 f0 70 3e 80 30 20   ......M....p>.0 
      +0040   65 0c 58 54 21 00 00 1a 28 68 00 a0 f0 70 3e 80   e.XT!...(h...p>.
      +0050   08 90 65 0c 58 54 21 00 00 1a 00 00 00 fd 00 38   ..e.XT!........8
      +0060   3d 1e 87 38 00 0a 20 20 20 20 20 20 00 00 00 fc   =..8..      ....
      +0070   00 4c 47 20 48 44 52 20 34 4b 0a 20 20 20 01 9e   .LG HDR 4K.   ..
      +0080   02 03 19 71 44 90 04 03 01 23 09 07 07 83 01 00   ...qD....#......
      +0090   00 e3 05 c0 00 e3 06 05 01 02 3a 80 18 71 38 2d   ..........:..q8-
      +00a0   40 58 2c 45 00 58 54 21 00 00 1e 56 5e 00 a0 a0   @X,E.XT!...V^...
      +00b0   a0 29 50 30 20 35 00 58 54 21 00 00 1a 00 00 00   .)P0 5.XT!......
      +00c0   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
      +00d0   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
      +00e0   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
      +00f0   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 29   ...............)

   Connector:   card0-HDMI-A-1
      i2c_busno:   -1
      connector_number: -1
      name:        (null)
      dev:         (null)
      enabled:     disabled
      status:      disconnected
      edid:        None

*** Sysfs_I2C_Info report ***
   All Sysfs_I2C_Info records
[... 51 lines cut ...]
         adapter_path:              (null)
         adapter_class:             (null)
         driver:                    (null)
         driver_version:            (null)
         conflicting_driver_names:  

*** Sysfs I2C devices possibly associated with displays ***
I2C buses to check: x00 x01 x02 x03

*** Sys_Conflicting_Driver report: Check for Conflicting Device Drivers ***
   No conflicting drivers found

*** Sysfs Reports Done ***

Drm_Connector_State at: 0x5602339f5050
   cardno:               0
   connector_id:         88
   connector_type:      10 - DRM_MODE_CONNECTOR_DisplayPort
   connector_type_id:   1
   connection:          1 - DRM_MODE_CONNECTED
      Properties:
      dpms:              0 - On
      link_status:       0 - Good
[1]    1476 segmentation fault (core dumped)  ./src/ddcutil detect -v --f2
```

digitaltrails, 2024-03-25:
Running with gdb and using where results in:
```
#0  get_enum_value_name (meta=0x0, value=0) at drm_connector_state.c:97
#1  0x00005555555f0e10 in dbgrpt_connector_state (state=0x555555689050, depth=depth@entry=0)
    at drm_connector_state.c:345
#2  0x00005555555f1222 in report_drm_connector_states (depth=depth@entry=0) at drm_connector_state.c:544
#3  0x0000555555596235 in main (argc=<optimized out>, argv=<optimized out>) at main.c:905

```

digitaltrails, 2024-03-25:
The previous test of of a service GetVCP that failed in the first comment of this issue, now works:
```
busctl --user call com.ddcutil.DdcutilService /com/ddcutil/DdcutilObject com.ddcutil.DdcutilInterface GetVcp isyu "2" "" "16" "0"
qqsis 92 100 "current value =    92, max value =   100" 0 "OK"
```
In doing these recent tests, I did as before.  I plugged in and connected the Beng after the system was already up.

One thing though, the status of the Beng is still disconnected:

```
cat /sys/class/drm/card0-DP-[123]/status
connected
disconnected
connected

```
But if I have root do `echo detect > /sys/class/drm/card0-DP-2/status`, the status does refresh.

digitaltrails, 2024-03-25:
Note that `./src/ddcutil detect -v` is working and produces an EDID. The service Detect method is working too.

rockowitz, 2024-03-26:
A fix for the segfault in get_enum_value_name() has  been pushed to branch 2.1.5

rockowitz, 2024-03-26:
What happens now if you run through your test sequence with **ddcutil detect -v --enable-try-get-edid-from-sysfs --f15**

What is the output of  **cat /sys/class/drm/card0-DP-2/edid | parse-edid** before and after the above command?