rockowitz/ddcutil #395: User defined features not overriding monitor values

unai-ndz, 2024-03-23:
```ini
> cat .local/share/ddcutil/SAM-LS27A800U-29092.mccs
MFG_ID       SAM
MODEL        LS27A800U
PRODUCT_CODE 29092
MCCS_VERSION 2.1
FEATURE_CODE 60 Input Source
  ATTRS NC RW
  VALUE x05 HDMI-1
  VALUE x06 HDMI-2
  VALUE x0f DisplayPort
FEATURE_CODE 70 Input Source2
  ATTRS NC RW
  VALUE x05 HDMI
  VALUE x0f DisplayPort
#  VALUE x06 HDMI Maybe usb-c if connected, currently switches to HDMI-1
```

```sh
> rm -rf .cache/ddcutil/capabilities; ddcutil --model "LS27A800U" cap --verbose
Unable to open directory /sys/devices/pci0000:00/0000:00:03.2/0000:10:00.0/drm: No such file or directory
Processed feature definition file: .local/share/ddcutil/SAM-LS27A800U-29092.mccs
Unparsed capabilities string: (prot(monitor)type(lcd)model(FALCON)cmds(01 02 03 07 0C 4E F3 E3)vcp(02 04 05 08 0B 0C 10 12 14(01 04 05 06 07 08 0A 0B) 16 18 1A 6C 6E 70 AC AE B6 C0 C6 C8 C9 CA CC(00 02 03 04 05 08 09 0A 0D) D6(01 04) DC(00 01 02 03 04) DF 60(01 03) 62 8D FF)mswhql(1)mccs_ver(2.0)asset_eep(32)mpu_ver(01))
Model: FALCON
MCCS version: 2.0
Commands:
   Op Code: 01 (VCP Request)
   Op Code: 02 (VCP Response)
   Op Code: 03 (VCP Set)
   Op Code: 07 (Timing Request)
   Op Code: 0C (Save Settings)
   Op Code: 4E (Unrecognized operation code)
   Op Code: F3 (Capabilities Request)
   Op Code: E3 (Capabilities Reply)
VCP Features:
   Feature: 02 (New control value)
   Feature: 04 (Restore factory defaults)
   Feature: 05 (Restore factory brightness/contrast defaults)
   Feature: 08 (Restore color defaults)
   Feature: 0B (Color temperature increment)
   Feature: 0C (Color temperature request)
   Feature: 10 (Brightness)
   Feature: 12 (Contrast)
   Feature: 14 (Select color preset)
      Values (unparsed): 01 04 05 06 07 08 0A 0B
      Values (  parsed):
         01: sRGB
         04: 5000 K
         05: 6500 K
         06: 7500 K
         07: 8200 K
         08: 9300 K
         0a: 11500 K
         0b: User 1
   Feature: 16 (Video gain: Red)
   Feature: 18 (Video gain: Green)
   Feature: 1A (Video gain: Blue)
   Feature: 6C (Video black level: Red)
   Feature: 6E (Video black level: Green)
   Feature: 70 (Input Source2)
   Feature: AC (Horizontal frequency)
   Feature: AE (Vertical frequency)
   Feature: B6 (Display technology type)
   Feature: C0 (Display usage time)
   Feature: C6 (Application enable key)
   Feature: C8 (Display controller type)
   Feature: C9 (Display firmware level)
   Feature: CA (OSD)
   Feature: CC (OSD Language)
      Values (unparsed): 00 02 03 04 05 08 09 0A 0D
      Values (  parsed):
         00: Reserved value, must be ignored
         02: English
         03: French
         04: German
         05: Italian
         08: Portuguese (Portugal)
         09: Russian
         0a: Spanish
         0d: Chinese (simplified / Kantai)
   Feature: D6 (Power mode)
      Values (unparsed): 01 04
      Values (  parsed):
         01: DPM: On,  DPMS: Off
         04: DPM: Off, DPMS: Off
   Feature: DC (Display Mode)
      Values (unparsed): 00 01 02 03 04
      Values (  parsed):
         00: Standard/Default mode
         01: Productivity
         02: Mixed
         03: Movie
         04: User defined
   Feature: DF (VCP Version)
   Feature: 60 (Input Source)
      Values (unparsed): 01 03
      Values (  parsed):
         01: VGA-1
         03: DVI-1
   Feature: 62 (Audio speaker volume)
   Feature: 8D (Audio Mute)
   Feature: FF (Manufacturer specific feature)
```

If I disable udf the only thing that changes is feature 70:
```
   Feature: 70 (Video black level: Blue)
```

unai-ndz, 2024-03-25:
>The parsed capabilities that you show is the output with option --enable-udf (which is the default). Is that correct?

Yes, and afaik it's detecting the udf file for that monitor as shown in the output: `Processed feature definition file: .local/share/ddcutil/SAM-LS27A800U-29092.mccs`

```bash
> ddcutil capabilities  --model "LS27A800U" --disable-udf --verbose --disable-capabilities-cache
Unable to open directory /sys/devices/pci0000:00/0000:00:03.2/0000:10:00.0/drm: No such file or directory
Unparsed capabilities string: (prot(monitor)type(lcd)model(FALCON)cmds(01 02 03 07 0C 4E F3 E3)vcp(02 04 05 08 0B 0C 10 12 14(01 04 05 06 07 08 0A 0B) 16 18 1A 6C 6E 70 AC AE B6 C0 C6 C8 C9 CA CC(00 02 03 04 05 08 09 0A 0D) D6(01 04) DC(00 01 02 03 04) DF 60(01 03) 62 8D FF)mswhql(1)mccs_ver(2.0)asset_eep(32)mpu_ver(01))
Model: FALCON
MCCS version: 2.0
Commands:
   Op Code: 01 (VCP Request)
   Op Code: 02 (VCP Response)
   Op Code: 03 (VCP Set)
   Op Code: 07 (Timing Request)
   Op Code: 0C (Save Settings)
   Op Code: 4E (Unrecognized operation code)
   Op Code: F3 (Capabilities Request)
   Op Code: E3 (Capabilities Reply)
VCP Features:
   Feature: 02 (New control value)
   Feature: 04 (Restore factory defaults)
   Feature: 05 (Restore factory brightness/contrast defaults)
   Feature: 08 (Restore color defaults)
   Feature: 0B (Color temperature increment)
   Feature: 0C (Color temperature request)
   Feature: 10 (Brightness)
   Feature: 12 (Contrast)
   Feature: 14 (Select color preset)
      Values (unparsed): 01 04 05 06 07 08 0A 0B
      Values (  parsed):
         01: sRGB
         04: 5000 K
         05: 6500 K
         06: 7500 K
         07: 8200 K
         08: 9300 K
         0a: 11500 K
         0b: User 1
   Feature: 16 (Video gain: Red)
   Feature: 18 (Video gain: Green)
   Feature: 1A (Video gain: Blue)
   Feature: 6C (Video black level: Red)
   Feature: 6E (Video black level: Green)
   Feature: 70 (Video black level: Blue)
   Feature: AC (Horizontal frequency)
   Feature: AE (Vertical frequency)
   Feature: B6 (Display technology type)
   Feature: C0 (Display usage time)
   Feature: C6 (Application enable key)
   Feature: C8 (Display controller type)
   Feature: C9 (Display firmware level)
   Feature: CA (OSD)
   Feature: CC (OSD Language)
      Values (unparsed): 00 02 03 04 05 08 09 0A 0D
      Values (  parsed):
         00: Reserved value, must be ignored
         02: English
         03: French
         04: German
         05: Italian
         08: Portuguese (Portugal)
         09: Russian
         0a: Spanish
         0d: Chinese (simplified / Kantai)
   Feature: D6 (Power mode)
      Values (unparsed): 01 04
      Values (  parsed):
         01: DPM: On,  DPMS: Off
         04: DPM: Off, DPMS: Off
   Feature: DC (Display Mode)
      Values (unparsed): 00 01 02 03 04
      Values (  parsed):
         00: Standard/Default mode
         01: Productivity
         02: Mixed
         03: Movie
         04: User defined
   Feature: DF (VCP Version)
   Feature: 60 (Input Source)
      Values (unparsed): 01 03
      Values (  parsed):
         01: VGA-1
         03: DVI-1
   Feature: 62 (Audio speaker volume)
   Feature: 8D (Audio Mute)
   Feature: FF (Manufacturer specific feature)
```

```bash
> ddcutil --model "LS27A800U" --enable-udf getvcp 60 70 --disable-capabilities-cache
Unable to open directory /sys/devices/pci0000:00/0000:00:03.2/0000:10:00.0/drm: No such file or directory
VCP code 0x60 (Input Source                  ): HDMI-2 (sl=0x06)
VCP code 0x70 (Input Source2                 ): Unrecognized value (sl=0x32)
```

Every `setvcp` outputs: `Unable to open directory /sys/devices/pci0000:00/0000:00:03.2/0000:10:00.0/drm: No such file or directory` and nothing else.

```bash
ddcutil --model "LS27A800U" --enable-udf setvcp 60 x05 --disable-capabilities-cache
# monitor switches to HDMI input
ddcutil --model "LS27A800U" --enable-udf setvcp 60 x11 --disable-capabilities-cache
# Turns monitor off for a second as if trying to change input but goes back to whatever the input was set before

ddcutil --model "LS27A800U" --enable-udf setvcp 70 x05 --disable-capabilities-cache
# Turns screen yellow
ddcutil --model "LS27A800U" --enable-udf setvcp 70 50 --disable-capabilities-cache
# Restores the original colors
```

```bash
> ddcutil --model "LS27A800U" --disable-udf getvcp 60 70 --disable-capabilities-cache
Unable to open directory /sys/devices/pci0000:00/0000:00:03.2/0000:10:00.0/drm: No such file or directory
VCP code 0x60 (Input Source                  ): Composite video 2 (sl=0x06)
VCP code 0x70 (Video black level: Blue       ): current value =    50, max value =   100
```

The `setvcp` commands using ` --disable-udf ` instead of enable have the same exact results as above but I'm including them anyways for explicitness.
```bash
ddcutil --model "LS27A800U" --disable-udf setvcp 60 x05 --disable-capabilities-cache
# monitor switches to HDMI input
ddcutil --model "LS27A800U" --disable-udf setvcp 60 x11 --disable-capabilities-cache
# Turns monitor off for a second as if trying to change input but goes back to whatever the input was set before

ddcutil --model "LS27A800U" --disable-udf setvcp 70 x05 --disable-capabilities-cache
# Turns screen yellow
ddcutil --model "LS27A800U" --disable-udf setvcp 70 50 --disable-capabilities-cache
# Restores the original colors
```

I also tried to add a similar mccs for another monitor with the same results, the file seems to be picked up by ddcutil but not applied.

[sudo ddcutil interrogate --verbose](https://pastebin.com/TS2qGHTd)

rockowitz, 2024-03-29:
There are two different contexts here. 

First, is whether user supplied feature definitions are used to interpret the output of **getvcp**.  This appears to be working properly.

Second, is whether UDF feature definitions are reflected in **capabilities** command output.  This was not happening correctly.   If a feature is user-specified, the feature name is now prefaced with "UDF:".  If the feature is non-continuous, the header for the values list is "UDF Values" instead of values. Also, for verbose output the "Values (unparsed)" line is not shown, as that is what is extracted from the monitor's capabilities string, but ignored for user defined features.

These changes have been applied to branch 2.1.5-dev.

hostmit, 2025-03-28:
Hello! Did you happen to find a way to switch to input - Usb-C code? Thanks.