rockowitz/ddcutil #396: Samsung LS25BG400EUXEN (Odyssey G25)  feature x10 not set

rockowitz, 2024-03-24:
I have a Samsung LS25BG400EUXEN (Odyssey G25) It apparently responds with it's 'Picture Mode' on feature 0x10 instead of brightness, i.e if 'Custom' mode is set, response is always 0, if 'sRGB' response is 5. How to workaround this? I would really like to change brightness with media keys, instead of my current solution of binding modifier + number keys to set specific brightness.

_Originally posted by @missonserch in https://github.com/rockowitz/ddcutil/issues/323#issuecomment-2016401535_

rockowitz, 2024-03-24:
It's not clear what you're describing here.  I gather that "Picture Mode" is a setting in the OSD. Unfortunately, more recent monitors have modes, such as for games or showing video, that disable the ability to change various features.  You need to poke around in the OSD to determine what setting disables your ability to set the brightness using feature x10.

digitaltrails, 2024-03-24:
I would suggest downloading the Samsung manual. The example manual I've seen states what disables the brightness control. For example, some monitors disable the brightness if *Eye Saver Mode* is enabled.

Presumably anything that adapts to ambient lighting such as *Eye Care: Ambient Picture* or *Eco Sensor Mode* might also cause issues.

missonserch, 2024-03-24:
> It's not clear what you're describing here. I gather that "Picture Mode" is a setting in the OSD. Unfortunately, more recent monitors have modes, such as for games or showing video, that disable the [...]

Yes, “Picture mode” is a setting in the ost ranging from “Custom” to “Dynamic contrast” (0 to 7 respectively.) For the record, I’ve tried disabling every feature, including freesync, low input lag mode, etc. I’ve updated to the latest firmware, the issue persists, I have no idea why they made it this way, it does not make any sense. Contrast and sharpness are reported properly. I’ve probed other available features in case brightness was assigned to something wierd for whatever reason, that’s not the case. My second monitor from dell works flawlessly, so that can’t be a gpu/driver/system issue.

digitaltrails, 2024-03-24:
I did a quick search, aside from Samsung's own manual, there's not a lot to find, this looked the most promising:
[https://superuser.com/questions/822479/cant-change-monitor-brightness](https://superuser.com/questions/822479/cant-change-monitor-brightness)
Other search results mentioned that different cables might have an effect.

My own LG monitor locks contrast on all Picture-Modes except custom, so I always leave it in custom.

Zhonglin-Niu, 2024-04-03:
I have the exact same problem.

[image]

For some reason, the brightness value is set to the current mode's index in Picture Mode. However, setvcp did work with a verification failed message every time since getvcp is working incorrectly.

digitaltrails, 2024-04-03:
Has anyone contacted Samsung?