rockowitz/ddcutil #397: Samsung 57 G95NC Feature definition

saltyautomation, 2024-03-27:
I am trying to control the inputs on the Samsung 57" G95NC. Specifically looking to change the input on the PBP source and also be able to enable / disable PBP/PIP. Is there some output I need to gather to create a feature definition file? Happy to help collect the data and enable input switching for PIP/PBP modes similar to the Dell monitors.

rockowitz, 2024-03-31:
Your best bet is to make changes using the OSD, and use the output of **ddcutil getvcp scan --brief** to see what has changed.

A more precise way to figure out what is needed is to sniff the I2C bus while making changes from Samsung's Windows application.   See [Reverse Engineering Proprietary DDC Extensions](http://localhost:8000/sniffing/#reverse-engineering-proprietary-ddc-extensions).

saltyautomation, 2024-04-02:
Well I made a bunch of changes to inputs using the OSD and ran the ddcutil getvcp command after each change. Captured the results and compared there were no differences. Although I am running the command and capturing the changes from an rpi zero attached to the 2nd HDMI interface. Not sure that matters though as I am able to query and change some inputs from that device. 

Guess it is on to read the reverse engineering guide. 
Thanks!

robertswiecki, 2025-06-22:
@saltyautomation Did you have any luck with PbP? I can change input (VCP 0x60), but didn't find any PbP controls.

leinardi, 2026-10-01:
Quick question for @ema987 or @mwd102, now that you're both on firmware 1009.2: for a plain input switch, does HDMI 1 / HDMI 2 still take 0x05 / 0x06 (what ema987 found on 1007.0), or 0x11 / 0x12 as in mwd102's notes?

I ask because I keep a small list of tested input values per monitor for `monmux` (https://github.com/leinardi/monmux), a command-line tool that runs `ddcutil` for you. Right now the G95NC entry has the 1007.0 values, and I don't want to send anyone a value that's wrong on newer firmware. If either of you wants to try it, the steps are here: https://github.com/leinardi/monmux/issues/80

ema987, 2026-10-10:
@leinardi sorry it took so long to give you an answer, I confirm it's `0x11 / 0x12` that works. I've replied in your repo issue, also added `0x32` to switch to HDMI3. I've also left some notes about what I found out during testing, but I think we should continue the discussion about this in your repo.

leinardi, 2026-10-10:
Hey no worries and thank you for testing! I'll reply you in the monmux repo 👍

dbibicoff, 2026-10-31:
Thanks for documenting the E2/E3 codes. Data point: on an LS57CG952NNXZA (North America), firmware **1008.2** (the newest on Samsung's CA/US support pages) does **not** respond to them. E2/E3/E8 reads fail, and `E2=0x03` does nothing on screen. 0x60 works normally. So 1007.0 and 1008.2 both lack the PBP codes.

@ema987, you mentioned upgrading to 1009.2. Where did you get it? It doesn't appear on any Samsung support page I can find (CA, US, EG list 1008.2 as newest). A different region's page, Samsung support directly, or something else?

ema987, 2026-11-01:
Hey @dbibicoff , thanks for confirming these codes do not work on `1008.2` either, unfortunately we did not have any evidence of it, so now we are sure these codes work only on the latest `1009.2` where Samsung seems to have enabled them.

Regarding the firmware, you are right it doesn't appear listed on any region but the korean one: it works perfectly for other regions too, my monitor is from EU. Have a look here on reddit:
- https://www.reddit.com/r/ultrawidemasterrace/comments/1puluzt/samsung_odyssey_neo_g9_57_g95nc_new_firmware/
- https://www.reddit.com/r/ultrawidemasterrace/comments/1pqsuxr/samsung_57_odyssey_neo_g9_firmware_10092/

dbibicoff, 2026-11-01:
Thanks @ema987, that did it. I flashed the Korean 1009.2 on my North America unit (LS57CG952NNXZA) and the codes work now.

Data point for anyone else on a US/CA model:
- `E2` layouts all work on 1009.2: `00` single, `01` split, `10` PIP, `03`/`04`/`05` 2-way (only from an existing 2-way split, ignored from single or PIP), `06`-`09` 3-way. `02` just blanks the screen on mine.
- `E3` per-pane source works (`(pane<<8)|src`, src `00`/`01`/`02` = HDMI1-3, `10` = DP).
- Changing a 3-way layout can reset a pane's source, so set sources after the layout.

Appreciate you pointing me to it.