rockowitz/ddcutil #399: Getvcp and multiuple feature codes

Alvarofg, 2024-03-31:
Hi, I'm trying to read more than one feature at once, based on what I found in the docs. However when I run 
`$ ddcutil getvcp 10 12`
I get this error:
`Too many arguments`

in [here](https://www.ddcutil.com/command_getvcp/) it is mentioned that 

> Command getvcp
> Syntax: getvcp <display-selection-options> <other options> <feature-codes-or-group>
> ...
> Briefly, you can specify one or more feature codes or a feature group, but feature codes and groups cannot be combined. For example, the folling commands are valid:
> 
> $ ddcutil getvcp 10 12

However [here](https://www.ddcutil.com/feature_selection/): it mentions:

> Commands getvcp, setvcp, and vcpinfo operate on a single VCP feature. Additionally, getvcp and vcpinfo can specify a named group of features.

Is this a documentation issue or I'm missing something. TIA

rockowitz, 2024-03-31:
What version of **ddcutil** are you running?  The ability to specify multiple features on **getvcp** was added in release 1.3.0.

The documentation you cite needs to be updated to properly specify multiple feature codes.  It has not kept up with the release notes.

Alvarofg, 2024-03-31:
That's it then, I'm running ddcutil 0.9.9 on a raspberry pi zero w (armv6l)

rockowitz, 2024-03-31:
The documentation cited has been updated to note that the ability to specify multiple features was added in release 1.3.0.

closed 2024-03-31 by rockowitz

Alvarofg, 2024-03-31:
Thanks, I only like to add for future reference I was able to build v 2.1.4 from source in the pi zero and I can confirm I can now retrieve more than one feature code