rockowitz/ddcutil #403: User Defined Feature enhancements/documentation and questions

digitaltrails, 2024-04-08:
I was attempting to use a User Defined Feature to work around a problem VDU.

First some general comments on the UDF's.

It would be good if filename was more easily derived or more exactly defined.  I had to guess a bit and only got it right by trial and error, dtrace of open, and a fortuitous error in the file content that caused ddcutil to out `Error(s) processing monitor definition file: blah.mccs`.

So the file name is constructed from the output of ddcutil detect as follows:

1. Split the Mfg Id by spaces and retain the first three letter acronym only.
2. Trim the Model and substitute "_" for each internal space.
3. Split the product code by spaces and retain the first (non-hex) part.
4. Combine 1, 2, and 3 with hyphens (-) and suffix with the ".mccs" file type.

More formally defined in awk:

```
ddcutil detect | awk -F: '
   $1 ~ /Mfg id/ { split($2, parts, " "); mfg_id=parts[1];}
   $1 ~ /Model/ { model=$2;  gsub(/^ +/, "", model) gsub(" ", "_", model);}
   $1 ~ /Product code/ {split($2, parts, " "); product_code=parts[1];
                        printf("%s-%s-%s.mccs\n", mfg_id, model, product_code);}
   '
```
Then use the same extracted part values at the top of the mccs file.  

### Suggestions
- It would be quite nice if ddcutil could print out the required filenames (perhaps there is already an option where it does this).
- It might be nice if there was an option to create a starter file with just the header.  I guess the above awk script could be repurposed for that.

rockowitz, 2024-04-10:
The file name is quite precisely defined.  The mfg-id portion of the file name is exactly the 3 character manufacturer code as specified in the EDID.  The product-code is an unsigned integer (16 bit) as specified in the EDID. The model name is that given in the EDID, with non-alphanumeric characters replaced by underscores. The latter transformation could have been better documented.  For verbose output of **capabilities**, **getvcp**, **setvcp**, **dumpvcp**, and **probe**, the fully qualified name of the feature definition file is reported (if it is found), or the simple name is reported (if it is not found).

I have updated section [User Defined Features](http://www.ddcutil.com/udf) of the online documentation to more clearly document the above.  In addition, with the latest changes to branch 2.1.5-dev, **detect --verbose** also reports the name of the feature definition file.

rockowitz, 2024-04-10:
Re the meaningful value in the SH byte in what otherwise is a SNC feature, the simplest extension would be to allow SNC and CNC as valid attributes, with SNC being a synonym for the currently valid NC. ddcutil and ddcui would just report such features as 2 byte hex values, without any interpretation. 

A more complete solution would be to have an extended SNC type, say "ESNC" what would allow for specific values in SL with an arbitrary value in SH, or perhaps even a list of specific values in SH.  This would require creating a new interpretation function for ddcutil, and a new widget for ddcui. The latter might prove tricky given the space constraints.  

I'm open to implementing the SNC/CNC extension at this time and seeing how it works out.

digitaltrails, 2024-04-10:
> I'm open to implementing the SNC/CNC extension at this time and seeing how it works out.

Being able to promote SNC up to CNC would solve such problems with no change to existing clients, so it seems to be a good option.  Given this is probably quite a rare situation, this seems quite a minimal way of dealing with it. 

I probably wouldn't have gone to the effort to change ddcutil-service and vdu_controls if this was already an option.  With hindsight I wonder if I should have given vdu_controls some ability to create/manage UDF files.

rockowitz, 2024-05-25:
Added a new NC subtype XNC (extended non-continuous, symbolic constant DDCA_EXTENDED_NC).

In common with SNC, XNC regards the SL byte as having a list of valid values.  However, it also regards the SH byte as significant.  While there are no features with this subtype automatically known to **ddcutil** from the Monitor Control Command Set),, it can be specified as an attribute in a user defined features file.  

For example, given the following .mccs file:

~~~
MFG_ID       SAM
MODEL        U32H75x
PRODUCT_CODE 3586
MCCS_VERSION 2.1
FEATURE_CODE 60 Input Source
  ATTRS XNC RW
  VALUE x05 HDMI-1
  VALUE x06 HDMI-2
  VALUE x0f DisplayPort
~~~

Given a value with SH=0x00, SL=0x05, 
   Normal output:  VCP code 0x60 (Input Source                  ): HDMI-1 (sl=0x05), sh=0x00
   Terse output: VCP 60 SNC x00 x05

Input: specify a two byte hex value, e.g  ddcutil setvcp 60 x0105

Shared library IO is performed in the normal way using ddca_get_non_table_vcp_value() and ddca_set_non_table_vcp_value()