rockowitz/ddcutil #407: Core dump on ddcutil getvcp

jcassee, 2024-04-22:
Not sure whether this is the same problem as the last issues, but this core dump happened with the latest 2.1.5-dev commit (bbb471b83b996458bb1b7140c613e869a0ec7c7a).

Output of `coredumpctl`:
```
           PID: 201823 (ddcutil)
           UID: [...]
           GID: [...]
        Signal: 6 (ABRT)
     Timestamp: Mon 2024-04-22 10:16:37 CEST (3min 47s ago)
  Command Line: /usr/bin/ddcutil getvcp --brief D6 --bus 15 --sleep-multiplier 5
    Executable: /usr/bin/ddcutil
    [...]
  Size on Disk: 120.6K
       Message: Process 201823 (ddcutil) of user 1000 dumped core.

                Stack trace of thread 201823:
                #0  0x000078938df2c32c n/a (libc.so.6 + 0x8d32c)
                #1  0x000078938dedb6c8 raise (libc.so.6 + 0x3c6c8)
                #2  0x000078938dec34b8 abort (libc.so.6 + 0x244b8)
                #3  0x000078938dec33dc n/a (libc.so.6 + 0x243dc)
                #4  0x000078938ded3d46 __assert_fail (libc.so.6 + 0x34d46)
                #5  0x00005e10f64d808a dsa2_record_final (ddcutil + 0x9308a)
                #6  0x00005e10f64df9a4 pdd_record_final (ddcutil + 0x9a9a4)
                #7  0x00005e10f64a2cde ddc_write_read_with_retry (ddcutil + 0x5dcde)
                #8  0x00005e10f64a85d0 ddc_get_nontable_vcp_value (ddcutil + 0x635d0)
                #9  0x00005e10f649a6f1 ddc_initial_checks_by_dh (ddcutil + 0x556f1)
                #10 0x00005e10f649b259 ddc_initial_checks_by_dref (ddcutil + 0x56259)
                #11 0x00005e10f64867f9 find_dref (ddcutil + 0x417f9)
                #12 0x00005e10f64847e9 main (ddcutil + 0x3f7e9)
                #13 0x000078938dec4cd0 n/a (libc.so.6 + 0x25cd0)
                #14 0x000078938dec4d8a __libc_start_main (libc.so.6 + 0x25d8a)
                #15 0x00005e10f6485905 _start (ddcutil + 0x40905)
                ELF object binary architecture: AMD x86-64
```

I can also provide the binary core dump if useful.

Output of `ddcutil interrogate`:
[ddcutil-interrogate.log](https://github.com/rockowitz/ddcutil/files/15059822/ddcutil-interrogate.log)

rockowitz, 2024-04-23:
Thank you for the clear report.  The bug should be addressed in patch .  It was triggered when the sleep-multiplier specified exceeded 2.0, the greatest value in the dynamic sleep step table.  

This bug is unrelated patch bbb471b, which addressed issue #405.

jcassee, 2024-04-23:
Thank you for the quick fix! No more crash.

closed 2024-04-23 by rockowitz