rockowitz/ddcutil #408: Segfault in scan_sys_drm_connectors called from ddcutil/chkusbmon

rockowitz, 2024-04-23:
@scr4bble posted this coredump on issue #405.  It is different from the crash reported on that issue so has been moved here.

Is this the same issue?

Apr 22 09:34:19 lenovo-t480 systemd-coredump[118855]: [🡕] Process 118813 (ddcutil) of user 0 dumped core.
                                                      
                                                      Module libcap.so.2 from rpm libcap-2.48-9.fc39.x86_64
                                                      Module libpcre2-8.so.0 from rpm pcre2-10.42-1.fc39.2.x86_64
                                                      Module libdrm.so.2 from rpm libdrm-2.4.120-1.fc39.x86_64
                                                      Module libudev.so.1 from rpm systemd-254.10-1.fc39.x86_64
                                                      Module libglib-2.0.so.0 from rpm glib2-2.78.3-1.fc39.x86_64
                                                      Module libjansson.so.4 from rpm jansson-2.13.1-7.fc39.x86_64
                                                      Module ddcutil from rpm ddcutil-2.1.2-1.fc39.x86_64
                                                      Stack trace of thread 118813:
                                                      #0  0x00007fa35251e8dd __strlen_avx2 (libc.so.6 + 0x1618dd)
                                                      #1  0x0000559033d8b222 assemble_sysfs_path2.constprop.0 (ddcutil + 0xc1222)
                                                      #2  0x0000559033d83aed rpt_attr_note_indirect_subdir (ddcutil + 0xb9aed)
                                                      #3  0x0000559033d43256 find_adapter (ddcutil + 0x79256)
                                                      #4  0x0000559033d44533 one_drm_connector (ddcutil + 0x7a533)
                                                      #5  0x0000559033d7b598 dir_filtered_ordered_foreach (ddcutil + 0xb1598)
                                                      #6  0x0000559033d454fb scan_sys_drm_connectors (ddcutil + 0x7b4fb)
                                                      #7  0x0000559033d1f409 submaster_initializer (ddcutil + 0x55409)
                                                      #8  0x0000559033d09784 main (ddcutil + 0x3f784)
                                                      #9  0x00007fa3523e514a __libc_start_call_main (libc.so.6 + 0x2814a)
                                                      #10 0x00007fa3523e520b __libc_start_main@@GLIBC_2.34 (libc.so.6 + 0x2820b)
                                                      #11 0x0000559033d0aa15 _start (ddcutil + 0x40a15)
                                                      ELF object binary architecture: AMD x86-64
Apr 22 09:34:19 lenovo-t480 (udev-worker)[118811]: hiddev1: Process '/usr/bin/ddcutil chkusbmon /dev/usb/hiddev1 -v' terminated by signal SEGV.
Apr 22 09:34:19 lenovo-t480 audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-coredump@2-118852-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Apr 22 09:34:19 lenovo-t480 (udev-worker)[118811]: hiddev1: Failed to wait for spawned command '/usr/bin/ddcutil chkusbmon /dev/usb/hiddev1 -v': Input/output error
Apr 22 09:34:19 lenovo-t480 (udev-worker)[118811]: hiddev1: /usr/lib/udev/rules.d/60-ddcutil-usb.rules:8 Failed to execute "/usr/bin/ddcutil chkusbmon /dev/usb/hiddev1 -v": Input/output error
Apr 22 09:34:19 lenovo-t480 systemd[1]: systemd-coredump@1-118848-0.service: Deactivated successfully.

Does this issue cause the OS to hang on login? I usually get these when waking up laptop from sleep, typing password in, clicking enter and then getting 1-2 min freeze before getting into gnome...

rockowitz, 2024-04-26:
It appears that /sys is in an unexpected state after waking from sleep.  Short of adding asserts, it's unclear how to appropriately harden the code in a way that doesn't just create some other obscure error. That will take some thought.

Assuming you do not have any displays that use USB instead of DDC for communicating with the monitor's virtual control panel, the same workaround applies as for issue #405. Delete file 60-ddcutil-usb.rules from /usr/lib/udev/rules.d and /etc/udev/rules.d. Alternatively, build **dcutil** without support for monitors using USB by specifying **configure** option ***--disable-usb*** (and check that no vestigial copy of 60-ddutil-usb.rules remains installed).

scr4bble, 2024-09-02:
> Assuming you do not have any displays that use USB instead of DDC for communicating with the monitor's virtual control panel

Assumption is right. I removed the rules as temporary solution for now. Thanks for suggestion.