rockowitz/ddcutil #409: ASUS VS247: Invalid display

benjamb, 2024-04-24:
From a glance at documentation for the monitor and browsing through the menu, it claims to support DDC/CI, yet it is detected as an invalid display.

Output of `sudo ddcutil detect -v`:
```
Invalid display
   I2C bus:  /dev/i2c-1
      DRM connector:                         card0-HDMI-A-1
      /sys/class/drm/card0-HDMI-A-1/dpms     On
      /sys/class/drm/card0-HDMI-A-1/enabled  enabled
      /sys/class/drm/card0-HDMI-A-1/status   connected
      Driver:                                i915
      I2C address 0x50 (EDID) responsive:    true 
      I2C address 0x37 (DDC)  responsive:    true 
      Is eDP device:                         false
      Is LVDS device:                        false
      /sys/bus/i2c/devices/i2c-1/name        i915 gmbus dpb
      PCI device path:                       /sys/devices/pci0000:00/0000:00:02.0/i2c-1
   EDID synopsis:
      Mfg id:               ACI - Ancor Communications Inc
      Model:                ASUS VS247
      Product code:         9370  (0x249a)
      Serial number:        BCLMTF075328
      Binary serial number: 75328 (0x00012640)
      Manufacture year:     2011,  Week: 49
      EDID version:         1.3
      Extra descriptor:        
      Video input definition:    0x80 - Digital Input
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.641, 0.338
      Green x,y:        0.311, 0.619
      Blue  x,y:        0.159, 0.059
      Extension blocks: 1
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 04 69 9a 24 40 26 01 00   .........i.$@&..
      +0010   31 15 01 03 80 34 1d 78 2a 2a c5 a4 56 4f 9e 28   1....4.x**..VO.(
      +0020   0f 50 54 b7 ef 00 71 4f 81 40 81 80 95 00 b3 00   .PT...qO.@......
      +0030   d1 c0 81 c0 81 00 02 3a 80 18 71 38 2d 40 58 2c   .......:..q8-@X,
      +0040   45 00 09 25 21 00 00 1e 00 00 00 ff 00 42 43 4c   E..%!........BCL
      +0050   4d 54 46 30 37 35 33 32 38 0a 00 00 00 fd 00 32   MTF075328......2
      +0060   4b 18 53 11 00 0a 20 20 20 20 20 20 00 00 00 fc   K.S...      ....
      +0070   00 41 53 55 53 20 56 53 32 34 37 0a 20 20 01 80   .ASUS VS247.  ..
   DDC communication failed. (getvcp of feature x10 returned Error_Info[DDCRC_RETRIES in ddc_write_read_with_retry, causes: DDCRC_DDC_DATA(10)])

Invalid display
   I2C bus:  /dev/i2c-9
      DRM connector:                         card0-eDP-1
      /sys/class/drm/card0-eDP-1/dpms        On
      /sys/class/drm/card0-eDP-1/enabled     enabled
      /sys/class/drm/card0-eDP-1/status      connected
      Driver:                                i915
      I2C address 0x50 (EDID) responsive:    true 
      I2C address 0x37 (DDC)  responsive:    false
      Is eDP device:                         true 
      Is LVDS device:                        false
      /sys/bus/i2c/devices/i2c-9/name        AUX A/DDI A/PHY A
      PCI device path:                       /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/i2c-9
   EDID synopsis:
      Mfg id:               CMN - Chimei Innolux Corporation
      Model:                
      Product code:         5183  (0x143f)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2021,  Week: 37
      EDID version:         1.4
      Extra descriptor:        N140JCA-EEL
      Video input definition:    0xa5 - Digital Input (DisplayPort), Bit depth: 8
      Supported features:
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.590, 0.350
      Green x,y:        0.330, 0.555
      Blue  x,y:        0.153, 0.119
      Extension blocks: 0
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 0d ae 3f 14 00 00 00 00   ..........?.....
      +0010   25 1f 01 04 a5 1e 13 78 03 28 65 97 59 54 8e 27   %......x.(e.YT.'
      +0020   1e 50 54 00 00 00 01 01 01 01 01 01 01 01 01 01   .PT.............
      +0030   01 01 01 01 01 01 42 3c 80 a0 70 b0 24 40 30 20   ......B<..p.$@0 
      +0040   a6 00 2d bc 10 00 00 18 00 00 00 fd 00 28 3c 43   ..-..........(<C
      +0050   43 10 01 0a 20 20 20 20 20 20 00 00 00 fe 00 43   C...      .....C
      +0060   4d 4e 0a 20 20 20 20 20 20 20 20 20 00 00 00 fe   MN.         ....
      +0070   00 4e 31 34 30 4a 43 41 2d 45 45 4c 0a 20 00 39   .N140JCA-EEL. .9
   DDC communication failed. (getvcp of feature x10 returned Error_Info[DDCRC_RETRIES in ddc_write_read_with_retry, causes: EREMOTEIO(10)])
   This is an eDP laptop display. Laptop displays do not support DDC/CI.
```

I've attached the output of `sudo ddcutil interrogate --verbose`.

[ddc.txt](https://github.com/rockowitz/ddcutil/files/15093567/ddc.txt)

rockowitz, 2024-04-26:
A display can be marked  "Invalid" for any number of reasons.  It may not support DDC/CI, but except for laptop displays this is quite rare. Some monitors allow for disabling DDC/CI in the OSD.   The Nvidia proprietary driver can require some special settings.  Most commonly, the received DDC data packets are not as expected. This is what status DCRC_DDC_DATA indicates.  For more detail, option ***--ddcdata*** (previously ***--ddc***) reports the invalid data instances and option ***--stats errors*** reports data error counts.

Sometimes data errors can be resolved by increasing the wait time (aka sleep time) between writing a request packet to the monitor and attempting to read the reply packet.  Disable dynamic sleep adjustment (***--disable-dynamic-sleep***) and try setting an explicit sleep multiplier, e.g. ***--sleep-multiplier 2.0*** which doubles the time **ddcutil** waits between writing a request packet and reading the reply.  Generally speaking, if a sleep-multiplier value of 2.0 does not resolve the problem, a greater value will not work either.

Looking at your interrogate output, it appears that is not the case that the reply packet contains invalid data, but that there's no returned data at all.  I'm not sure that to make of this, but I would try explicitly setting a high sleep multiplier.

rockowitz, 2024-04-30:
You can ignore that last paragraph about it appearing that no reply packet data was returned.   That was due to a bug in **interrogate**.