rockowitz/ddcutil #413: ddcutil 2.1.4 - display not found; ddcutil 1.4.1 works

p1r473, 2024-05-06:
hey!
having an issue with ddcutil 2.1.4 where it cant find my monitor `Display not found`

however, ddcutil 1.4.1 installed via debian's `apt install dcutil` works fine

I compiled 2.1.4 myself as the debian version is very out of date, didnt seem to get any errors
```
#ddcutil
wget -O /home/pi/jansson-2.13.tar.gz https://digip.org/jansson/releases/jansson-2.13.tar.gz
tar -xzf /home/pi/jansson-2.13.tar.gz  -C /home/pi
cd /home/pi/jansson-2.13
autoreconf --force --install 
./configure
make
sudo make install

wget -O /home/pi/libusb-1.0.27.tar.bz2 https://github.com/libusb/libusb/releases/download/v1.0.27/libusb-1.0.27.tar.bz2
tar -xf /home/pi/libusb-1.0.27.tar.bz2 -C /home/pi
cd /home/pi/libusb-1.0.27
autoreconf --force --install 
./configure
make
sudo make install

wget -O /home/pi/ddcutil-2.1.4.tar.gz https://www.ddcutil.com/tarballs/ddcutil-2.1.4.tar.gz
tar -xzf /home/pi/ddcutil-2.1.4.tar.gz -C /home/pi
cd /home/pi/ddcutil-2.1.4 
autoreconf --force --install 
./configure
make
sudo make install
```

im controlling it with a little script:
```
#!/bin/bash

# Enable and disable HDMI output on the Raspberry Pi

is_off ()
{
    ddcutil getvcp d6 | grep "0x04" >/dev/null
}

turn_on ()
{
    output=$(ddcutil setvcp d6 01 2>&1)

    # Loop until the output does not contain the verification failed message
    while [[ $output == *"Verification failed for feature d6"* ]]; do
        sleep 1 # Wait a bit before retrying to avoid spamming the command
        output=$(ddcutil setvcp d6 01 2>&1)
    done
}

# Function to set brightness
set_brightness ()
{
    local brightness=$1

    # Check if the brightness level is an integer and within the 0-100 range
    if [[ $brightness =~ ^[0-9]+$ ]] && [[ $brightness -ge 0 && $brightness -le 100 ]]; then
        ddcutil setvcp 10 $brightness
    else
        echo "Error: Brightness level must be an integer between 0 and 100." >&2
        return 1
    fi
}

case $1 in
    off)
        ddcutil setvcp d6 04
    ;;
    on)
        if is_off
        then
            turn_on
        fi
    ;;
    status)
        if is_off
        then
            echo off
        else
            echo on
        fi
    ;;
    toggle)
        if is_off
        then
            turn_on
        else
            ddcutil setvcp d6 04 
        fi
    ;;
    set)
        set_brightness $2
    ;;
    *)
        echo "Usage: $0 on|off|toggle|status|set [brightness_level]" >&2
        exit 2
    ;;
esac

exit 0

```

p1r473, 2024-05-06:
I tried 2.1.5-dev too and it didnt fix it
```
Failure retrieving DRM resources, errno=95=Operation not supported
```

p1r473, 2024-05-07:
Ok let me know if you'd like me to test a fix when you are able to

rockowitz, 2024-05-08:
@p1r473 I've made a couple changes that reduce the use of /sysfs for video drivers that may not use /sys properly.  Let me know if they address your problem.

p1r473, 2024-05-08:
@rockowitz Its working sometimes but I am also getting mostly error output 
[ddc.log](https://github.com/rockowitz/ddcutil/files/15253058/ddc.log)

`Failure retrieving DRM resources, errno=95=Operation not supported `
and 
```
Setting value failed for feature xD6, rc=DDCRC_RETRIES(-3007): maximum retries exceeded
    Try errors: DDCRC_RETRIES
```
However it is not occasionally working

Also an issue with `interrogate`- it dumps the entire dir of my home folder. causing this massive 5mb log file
```
*** Extended dump of sysfs video devices for ARM architecture ***
Display devices: (class 0x03nnnn)
      dirname: missing operand
      Try 'dirname --help' for more information.
      .:
      total 166968
      drwxr-xr-x   3 pi   pi       4096 Apr 16 19:08 AAXtoMP3
      drwxr-xr-x   3 pi   pi       4096 Feb 12 15:33 absminitowerkit
      drwxr-xr-x   3 pi   pi       4096 Feb 12 15:33 Adafruit_LED_Backpack
      -rwxr-xr-x   1 pi   pi       1039 May 25  2021 adlists-deprecated.sh
      -rwxr-xr-x   1 pi   pi       2255 Dec  5 16:11 adlist-sequence.sh
      drwxr-xr-x   3 pi   pi       4096 Apr 26 15:48 almostontop
      -rwxr-xr-x   1 pi   pi        524 Jun  1  2021 android-otp-extractor.sh
      drwxr-xr-x   4 pi   pi       4096 Apr 26 17:05 ansiweather
      drwxr-xr-x   5 pi   pi       4096 May  6 15:57 archivemount
      drwxr-xr-x   7 pi   pi       4096 May  6 15:13 autojump
```

p1r473, 2024-05-21:
@rockowitz Its working sometimes but I am also getting mostly error output 
[ddc.log](https://github.com/rockowitz/ddcutil/files/15253058/ddc.log)

```
Failure retrieving DRM resources, errno=95=Operation not supported
```
and 
```
Setting value failed for feature xD6, rc=DDCRC_RETRIES(-3007): maximum retries exceeded
    Try errors: DDCRC_RETRIES
```
However it is now sort of actually working

Also an issue with `interrogate`- it dumps the entire dir of my home folder. causing this massive 5mb log file
```
*** Extended dump of sysfs video devices for ARM architecture ***
Display devices: (class 0x03nnnn)
      dirname: missing operand
      Try 'dirname --help' for more information.
      .:
      total 166968
      drwxr-xr-x   3 pi   pi       4096 Apr 16 19:08 AAXtoMP3
      drwxr-xr-x   3 pi   pi       4096 Feb 12 15:33 absminitowerkit
      drwxr-xr-x   3 pi   pi       4096 Feb 12 15:33 Adafruit_LED_Backpack
      -rwxr-xr-x   1 pi   pi       1039 May 25  2021 adlists-deprecated.sh
      -rwxr-xr-x   1 pi   pi       2255 Dec  5 16:11 adlist-sequence.sh
      drwxr-xr-x   3 pi   pi       4096 Apr 26 15:48 almostontop
      -rwxr-xr-x   1 pi   pi        524 Jun  1  2021 android-otp-extractor.sh
      drwxr-xr-x   4 pi   pi       4096 Apr 26 17:05 ansiweather
      drwxr-xr-x   5 pi   pi       4096 May  6 15:57 archivemount
      drwxr-xr-x   7 pi   pi       4096 May  6 15:13 autojump
```

rockowitz, 2024-11-24:
Dump of the home directory was fixed in commit 0f9254e4df2764f9dd5f77083d762371fada74b7.

The "Failure retrieving DRM resources" message should now only be possible **ddcutil environment --verbose**.

However, I am still getting DDC errors on my Raspberry Pi 4 running Debian Bullseye.

rockowitz, 2024-11-25:
DDC communication should now be working on the Raspberry Pi.  Give it a try.