rockowitz/ddcutil #417: libddcutil ddca_get_display_info_list2 returns DDCRC_OTHER (-3022)

digitaltrails, 2024-05-15:
A ddcutil-service user raised an [issue](https://github.com/digitaltrails/ddcutil-service/issues/23) where detect is failing resulting in DDCRC_OTHER (-3022) being returned by ddca_get_display_info_list2().

It appears the user has some non-display i2c devices that cannot be accessed. This causes set_ddca_error_detail_from_open_errors()` to return DDCRC_OTHER.  Which in turn results in `ddca_get_display_info_list2()` also returning DDCRC_OTHER even though there are displays in the return list from i2c devices that could be opened.  If some devices are in the list, perhaps `DDCRC_OTHER` is  inappropriate, or needs documenting.

I've pushed a change to `ddcutil-service` that filters `DDCRC_OTHER` which has fixed the user's issue.

If other projects are using libddcutil, it may be causing them issues as well.  KDE *PowerDevil* comes to mind and, of course, *ddcui*.

rockowitz, 2024-06-18:
Commit b3573c2 changes the behaviour of ddca_get_display_refs() and ddca_get_display_info2() to always return 0 (as documented in ddcutil_c_api.h.)

So that information about errors that occur opening individual displays or reading their EDIDs is not lost, they are still reported using ddca_get_error_detail().    In addition, error messages are written to the terminal (which may or may not exist) and, depending on the current syslog level, to the system log.

digitaltrails, 2024-06-20:
Thanks. I modified `ddcutil-service` to check for either behaviour and issue a warning if encountered and return DDCRC_OK for both.

closed 2024-06-25 by rockowitz