rockowitz/ddcutil #418: libddcutil behaves differently/fails after a hotplug compared to ddcutil

digitaltrails, 2024-05-18:
I did a hotplug of the Beng BegQ T2200HD, after which `libddcutil` correctly detected the new monitor, but an attempt to use `libddcutil` getvcp failed with `DDCRC_INTERNAL_ERROR`.  The `ddcutil` command worked fine, bot with detect and setvcp.

I traced this back to `src/ddc/ddc_displays.c` inside 

```
ddc_validate_display_ref(Display_Ref * dref, bool basic_only, bool 
require_not_asleep)
```

Where it appear that the `drm_connector` was NULL:

```
else if (drm_enabled && !basic_only) {
              if (!dref->drm_connector)    <----------------------- NULL
                  ddcrc = DDCRC_INTERNAL_ERROR;
```

Suspecting the driver had not initialised some aspect of drm, I then hacked the call to `ddc_validate_display_ref` to force `basic_only`, and then the setvcp worked.

I then rebooted figuring drm would then be correctly initialised.  Sure enough the `basic_only` hack was no longer required, a getvcp via libddcutil worked just fine.

I don't think you can trust the drm structures if the Nvidia driver is in play.  I suppose this could be covered off with documentation.  Or maybe an option to completely switch off drm.

rockowitz, 2024-06-21:
This looks like another instance of stale  value in /sys/class/drm/<card-connector>/edid, causing dref->drm_connector not to be set.  Probably simplest to use ** ddcutil environment --verbose** to confirm.   Execute the command after a hotplug and also  when the system is started with the monitor plugged in.

digitaltrails, 2024-06-21:
Attachments:
1) Already connected and `nvidia-settings` enabled `ddcutil environment --verbose 2>&1 | tee boot-nvidia-config.log`

[boot-nvidia-config.log](https://github.com/user-attachments/files/15933579/boot-nvidia-config.log)

2) Reboot and hotplug, enable Beng via `nvidia-settings` then `ddcutil environment --verbose 2>&1 | tee hotplug-nvidia-config.log`

[hotplug-nvidia-config.log](https://github.com/user-attachments/files/15933552/hotplug-nvidia-config.log)

`sdiff boot-nvidia-config.log hotplug-nvidia-config.log |less`
```
...
   Display 2                                                       Display 2
      I2C bus:  /dev/i2c-4                                            I2C bus:  /dev/i2c-4
         DRM connector:                         card1-DP-2    <
         /sys/class/drm/card1-DP-2/dpms         On            <
         /sys/class/drm/card1-DP-2/enabled      disabled      <
         /sys/class/drm/card1-DP-2/status       connected     <
         Driver:                                Unknown                  Driver:                                Unknown
         I2C address 0x50 (EDID) responsive:    true                     I2C address 0x50 (EDID) responsive:    true 
         I2C address 0x37 (DDC)  responsive:    true                     I2C address 0x37 (DDC)  responsive:    true 
         Is LVDS or EDP display:                false                    Is LVDS or EDP display:                false
         Is laptop display by EDID:             false                    Is laptop display by EDID:             false
         Is laptop display:                     false                    Is laptop display:                     false
         /sys/bus/i2c/devices/i2c-4/name        NVIDIA i2c ad            /sys/bus/i2c/devices/i2c-4/name        NVIDIA i2c ad
         PCI device path:                       /sys/devices/            PCI device path:                       /sys/devices/
      EDID synopsis:                                                  EDID synopsis:
         Mfg id:               BNQ - UNK                                 Mfg id:               BNQ - UNK
         Model:                BenQ T2200HD                              Model:                BenQ T2200HD
         Product code:         30502  (0x7726)                           Product code:         30502  (0x7726)
         Serial number:        2C803527SL0                               Serial number:        2C803527SL0
...
```