rockowitz/ddcutil #419: Bricked LG Monitor

Slothinger, 2024-05-19:
Hi, i bricked my LG 38WN95CP-W by trying to switch the input with ddcutil. Now it stays black with no osd.
I tried to do a hard reset (remove power + wait  + press the button 10 sec) and i also tried the factory reset option in ddcui.
My pc recognizes the display but it stays black with no sound over hdmi. (tried hdmi, dp, and usbc)

Is there any chance to bring it back to life?

`ddcutil detect --verbose
Display 1
   I2C bus:  /dev/i2c-3
      Driver:                                nvidia
      I2C address 0x50 (EDID) responsive:    true 
      I2C address 0x37 (DDC)  responsive:    true 
      Is LVDS or EDP display:                false
      Is laptop display by EDID:             false
      Is laptop display:                     false
      /sys/bus/i2c/devices/i2c-3/name        NVIDIA i2c adapter 5 at b:00.0
      PCI device path:                       /sys/devices/pci0000:00/0000:00:03.1/0000:0b:00.0/i2c-3
   EDID synopsis:
      Mfg id:               SAM - Samsung Electric Company
      Model:                SyncMaster
      Product code:         1415  (0x0587)
      Serial number:        H9XZ503660
      Binary serial number: 1297691188 (0x4d593234)
      Manufacture year:     2010,  Week: 18
      EDID version:         1.3
      Extra descriptor:        
      Video input definition:    0x80 - Digital Input
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:4:4
         Standard sRGB color space: False
      White x,y:        0.312, 0.329
      Red   x,y:        0.640, 0.330
      Green x,y:        0.300, 0.600
      Blue  x,y:        0.150, 0.060
      Extension blocks: 1
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 4c 2d 87 05 34 32 59 4d   ........L-..42YM
      +0010   12 14 01 03 80 34 20 78 2a ee 91 a3 54 4c 99 26   .....4 x*...TL.&
      +0020   0f 50 54 23 08 00 a9 40 81 80 81 40 81 00 95 00   .PT#...@...@....
      +0030   b3 00 01 01 01 01 28 3c 80 a0 70 b0 23 40 30 20   ......(<..p.#@0 
      +0040   36 00 06 44 21 00 00 1a 00 00 00 fd 00 32 3f 1e   6..D!........2?.
      +0050   51 11 00 0a 20 20 20 20 20 20 00 00 00 fc 00 53   Q...      .....S
      +0060   79 6e 63 4d 61 73 74 65 72 0a 20 20 00 00 00 ff   yncMaster.  ....
      +0070   00 48 39 58 5a 35 30 33 36 36 30 0a 20 20 01 b4   .H9XZ503660.  ..
   VCP version:         2.0
   Controller mfg:      DDC communication failed
   Firmware version:    DDC communication failed
   Monitor uses invalid feature flag in DDC reply packet to indicate unsupported feature.
   Current dynamic sleep adjustment multiplier:  1.00

Display 2
   I2C bus:  /dev/i2c-4
      Driver:                                nvidia
      I2C address 0x50 (EDID) responsive:    true 
      I2C address 0x37 (DDC)  responsive:    true 
      Is LVDS or EDP display:                false
      Is laptop display by EDID:             false
      Is laptop display:                     false
      /sys/bus/i2c/devices/i2c-4/name        NVIDIA i2c adapter 6 at b:00.0
      PCI device path:                       /sys/devices/pci0000:00/0000:00:03.1/0000:0b:00.0/i2c-4
   EDID synopsis:
      Mfg id:               GSM - Goldstar Company Ltd (LG)
      Model:                LG HDR WQHD+
      Product code:         30541  (0x774d)
      Serial number:        304NTGYA7791
      Binary serial number: 347791 (0x00054e8f)
      Manufacture year:     2023,  Week: 4
[... 127 lines cut ...]
         f0: Dynamic contrast
         f2: Unrecognized value
         f9: Unrecognized value
         fa: Unrecognized value
   Feature: DB (Image Mode)
      Values:
         00: No effect
         04: Variable
         fd: Unrecognized value
         fe: Unrecognized value
   Feature: DF (VCP Version)
   Feature: E8 (Manufacturer specific feature)
      Values: 00 01 06 07 FE (interpretation unavailable)
   Feature: E9 (Manufacturer specific feature)
   Feature: EA (Manufacturer specific feature)
      Values: 00 01 02 03 04 (interpretation unavailable)
   Feature: EB (Manufacturer specific feature)
      Values: 00 02 09 FD (interpretation unavailable)
   Feature: F0 (Manufacturer specific feature)
      Values: 00 01 02 03 (interpretation unavailable)
   Feature: F2 (Manufacturer specific feature)
   Feature: F6 (Manufacturer specific feature)`
   
The detect command says something about the "dynamic sleep adjustment multiplier" is this a normal behavior?
Because the display worked while trying the ddcutil commands, but when i went away it didn't wake up from standby anymore.

digitaltrails, 2024-05-21:
> Now it stays black with no osd.

I have a different (older) LG monitor 27UL650 with a *display controller type*  `Mstar (sl=0x05), controller number: mh=0x00, ml=0xff, sh=0x00`.  Not sure if that's in anyway similar, but I thought I ask some questions anyway...

Are you saying he little button on the monitor no longer brings up the monitor's internal on-screen-display?  That should be independent of the input connection. It should work no matter what the monitor is connected to.  If the monitors internal OSD isn't coming up, that may mean the panel, or some part of monitor, is no longer working.  Could it be a warranty-type defect kicking in and perhaps only be coincidentally related to ddcutil?  In which case you could just send it in for warranty repair reporting that the panel seems dead.

Given you are posting outputs, it seems the monitor is still responding to DDC commands.  It seems like the current input source is DisplayPort-1:

> VCP code 0x60 (Input Source                  ): DisplayPort-1 (sl=0x0f)

Experimenting with my own monitor...  

If I switch to an invalid input I can definitely use the monitor's internal OSD to switch back (or also use ddcutil to switch back).  

I even have a input listed in the ddcutil output which does not actually exist on the back of the monitor and isn't listed in the OSD.  Selecting this non-input just results in it naturally resetting to a real input.

One issue I've had is if the nvidia driver loses track of the monitor, I may have to use the GUI *NVIDIA Settings->Display Configuration* to bring it back online.  But the monitors OSD still works in those circumstances, just the PC output is blank.

Sounds like you've tried different cables.  Have you tried a different PC/Laptop?

Slothinger, 2024-05-21:
When i press the powerbutton of the monitor, only the button lights up and the display stays completely dark.
This would be a very big coincidence, if the monitor randomly dies when i was messing around with this tool some minutes before.
I dont think its Nvidia related as i tried it with a notebook (intel), android phone over usbc and an old mac. All devices recognize the display. (resolution, power, etc)

@digitaltrails Thank you for your answer and the link! I'll add this file asap to this issue.

Does someone know if there is a different way to get the firmware files for lg monitors? Sadly the support website doesnt list one for this monitor.

I dont know if i can fix this on my own, but i hope i can prevent other users of this monitor to get into the same trouble...

rockowitz, 2024-05-26:
I'm sorry to hear of your apparently bricked monitor.  There have been a few reports of this, but always with brand-X Chinese imports. So I would not give up hope just yet.

I have a few questions and comments. 

You say that the problem is associated with using feature x60 to change input source.  But you also say that input source switching worked, but the monitor wouldn't wake up after it subsequently went to sleep. This suggests that the problem is not with feature x60.

Were you trying to do something non-standard with feature x60, such as using it in conjunction with features in the mfg range xEO..xFF, or using an alternate source address, as has been described by people trying to reverse engineer Picture-By-Picture, etc.

When diagnosing problems you want to use option ***--disable-dynamic-sleep*** .   The dynamic sleep multiplier reported (.0.00)  is VERY aggressive and can be the source of communication errors, though I don't see evidence of that in your output.

Some monitors will respond to DDC requests no matter which input it is sent on.  Others will respond only if the ddcutil command is issued on the currently active input.

It is possible that the brightness value is so low that the monitor is black. Try issuing command **ddcutil setvcp x10 100 --dis 2** to turn the brightness up to max.

Some monitors do not use the input source values specified in the Monitor Control Command Set spec, but instead use non-standard values. Play around with alternate values.

Slothinger, 2024-05-27:
> I'm sorry to hear of your apparently bricked monitor. There have been a few reports of this, but always with brand-X Chinese imports. So I would not give up hope just yet.
>
> I have a few questions [...]
When i was trying to switch the input it only went black for half a sec and then it came back to the current input.
I tried some commands with ddcutil and after that the monitor was definitely working, but the first sleep 10 min later or so killed it. mby i changed some power settings or something?

> Were you trying to do something non-standard with feature x60, such as using it in conjunction with features in the mfg range xEO..xFF, or using an alternate source address, as has been described by people trying to reverse engineer Picture-By-Picture, etc.
> 

I was trying some commands which are described here:
https://github.com/rockowitz/ddcutil/wiki/Switching-input-source-on-LG-monitors
Something like this: "ddcutil -d 1 setvcp xF4 x0090 --i2c-source-addr=x50 --noverify"

> When diagnosing problems you want to use option _**--disable-dynamic-sleep**_ . The dynamic sleep multiplier reported (.0.00) is VERY aggressive and can be the source of communication errors, though [...]

Thx for your help! I'll try this and report back asap!

> Some monitors do not use the input source values specified in the Monitor Control Command Set spec, but instead use non-standard values. Play around with alternate values.

digitaltrails, 2024-05-28:
I see LG offers `OnScreen Control` software to control this monitor from MacOS or Windows. Perhaps you could plug it into a MacOS or Windows machine and try you luck with LG's software.

Slothinger, 2024-05-28:
> I see LG offers `OnScreen Control` software to control this monitor from MacOS or Windows. Perhaps you could plug it into a MacOS or Windows machine and try you luck with LG's software.

I already tried messing around with different tools so set the default values like brightness, contrast etc but that didn't help.
I was hoping that the lg onscreen control provides a reset function but it is missing this and many features from the real osd. 
The other software i tried was "EnTech Monitor Asset Manager" which had a reset function but that didn't work either.
As the monitor was under warranty i had to turn it in for a repair, so i think we can no longer analyze the problem...

leinardi, 2024-10-12:
Hi @Slothinger,

I also have an LG 38WN95CP-W, and I've been considering using ddcutil to switch inputs. After reading about your experience, though, I'm a bit hesitant since I don't want to risk damaging my monitor.

Do you happen to recall the exact command that caused the issue with your monitor? It would be really helpful to know, as I'd like to switch the input via ddcutil, but I want to avoid any potential pitfalls.

quickishfm, 2024-10-12:
I'm facing a similar issue on LG 27GN850-B. My monitor actually works but it keeps saying "Out of Range" for 1080p, 1440p, and 4k (despite the fact it's a 1440p 144hz monitor).

In my KDE display configuration it now identifies as "LG HDR 4K" instead of ULTRAGEAR as its supposed to. Same in `ddcutil detect`.

I got to this predicament by
- Having been following the discussions around input switching not working and trying a few commands, including the same one as OP listed
- Running `i2cdetect` on the i2c bus and trying commands not only on x50 but on x30 and x37 (the only other things present on the bus) in case it was that

Nothing changed _in the moment_, but I went away for a bit and the monitor entered its power saving mode. When I woke it, this error started.

Factory resetting didn't do anything to fix it, same with unplugging power and displayport cables and holding the button etc, nothing works.
I don't think it's a hardware issue. This happened so soon after my DDCUtil experiments compared to having owned the monitor for a few weeks, it cannot be a coincidence. I don't know exactly which commands I ran (it's probably in my bash history somewhere) to get to this point, especially because my issue only became known after it power cycled (or suspend-resume cycled I guess).
I fear I wrote a value to a configuration register somewhere with ddcutil that told the firmware it's an LG HDR 4K display or something. Short of reverse engineering the firmware myself I'm not sure what to do, so I'm posting on this issue because it's the only still-active post that's related to my problem (and the initial cause is very similar too).

I can have a look at reverse engineering the FW in something like Ghidra, I've done it before, but I don't know how I'd get a hold of it without disassembling my monitor and reading the EEPROM because I don't see any firmware downloads online for my specific model.

Any help on this would be really helpful. I'm happy to run any diagnostic commands for this.

digitaltrails, 2024-10-12:
> I don't know exactly which commands I ran (it's probably in my bash history somewhere) to get to this point,

It's probably worth using grep to find all the commands you ran, they might yield clues.

At this point I would not exclude a hardware fault. Exercising previously unused features may trip over a fault. Changing a setting causes an NVRAM rewrite - what if the NVRAM has a bad block?

Slothinger, 2024-10-12:
> Hi @Slothinger,
>
> I also have an LG 38WN95CP-W, and I've been considering using ddcutil to switch inputs. After reading about your experience, though, I'm a bit hesitant since I don't want to risk [...]

Hi, i think i dont have this bash history anymore.
I havent tried it since i got the monitors mainboard replaced.
One thing i noticed was the really fast on screen menu after i got the monitor back. At first i thougt they gave me a newer revision but that only lasted some days and now its slow as f agian. (Would be so nice if i could use ddcutil :)

@quickishfm
I think my monitor also died when it went to sleep after messing with the commands.
Btw im also using KDE (fedora) but i dont think this could be the problem.
Back then i asked LG Support for the firmware files, but they only wanted to help me get my mainboard replaced...

quickishfm, 2024-10-12:
Thanks @digitaltrails I'll go through the bash history thoroughly. My issue is that whatever command I'll find, I won't know exactly which value to set it to fix it again, if it's even possible (but i think it is). I'll also need to powercycle the monitor each time to check if it made a difference or not.

I didn't have any luck finding the firmware online, as expected, so I took it apart and found an SPI ROM (GD25Q127CSIG) that I dumped using CH341A programmer. Running binwalk on it I only see
```
DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
1759907       0x1ADAA3        DER SHA256 hash
```

Running it through Ghidra it doesn't recognise the language automatically. I'm thinking it might be encrypted, which is kinda annoying, I don't have the skills to dive into it more than just extracting the firmware with binwalk. Will look into this some more.

If this firmware will be useful to anyone, please contact me and I'll upload it. I don't want to pollute the thread with files since I don't think it'll be useful in this state.

digitaltrails, 2024-10-12:
> firmware

Presumably the firmware persists settings to NVRAM reserved for data, so unless you can restore the data related NVRAM, I can't see that reinstalling the firmware would make a difference.  

My speculation was that if you could identify the setting that was changed, the possible valid values for that setting may be teased out of the DDC metadata, or by seeing what a working LG display has in that setting.

Personally, I would have sent it in under warranty, reporting the exact symptoms without confusing them with any description of what might have caused the issue (which is just speculation).

quickishfm, 2024-10-14:
I bought this 2nd hand from a colleague so I don't think it's applicable for warranty. I've emailed the support team nonetheless.

I've gone through my bash history and didn't seem to find anything. I tried a brute force type thing to spam values for setvcp xd6/x60/xf4 with numbers and reboot the display and then grep `ddcutil detect` for any change, but nothing so far. (I achieve a reboot programmatically by writing to xf6 which I found while playing around.)

Will play around some more but ultimately I think I may have overwritten the internet configuration of this monitor into a very bad state. The EDID presents itself as an "LG HDR 4K" yet anything above 1080p (or maybe lower) shows "Out of Range" despite it displaying the image just fine. So I think something must have gone wonky somewhere internally for this to occur. Kinda annoying that something done via software can affect a monitor permanently like this - one would assume the configuration for the screen is burnt into a ROM and not writeable - at the very least not with DDC commands  :/

The price for just wanting to input switch with a hotkey, I guess :P

quickishfm, 2024-10-15:
I cannot believe it; my debug method of thrashing random values at the manufacturer VCP codes fixed my issue. I wish I could explain why, but I haven't a clue. Here's how I fixed it.

I issued the command
```
ddcutil setvcp xe4 x00 --disable-dynamic-sleep --i2c-source-addr=x50
```

This put the monitor into some sort of diagnostic mode. See this image
[image]

I didn't know what to do so I pressed the control knob toward me. This flashed it white and when my computer display came back, everything was good. 1440p at 144hz, no Out of Range issue.
I think this did some sort of "Master Factory Reset" because my Power On Hours value has gone down to 0, from 5000+ from the previous owner. I don't think this is a huge issue personally but this is a fair warning to anyone perhaps trying this. I still don't know if this has any adverse effects but it looks for now to all be working.

The `ddcutil detect` input now shows ULTRAGEAR like before, too.

I'm still rather annoyed that such an issue can be caused by DDC commands but I'm posting this in case it helps someone else. Still happy to provide diagnostic information to @rockowitz or anyone who wants it. Rockowitz it might be worth mentioning this command in your wiki in case someone comes across the same issue. I know I was googling for days trying to resolve this.

Input Switching remains a pipe dream for this monitor though :/

parth-gith, 2026-09-27:
thanks a lot bro @quickishfm !!!!!!!!!!!!!!!!!!!!!!!!
You are life saver !!

I was trying out some vcp commands on my LG 32UN650 for automation and remote control and started facing this issue out of nowhere. Now its working like before. Only serial no and mfg month is changed to factory defaults if i try ddcutil detect. Everything else looks good !

Would love to know how you reverse engineered the firmware and found this vcp code. 
thanks again @quickishfm