rockowitz/ddcutil #420: Ubuntu 24.04 LTS prebuild package

nitramek, 2024-05-21:
Hi,

It would be great if there was an option to download prebuild package for Ubuntu 24.04.
I tried to look for them in openSUSE Build Service and Launchpad but looks like they are not yet available.

I am kind of new to Linux and I know I can build from source in the meantime, but the prebuild option would be great.

w8jcik, 2024-05-28:
Maybe I misunderstood something, but package for Ubuntu 24.04 is available with `sudo apt install ddcutil`.

w8jcik, 2024-05-31:
That being said, the PPA is missing package for Ubuntu 24.04
https://launchpad.net/~rockowitz/+archive/ubuntu/ddcutil/+packages

For older versions of Ubuntu one can install the latest `ddcutil` from binaries, but not for Ubuntu 24.04.