rockowitz/ddcutil #421: Unable to set input source on Gigabyte M27Q

v4u6h4n, 2024-05-23:
Hey everyone :-) I've been able to use ddcutil to put my monitors to sleep, but I haven't had any luck setting the input source with `ddcutil setvcp 60 0x12 --sn 23190B004259 --verbose`, I get the following error:

```
Feature definition file not found: GBT-M27Q-9997.mccs
Feature 0x60 does not support verification
```

Does this indicate I need to setup a [udf file](https://www.ddcutil.com/udf/)?

Also, unrelated, but thought I'd note that I'm happy to post some logs about my monitor model for the ddcutil monitor compatibility list you maintain, if it helps, just need instructions on what do to.

rockowitz, 2024-05-26:
Feature definition file are optional.  The message you are seeing is informational.  You are seeing it because ***--verbote*** output is requested.

Similarly, "Feature x60 does not support verification is also an informational message.  For most features, a **setvcp** command internally does a **getvcp** to ensure that the value has been set correctly.  For some features, verification does not occur, commonly because a feature is write-only.  In case of feature x60, once the input source has changed it may or may not be the case that it is possible to issue DDC requests on the input that was just switched away from. So verification does not automatically occur for that feature.

Some monitors do not use the values given in the Monitor Control Command Set spec, so it may be that x12 is unrecognized.  Try other feature values and see what happens.   If that is the case, you may want to use a UDF file so that **ddcutil** can correctly interpret values read by **getvcp**

v4u6h4n, 2024-05-26:
Hey @rockowitz,thanks for the response :-)

I've been using the ddcutil to turn the displays on and off for a while without any issues, so perhaps your onto something regarding the monitor not using the values given in the monitor control command set spec, I'll look into UDF and see if I have any luck.

10Nico17, 2024-12-13:
> Hey everyone :-) I've been able to use ddcutil to put my monitors to sleep, but I haven't had any luck setting the input source with `ddcutil setvcp 60 0x12 --sn 23190B004259 --verbose`, I get the [...]

Hi,
I have exactly the same issue, did you find a solution ?

v4u6h4n, 2024-12-14:
Hey there, unfortunatley no, but I didn't get much further, so there may be a way around the issue. I'll probably give it another crack in a few months once ive got some spare time.