rockowitz/ddcutil #422: Failed assertion in  check_video_adapters_list_implements_drm

maralorn, 2024-05-23:
Hello and thank you for ddcutil!

After updating my system (to nixos-24.05) I am experiencing crashes when running `ddcutil` (2.1.4) from within the `systemd-udevd` systemd (255.6) services. The same command works when run as my user so this is not necessarilly a bug in `ddcutil` and I suspect it has something to do with the hardening of the systemd unit. The systemd unit runs as root so it is not simply the missing i2c group, I think.

The same setup worked fine before the update (ddcutil 2.0.0). But I don’t know if this breakage is because of the newer systemd version (was 254.10 before) or the newer ddcutil version or something completely else, e.g. some default config change from the distro.

```
           PID: 692027 (ddcutil)
           UID: 0 (root)
           GID: 0 (root)
        Signal: 6 (ABRT)
     Timestamp: Fri 2024-05-24 00:31:04 CEST (57min ago)
  Command Line: /nix/store/s9nim482nrzaxd7y2divg8i56l542ax3-ddcutil-2.1.4/bin/ddcutil -l $'DELL S2721QS' setvcp 60 15
    Executable: /nix/store/s9nim482nrzaxd7y2divg8i56l542ax3-ddcutil-2.1.4/bin/ddcutil
 Control Group: /system.slice/systemd-udevd.service/udev
          Unit: systemd-udevd.service
         Slice: system.slice
       Boot ID: 0f769c3491ad415ba02734b7b4f30a8c
    Machine ID: 56308aa9949b4926946f1c9146b0819d
      Hostname: zeus
       Storage: /var/lib/systemd/coredump/core.ddcutil.0.0f769c3491ad415ba02734b7b4f30a8c.692027.1716503464000000.zst (inacces>
       Message: Process 692027 (ddcutil) of user 0 dumped core.
                
                Module libcap.so.2 without build-id.
                Module libpcre2-8.so.0 without build-id.
                Module libdrm.so.2 without build-id.
                Module libusb-1.0.so.0 without build-id.
                Module libudev.so.1 without build-id.
                Module libjansson.so.4 without build-id.
                Module ddcutil without build-id.
                Stack trace of thread 692027:
                #0  0x00007f2c57d69efc __pthread_kill_implementation (libc.so.6 + 0x8fefc)
                #1  0x00007f2c57d19e86 raise (libc.so.6 + 0x3fe86)
                #2  0x00007f2c57d02935 abort (libc.so.6 + 0x28935)
                #3  0x00007f2c57d02859 __assert_fail_base.cold (libc.so.6 + 0x28859)
                #4  0x00007f2c57d126e6 __assert_fail (libc.so.6 + 0x386e6)
                #5  0x0000556d3afb94d3 check_video_adapters_list_implements_drm (ddcutil + 0x9e4d3)
                #6  0x0000556d3afb94f8 check_all_video_adapters_implement_drm (ddcutil + 0x9e4f8)
                #7  0x0000556d3af6d3e4 submaster_initializer (ddcutil + 0x523e4)
                #8  0x0000556d3af5c860 main (ddcutil + 0x41860)
                #9  0x00007f2c57d0410e __libc_start_call_main (libc.so.6 + 0x2a10e)
                #10 0x00007f2c57d041c9 __libc_start_main@@GLIBC_2.34 (libc.so.6 + 0x2a1c9)
                #11 0x0000556d3af5d8f5 _start (ddcutil + 0x428f5)
                ELF object binary architecture: AMD x86-64
```
The systemd unit is this here: https://github.com/systemd/systemd/blob/c09f7e5a8208fab5680bfeba8df5707887a8c080/units/systemd-udevd.service.in
and the udev rule (although problem irrelevant)
```
ACTION=="add", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="3496/6/100", RUN+="/nix/store/s9nim482nrzaxd7y2divg8i56l542ax3-ddcutil-2.1.4/bin/ddcutil -l \"DELL S2721QS\" setvcp 60 15
```

I assume this is a systemd configuration problem. My best guess from reading the ddcutil code is that ddcutil cannot access /sys? But I am quite uncertain and maybe you can point me in the right direction. Thanks in advance!

rockowitz, 2024-05-27:
Thanks for the report.

I have hardened branch 2.1.5-dev to avoid the abort.

However, there's a more fundamental problem.  Release 2.x makes extensive use of /sys and drm in an attempt to improve performance.  However, it turns out that there is variation in how video drivers populate /sys.  (What is your hardware platform, what is the video driver?)  In your case, it appears that /sys is not fully populated when **ddcutil** is run by systemd at startup.

maralorn, 2025-07-23:
Sorry for never getting back to you on this. I just want to report that since nixos-25.05 with ddcutil 2.2.0 my setup simply started working again. So whatever you did. Thank you!