rockowitz/ddcutil #425: libddcutil-dev package is missing libddcutil.so

w8jcik, 2024-05-31:
```Dockerfile
FROM ubuntu:24.04

ARG SUSE_NAME="xUbuntu_24.04"

RUN apt update -y
RUN apt upgrade -y

RUN apt install build-essential -y
RUN apt install libglib2.0-dev -y
RUN apt install curl -y

RUN echo "deb http://download.opensuse.org/repositories/home:/rockowitz/${SUSE_NAME}/ /" | tee /etc/apt/sources.list.d/home:rockowitz.list
RUN curl -fsSL "https://download.opensuse.org/repositories/home:rockowitz/${SUSE_NAME}/Release.key" | gpg --dearmor | tee /etc/apt/trusted.gpg.d/home_rockowitz.gpg > /dev/null

RUN apt update -y
RUN apt install ddcutil libddcutil-dev -y
```

Listing files `dpkg -L libddcutil-dev`

```
/.
/usr
/usr/include
/usr/include/ddcutil_c_api.h
/usr/include/ddcutil_macros.h
/usr/include/ddcutil_status_codes.h
/usr/include/ddcutil_types.h
/usr/lib
/usr/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/cmake
/usr/lib/x86_64-linux-gnu/cmake/ddcutil
/usr/lib/x86_64-linux-gnu/cmake/ddcutil/FindDDCUtil.cmake
/usr/lib/x86_64-linux-gnu/pkgconfig
/usr/lib/x86_64-linux-gnu/pkgconfig/ddcutil.pc
/usr/share
/usr/share/doc
/usr/share/doc/libddcutil-dev
/usr/share/doc/libddcutil-dev/AUTHORS
/usr/share/doc/libddcutil-dev/CHANGELOG.md.gz
/usr/share/doc/libddcutil-dev/NEWS.md.gz
/usr/share/doc/libddcutil-dev/changelog.Debian.gz
/usr/share/doc/libddcutil-dev/copyright
/usr/lib/x86_64-linux-gnu/libddcutil.so
```

```
ls /usr/lib/x86_64-linux-gnu -l
```

```
...
lrwxrwxrwx  1 root root       19 Feb  7 09:00 libddcutil.so -> libddcutil.so.5.1.2 
...
```

(boken link, `/usr/lib/x86_64-linux-gnu/libddcutil.so.5.1.2` does not exist)

w8jcik, 2024-06-06:
Sorry for closing and reopening the issue. I thought I found the solution, but I was wrong. The issue is still here.

w8jcik, 2024-06-18:
Thank you for fixing it. The library is present now and it works fine. The issue can be closed.

rockowitz, 2024-06-18:
Fixed flibddcutil5-2.1.3-1  and libddcutil5-2.1.4-1 for Debian and Ubuntu releases.   For RPM based distributions (e.g. openSUSE, Fedora), the packaging is orchestrated by file ddcutil.spec.  For Debian based distributions, build is orchestrated by the files in directory debian of packages ddcutil_2.1.4-1.debian.tar.xz, etc.

closed 2024-06-18 by rockowitz