rockowitz/ddcutil #426: Enhancement - consider packaging module loading

digitaltrails, 2024-06-02:
When I packaged `ddcutil-service` for OpenSUSE, I was advised to add module loading to the spec so that users would not have to deal with that themselves.

I added the following to the [ddcutil-service.spec](https://github.com/digitaltrails/ddcutil-service/blob/master/ddcutil-service.spec):

```
# Module loading on subsequent boots
install -m 0644 -D -t %{buildroot}%{_prefix}/lib/modules-load.d/ i2c-dev/%{name}.conf

# Module load for the current boot:
# The post modprobe is helpful for the user, but not really necessary, don't error if it fails
%post
%{_prefix}/sbin/modprobe i2c-dev ||:
```

You might like to consider whether `ddcutil` or `ddcui` should do something similar. Or whether this might be bundled in with the i2c rules package.

rockowitz, 2024-06-05:
**make install**  has done this since ddcutil release 2.0.0 and ddcui 0.5.4  See the [release notes](http://www.ddcutil.com/release_notes/#installation and [ddcui release notes](http://www.ddcutil.com/ddcui_release_notes/).   Whether the distribution packages (rpm, deb, etc.) install the file depends on the distribution and its procedures.  I supply the official debian packages which does this (or more precisely will do so once the packages are updated to the current version.

digitaltrails, 2024-06-05:
This issue can therefore be closed?  

In my case, I should raise this as an issue with the OpenSUSE package/maintainer on OpenSUSE build system.