rockowitz/ddcutil #431: ddcutil 2.1.4-1: DDC communication failed

Cherkah, 2024-06-20:
please find bellow the log attachment

[attach.txt](https://github.com/user-attachments/files/15916300/attach.txt)

After updated my KDE-plasma system from 6.05 to 6.1, i get in my logs:

ddcutil:
```
busno=6, sleep-multiplier =  2.00. Testing for supported feature 0x10 returned Error_Info[EIO in ddc_write_read_with_retry, causes: EIO]

busno=5, sleep-multiplier =  2.00. Testing for supported feature 0x10 returned Error_Info[DDCRC_RETRIES in ddc_write_read_with_retry, causes: DDCRC_DDC_DATA(15)]
```

ddcutil detect
```
Invalid display
I2C bus:  /dev/i2c-5
DRM connector:           card0-eDP-1
EDID synopsis:
Mfg id:               AUO - UNK
Model:                
Product code:         8685  (0x21ed)
Serial number:        
Binary serial number: 0 (0x00000000)
Manufacture year:     2016,  Week: 0
DDC communication failed. (getvcp of feature x10 returned Error_Info[DDCRC_RETRIES in ddc_write_read_with_retry, causes: DDCRC_DDC_DATA(10)])
This appears to be a laptop display.  Laptop displays do not support DDC/CI

Invalid display
I2C bus:  /dev/i2c-6
DRM connector:           card0-HDMI-A-1
EDID synopsis:
Mfg id:               PHL - Philips Consumer Electronics Company
Model:                PHILIPS FTV
Product code:         0  (0x0000)
Serial number:        
Binary serial number: 16843009 (0x01010101)
Manufacture year:     2017,  Week: 12
DDC communication failed. (getvcp of feature x10 returned Error_Info[EIO in ddc_write_read_with_retry, causes: EIO])
```
inxi -G:
```
Graphics:
Device-1: NVIDIA TU117M [GeForce GTX 1650 Mobile / Max-Q] driver: nvidia
v: 550.90.07
Device-2: Chicony Integrated Camera driver: uvcvideo type: USB
Display: wayland server: X.org v: 1.21.1.13 with: Xwayland v: 24.1.0
compositor: kwin_wayland driver: X: loaded: nvidia unloaded: modesetting
gpu: nvidia resolution: 1: 1920x1080 2: 1920x1080
API: EGL v: 1.5 drivers: nvidia,swrast,zink
platforms: gbm,wayland,x11,surfaceless,device
API: OpenGL v: 4.6.0 compat-v: 4.5 vendor: nvidia mesa v: 550.90.07
renderer: NVIDIA GeForce GTX 1650/PCIe/SSE2
API: Vulkan v: 1.3.279 drivers: nvidia surfaces: xcb,xlib,wayland
```

rockowitz, 2024-06-21:
As the problem arose with the update to Plasma 6.1, I suspect it has to do with changes to PowreDevil, which has had a lot of changes.  See [PowerDevil in Plasma 6.0 and Beyond](https://blogs.kde.org/2024/04/23/powerdevil-in-plasma-6.0-and-beyond/).  You can disable PowerDevil's use of ddcutil by setting environment variable POWERDEVIL_NO_DDCUTIL=1 on startup. Does this resolve the ddcutil failures?

Cherkah, 2024-06-21:
i still have the logs issue at start even with the environment variable (/etc/environment):
`busno=6, sleep-multiplier =  2,00. Testing for supported feature 0x10 returned Error_Info[EIO in ddc_write_read_with_retry, causes: EIO]`

`busno=5, sleep-multiplier =  2,00. Testing for supported feature 0x10 returned Error_Info[DDCRC_RETRIES in ddc_write_read_with_retry, causes: DDCRC_DDC_DATA(10)]`

i 've already reported it on kdebugs (longtime bug) & waiting for a powerdevil update.

fanzhuyifan, 2024-06-22:
> i still have the logs issue at start even with the environment variable (/etc/environment):

You might need the patch in this MR: https://invent.kde.org/plasma/powerdevil/-/merge_requests/379