rockowitz/ddcutil #432: [debian] updating linux-image pkg fails due to ddcutil dkms module failure

laur89, 2024-06-23:
Upgrading `linux-image` package fails with following error:

```
Setting up linux-image-6.8.12-amd64 (6.8.12-1) ...
/etc/kernel/postinst.d/dkms:
dkms: running auto installation service for kernel 6.8.12-amd64.
Sign command: /lib/modules/6.8.12-amd64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
Building module(s)...(bad exit status: 2)
Failed command:
make -j16 KERNELRELEASE=6.8.12-amd64 KVER=6.8.12-amd64 -C /var/lib/dkms/ddcci/0.4.4/build
Error! Bad return status for module build on kernel: 6.8.12-amd64 (x86_64)
Consult /var/lib/dkms/ddcci/0.4.4/build/make.log for more information.
dkms autoinstall on 6.8.12-amd64/x86_64 succeeded for virtualbox
dkms autoinstall on 6.8.12-amd64/x86_64 failed for ddcci(10)
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel: 6.8.12-amd64 failed!
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
dpkg: error processing package linux-image-6.8.12-amd64 (--configure):
 installed linux-image-6.8.12-amd64 package post-installation script subprocess returned error exit status 11
Setting up libmalcontent-0-0:amd64 (0.12.0-1.1) ...
dpkg: dependency problems prevent configuration of linux-image-amd64:
 linux-image-amd64 depends on linux-image-6.8.12-amd64 (= 6.8.12-1); however:
  Package linux-image-6.8.12-amd64 is not configured yet.

dpkg: error processing package linux-image-amd64 (--configure):
 dependency problems - leaving unconfigured
Setting up bind9-dnsutils (1:9.19.24-185-g392e7199df2-1) ...
dpkg: dependency problems prevent configuration of linux-headers-6.8.12-amd64:
 linux-headers-6.8.12-amd64 depends on linux-image-6.8.12-amd64 (= 6.8.12-1) | linux-image-6.8.12-amd64-unsigned (= 6.8.12-1); however:
  Package linux-image-6.8.12-amd64 is not configured yet.
  Package linux-image-6.8.12-amd64-unsigned is not installed.

dpkg: error processing package linux-headers-6.8.12-amd64 (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of linux-headers-amd64:
 linux-headers-amd64 depends on linux-headers-6.8.12-amd64 (= 6.8.12-1); however:
  Package linux-headers-6.8.12-amd64 is not configured yet.

dpkg: error processing package linux-headers-amd64 (--configure):
 dependency problems - leaving unconfigured
Setting up dnsutils (1:9.19.24-185-g392e7199df2-1) ...
Processing triggers for libc-bin (2.38-13) ...
Processing triggers for man-db (2.12.1-2) ...
Errors were encountered while processing:
 linux-image-6.8.12-amd64
 linux-image-amd64
 linux-headers-6.8.12-amd64
 linux-headers-amd64
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
---

ddcutil built&installed off github.

```sh
$ ddcutil --version
ddcutil 2.1.4
```

Is this issue with linux-image package, ddcutil, or something third?

rockowitz, 2024-06-23:
There is a problem building driver ddcci, from package ddcci-dkms.   It is not a problem with ddcutil, which provides no drivers.  As per the output submitted, "Consult /var/lib/dkms/ddcci/0.4.4/build/make.log for more information."

closed 2024-06-23 by rockowitz

laur89, 2024-06-23:
Thanks as always.

If someone else stumbles upon it, [here](https://gitlab.com/ddcci-driver-linux/ddcci-driver-linux/-/issues/39) is the relevant ddcci-driver-linux issue.