rockowitz/ddcutil #433: Cannot get capabilities: Maximum DDC retries exceeded.

SamuelBorn, 2024-06-25:
## System Information:
- **ddcutil Version:** 2.1.2
- **Processor:**                                   Intel® Core™ i5-4690K × 4
- **Graphics:**                                    NVIDIA GeForce GTX 970
- **Firmware Version:**                            2405
- **OS Name:**                                     Fedora Linux 40 (Workstation Edition)
- **GNOME Version:**                               46
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.9.5-200.fc40.x86_64

## Problem Description

One of my monitors is recognized by `ddcutil detect` but everything else fails.

Is this a problem with my hardware not being supported or a software error?

---

Output of `ddcutil detect`:
```
Display 1
   I2C bus:  /dev/i2c-3
   DRM connector:           card0-HDMI-A-1
   EDID synopsis:
      Mfg id:               VSC - ViewSonic Corporation
      Model:                VG2433Smh
      Product code:         4911  (0x132f)
      Serial number:        TXC143300082
      Binary serial number: 16843009 (0x01010101)
      Manufacture year:     2014,  Week: 33
   VCP version:         2.2

Display 2
   I2C bus:  /dev/i2c-4
   DRM connector:           card0-DP-1
   EDID synopsis:
      Mfg id:               VSC - ViewSonic Corporation
      Model:                VX2457
      Product code:         47409  (0xb931)
      Serial number:        
      Binary serial number: 16843009 (0x01010101)
      Manufacture year:     2016,  Week: 22
   VCP version:         2.1
```

Example output of `ddcutil -d 2 capabilities` (took 24s)

```
Unable to get capabilities for monitor on Display_Handle[i2c-4: fd=3].  Maximum DDC retries exceeded.
```

rockowitz, 2024-06-27:
Please submit the output of **ddcutil interrogate** as an attachment.

rockowitz, 2024-06-29:
When you say everything else fails, are you including **getvcp** etc? The **detect** command internally performs **getvcp** operations to test DDC communication and read the VCP version.  The **interrogate** output also shows successful operations.

**capabilities** is the most complicated operation.  It performs several **getvcp** operations, each based on what is returned in the prior operation.   So if you have a marginal video card/monitor, it's the most likely to fail.

Option ***--ddcdata*** (previously named ***--ddc***) reports DDC data errors.  I suspect you'll see lot of them for commands that fail.

It appears that you are using a high sleep multiplier for the monitor (2.0) so there's nothing more to be done there.

Your video card is rather old.  It likely needs the [special Nvidia settings ](https://www.ddcutil.com/nvidia).

SamuelBorn, 2024-07-03:
> When you say everything else fails, are you including getvcp etc? The detect command internally performs getvcp operations to test DDC communication and read the VCP version. The interrogate output also shows successful operations.

Sorry, you are right. `getvcp` works. I only tested setvcp.

```
~ ❯ ddcutil -d 2  getvcp 10         
VCP code 0x10 (Brightness                    ): current value =    70, max value =   100
```

> Option --ddcdata (previously named --ddc) reports DDC data errors. I suspect you'll see lot of them for commands that fail.

Yes, I do:

```
~ ❯ ddcutil -d 2 setvcp 10 50 --ddcdata
DDC: Unexpected source address 0x82, should be 0x6e
DDC: i2c_response_bytes: 82 01 10 ac df 00 ff ff 02 02 6b 32 29 29 29 73 65 74 5f 65 65 70 28 6e 88 02 00 10 00 00 64 00 64 a4 32 29 29 29 73 65
DDC: Unexpected source address 0x82, should be 0x6e
DDC: i2c_response_bytes: 82 01 dd 61 10 00 00 64 00 64 a4 32 6e 88 02 01 dd 00 00 64 00 64 68 32 29 29 29 73 65 74 5f 65 65 70 28 34 30 29 6d 63
DDC: Unexpected source address 0x82, should be 0x6e
DDC: i2c_response_bytes: 82 01 df 63 dd 00 00 64 00 64 68 32 29 29 29 73 65 74 5f 65 65 70 28 34 30 88 02 00 df 00 ff ff 02 02 6b 32 29 29 29 73
DDC: Unexpected source address 0xff, should be 0x6e
DDC: i2c_response_bytes: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
DDC: Unexpected source address 0xff, should be 0x6e
DDC: i2c_response_bytes: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
DDC: Unexpected source address 0xff, should be 0x6e
DDC: i2c_response_bytes: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
DDC: Unexpected source address 0xff, should be 0x6e
DDC: i2c_response_bytes: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
DDC: Unexpected source address 0xff, should be 0x6e
DDC: i2c_response_bytes: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
DDC: Unexpected source address 0xff, should be 0x6e
DDC: i2c_response_bytes: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
Verification failed for feature 10
```

> It appears that you are using a high sleep multiplier for the monitor (2.0) so there's nothing more to be done there.

I did not (consciously) set it this high. Is it because of my old monitor?

rockowitz, 2024-07-05:
I may have misinterpreted your cached sleep multiplier data.  Try explicitly setting a high sleep multiplier using option ***sleep-multiplier 2.0***

rockowitz, 2024-07-09:
I'm afraid there's nothing more I can suggest.

SamuelBorn, 2024-07-10:
Okay, thank you for your time and attention anyway.

Lassebq, 2024-12-05:
Experiencing same issue with Philips 27M2N3500NL. It's brand new.
My other older philips monitor is able to return capabilities though.
```
Display 1
   I2C bus:  /dev/i2c-2
   DRM connector:           card0-HDMI-A-1
   EDID synopsis:
      Mfg id:               PHL - Philips Consumer Electronics Company
      Model:                PHL 246E7
      Product code:         49415  (0xc107)
      Serial number:        AU01720001816
      Binary serial number: 1816 (0x00000718)
      Manufacture year:     2017,  Week: 20
   VCP version:         2.2

Display 2
   I2C bus:  /dev/i2c-5
   DRM connector:           card0-DP-1
   EDID synopsis:
      Mfg id:               PHL - Philips Consumer Electronics Company
      Model:                27M2N3500N
      Product code:         49985  (0xc341)
      Serial number:        UK02412088067
      Binary serial number: 88067 (0x00015803)
      Manufacture year:     2024,  Week: 12
   VCP version:         2.2

```

Lassebq, 2024-12-05:
Experiencing same issue with Philips 27M2N3500NL. It's brand new.
My other older philips monitor is able to return capabilities though.
```
Display 1
   I2C bus:  /dev/i2c-2
   DRM connector:           card0-HDMI-A-1
   EDID synopsis:
      Mfg id:               PHL - Philips Consumer Electronics Company
      Model:                PHL 246E7
      Product code:         49415  (0xc107)
      Serial number:        
      Binary serial number: 1816 (0x00000718)
      Manufacture year:     2017,  Week: 20
   VCP version:         2.2

Display 2
   I2C bus:  /dev/i2c-5
   DRM connector:           card0-DP-1
   EDID synopsis:
      Mfg id:               PHL - Philips Consumer Electronics Company
      Model:                27M2N3500N
      Product code:         49985  (0xc341)
      Serial number:        
      Binary serial number: 88067 (0x00015803)
      Manufacture year:     2024,  Week: 12
   VCP version:         2.2

```

xosevp, 2026-10-22:
On my desktop (intel i5-9500 DP + Samsung LS27A600U) with ddcutil-3.0.2 :
```
$ ddcutil capabilities
Unable to get capabilities for monitor on Display_Handle[i2c-5: fd=3].  Maximum DDC retries exceeded.
```

I have to add "--disable-dynamic-sleep --sleep-multiplier 4" to make it work.

rockowitz, 2026-10-23:
Did **ddcutil capabilities** work without options on 2.2.7?  If so, it's something to investigate.  Otherwise ***-sleep-multiplier 4*** says it all.  The DDC connection is marginal and requires that ddcutil wait 4 times the time given in the spec before attempting to read the reply packet.  Specifying ***--sleep-multiplier*** on the command line should make ***--disable-dynamic-sleep*** unnecessary;  it resets the multiplier to the specified value at the start of execution. Are you finding that it's actually necessary?

rockowitz, 2026-10-24:
I need to correct something I wrote yesterday,  Specifying ***--disable-dynamic-sleep*** is necessary if specifying ***--sleep-multiplier*** with a value greater than 2, e.g. 4.   For details, see my response to issue #657