rockowitz/ddcutil #434: Add an option to return the capabilities for a specific feature.

ventsyv, 2024-07-19:
`ddcutils capabilities`  returns (in part) 

```
   Feature: 60 (Input Source)
      Values:
         1b: Unrecognized value
         0f: DisplayPort-1
         11: HDMI-1
         12: HDMI-2
```

`ddcutils vcpinfo 60 --verbose` does not list 1b (which happened to the the USB-C port)

Suggestion: 
1. Add 1b as a USB-C option
2. Add a command that returns the possible values for a feature. For example:
```
> ddcutils capabilities 60

      Values:
         1b: Unrecognized value
         0f: DisplayPort-1
         11: HDMI-1
         12: HDMI-2
```

This will simplify scripting / GUI development

rockowitz, 2024-07-19:
**vcpinfo** and **capabilities** get their feature value lists from different places.

**vcpinfo** reports the values in the MCCS spec.  Its information is not monitor-specific.  Option ***--mccs*** allows for selection of a specific MCCS version; if not given the most recent MCCS version in which the feature is defined, typically 2.2, is used.  

**capabilities** reports the values shown on the capabilities string returned by the monitor.   Normally, the interpretation of the value is that listed in the MCCS spec (reflecting the MCCS version of the monitor, as given in the capabilities string or, if that is not given, by querying feature xDF).  If the value is unknown to the MCCS spec, "Unrecognized value" is reported. 

However, if the feature is defined in a user defined feature file for the monitor,  the values reported for that feature are those in the UDF file, not those in the capabilities string, and the interpretation is that in the UDF file.   (The uninterpreted string reported by **capabilities** is not modified by the UDF file; it is exactly that returned by the monitor.)

Unfortunately, no value for USB-C is defined in any version of the MCCS spec, which is rather dated.  Some manufacturers use x0b for its value, other manufacturers use a different value, and there's even no guarantee that a given manufacturer uses the same value on all its monitors.  The only way for **ddcutil** to know about a USB-C value is for it to be defined it in a monitor-specific user defined features file. 

As a further comment, the capabilities string is not necessarily reliable.  For example, the capabilities string reported by my Samsung U32H750 is "60( 11 12 0F)".  The interpretation of those values, per the MCCS spec, is HDMI-1, HDMI-2, and DisplayPort-1.  However those are not the actual values used by the monitor.   The monitor recognizes x05 as HDMI-1 and x06 as HDMI-2.  My user defined features file for the monitor, SAM_U32H75x-3586.mccs, contains: 
~~~
MFG_ID       SAM
MODEL        U32H75x
PRODUCT_CODE 3586
MCCS_VERSION 2.1
FEATURE_CODE 60 Input Source
  ATTRS NC RW
  VALUE x05 HDMI-1
  VALUE x06 HDMI-2
  VALUE x0f DisplayPort
~~~

ventsyv, 2024-07-19:
Very informative, thank you.
I'll define a custom MCCS for my monitor then and pub it on github. Is there any keyword or tag that I can apply to make it more easily discoverable?

rockowitz, 2024-07-19:
I don't have any keyword or tag to use, but I would suggest that you (at least) put it in section [Notes on Specific Monitors](https://github.com/rockowitz/ddcutil/wiki/Notes-on-Specific-Monitors) in the wiki.

ventsyv, 2024-07-19:
Thanks, will do!

closed 2024-07-19 by ventsyv